bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #141: ender 3 skr pro v1.1 bltouch problem

3dmadnessnl, 2020-05-17:
hi all
i have a problem whit the bltouch when i wil print with octoprint it wil work 1 time 
wil i print the next print it wil not the norzel go in to the bed 
if i print from sd it go all times good 

setup
ender 3 
bltouch v3 orgineel
skr pro v1.1
TFT35-E3 v3.0 LCD 
marlin bugfix last one
octoprint last one

sorry my english is not too good
who has had the same problem and can help me with this

best regards Dennis

3dmadnessnl, 2020-05-19:
????

Shamoon78, 2020-06-15:
I use bltouch skr pro and octoprint never have issue

Shamoon78, 2020-06-15:
Try demo btt marlin it might work better

3dmadnessnl, 2020-06-21:
no it is the same

SavingLifes, 2020-06-21:
Can you post the Config.h, Congig_adv.h and pins file please? I will investigate. Thank you