bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #143: Want to use E2 for X2

Kiyotimun, 2020-05-21:
Has anyone tried to use E2 for X2. I have a dual extruder and I have everything working but this. I am going crazy when I have config, config.adv and pins and bigtree.pins open trying to put this together.

battlezone, 2020-05-27:
Search for `X_DUAL_STEPPER_DRIVERS` in config advanced and enable it. Then in the pins file change the E2 pins to X2: 
`#define X2_STEP_PIN        PD13
#define X2_DIR_PIN         PG9
#define X2_ENABLE_PIN      PF0
#ifndef X2_CS_PIN
  #define X2_CS_PIN        PG12
#endif`
And same for the E2_serial:
`  #define X_SERIAL_TX_PIN  PD6
  #define X_SERIAL_RX_PIN  PD6`

Kiyotimun, 2020-05-27:
I understand this, maybe I wasn't clear. I am using dual x carriages for the second extruder. I have everything programmed right (IMO) because I can print from extruder one with a few pretty good prints. Extruder 2 leaves a puddle of material since X2 isn't moving to try mirror mode which I actived in gcode in the print.

sensenmann, 2020-06-05:
I think I can help you, please contact me (rene . huber [at] rhse . at). Please inform me here, when you sent a mail (I get a lot of spam recently)