bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #144: SKR pro 1.1, TMC2208 UART, unable to set micro steps in firmware.

DarkShadowsX5, 2020-05-24:
M122 reports that the micro steps is set to 8 but i set them to 32 in firmware. 
01:11:13.042 : msteps		8	8	8	8
01:11:13.074 : tstep		        max	max	max	max
previously it reported that they were set to 256 before soldering the jumper but still moved as if in 1/8th mode. 

#define X_MICROSTEPS 32
set on each driver in use. each have the jumper soldered and the single red jumper on uart mode next to the driver on the SKR pro 1.1 board. 
my steps in configuration.h are currently set to 80 for xy but it moves 20mm instead of 5mm.
(i will set it to a proper 160 so it moves 10mm for 10mm when its working. )

not sure what i can do at this point. Maybe im missing something for proper UART. the documentation and video tutorials are all over the place with all the changes. everything seems out of date. 

for the jumpers i soldered on the driver, there was 2 options and i chose this
[image]

DarkShadowsX5, 2020-05-25:
Upon further investigation i found that hitting the reset button on the tft24 screen board fixes the issue BUT when i do a full power reset and wait for the led to turn off. its reset to 1/8th mstep movement again. hit the reset on screen board and its at 32 msteps.... 

something must be conflicting causing it to boot in 1/8th msteps but be overwritten to what you set the drivers at in Configuration_Adv.h with a soft reset. so the issue is only temporally corrected. 

someone who knows marlin code more intricately may be able to permanently fix the issue.

MarciV2, 2020-10-17:
I have the same issue on my E0-driver slot.
It's not the stepper driver but the slot on the board (at least for me).
Also looking for a solution...

DarkShadowsX5, 2020-10-17:
@MarciV2 
i dont know what exactly caused it to finally work correctly but it takes my micro step changes without a fuss now. 
i reset it and reprogrammed it a bunch is all i could say on that. the reset button on my LCD panel had the most helpful impact in getting it to work temporary. there might be something in the firmware your overlooking for E0. 
i'm sorry i forgot all that i changed at this point that might have fixed it for good.

i recommend keeping E0 at 16 micro-steps. it has the least impact on print quality compared to 32 micro steps on xyz

I had to change a lot of setting to get my printer working correctly and smoothly. 
nothing i changed had anything to do with the memory or Eprom. I can give you my firmware files but it likely wont work well with your printer. I have a 5 year old acrylic frame prusa clone. I'm just offering it as something to look over and compare.

MarciV2, 2020-10-17:
@DarkShadowsX5 The weird thing is, that suddenly the board doesn't even connect the driver anymore...
I've already contacted the BTT support too because the fan pins don't work either...

I have set all of my axis to 1/16 ustepping and sometimes e0 changes to 1/8 while e1 gets set to 1/256 which the drivers (TMC2209) don't even support.

DarkShadowsX5, 2020-10-18:
you might have an actual hardware issue with the stepper drivers. did you try swapping them around to see if the problem moved? 

as for the fans its a bit off topic but this is my settings.  
the fan pins were a pain to setup in firmware.  

in pins_BTT_SKR_PRO_V1 i have the fans pins as:
#define FAN_PIN        PC8
#define FAN1_PIN      PE5
#define FAN2_PIN      PE6 

#define E0_AUTO_FAN_PIN      FAN_PIN

in configuration_adv.h, i have 
#define E0_AUTO_FAN_PIN PE5

not sure if this change was necessary but 
#define FANMUX0_PIN  PE5

i had EXTRUDER_AUTO_FAN_TEMPERATURE working for FAN1_PIN at 50c 
without doing these changes i was getting heat creep and E0 skipping issues. one time it was so bad i had to change the bowden tube adapter because it jammed up.

closed 2022-01-07 by DarkShadowsX5