bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #145: Z mechanical endstop not wokring with X/Y Sensorless

rhukster, 2020-05-24:
I've spent hours scouring the web looking for a solution to this with no luck.  Let me outline my configuration:

I'm building a BLV MGN Cube printer with SKR Pro 1.1.  It has a triple-z axis setup that i'm going to level with a TH3D EZABL Pro sensor.  However, even before I to fully testing the sensor, I'm stuck getting the Z endstop to detect being triggered using a simple OMRON mechanical switch.  

This issue is that I have X/Y axes configured with sensorless homing  with 2209s (diag pin intact), and my Z0, Z2 (E1) and Z3 (E2) drivers, also 2209s (diag pin de-soldered).  The endstop lights next to the extruder connectors are completely off for X/Y/E0, and are dim for Z1, Z2, Z3.  

Because I have my Z endstop inverted in Marlin, when I plugin in the endstop the Z1 light turns on, and when i trigger my the endstop by hand (which is connected to Z_MIN endstop on the board), the Z1 light turns off.  So it seems the board itself is registering the endstop being triggered (even if it's inverted at the moment).

However, while doing this, the `M119` command always reports the end stop is `open`:

```
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open // not activated by hand (board light on)
Recv: ok
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open // activated by hand (board light off)
Recv: ok
```
I've tried pulling all the drivers, and also removing all stepper cables, so only the 1 single Z-endstop cable is attached, and the problem still remains! It seems to me that either there is a physical issue with the board (strange as the light on the board is changing properly), or it's some weird Marlin thing that I've set by accident? I'm using the very latest Marlin `2.x-bugfix` branch that I merge daily so that is 100% fully up to date as of right now.

BTW, sensorless homing on X and Y is working just fine with `G28 X Y`, no issues there at all!

Attached are the configuration files.

Any help with this would be greatly appreciated!

[Configuration-Files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4674537/Configuration-Files.zip)

GhostlyCrowd, 2020-05-24:
Andy, have you followed the correct guidelines for TMC 2209's and cut the proper pin off the stepper driver for the Z so it allows the limit switch to work?

rhukster, 2020-05-24:
Yes, as I originally stated, i have de-soldered (removed) the diag pins on all 3 of the Z-Axis 2209s.  The X, Y and E 2209s still retain their pins.. XY is working fine for sensorless homing. I don't believe the endstop lights on the board would illuminate at all if this was not the case?

GhostlyCrowd, 2020-05-25:
Sorry missed that on mobile i just assumed it was the easiest answer. Ive had a look at your configs nothing is glaring at me.  I'd probably post a bug report on the Marlin bugfix main repo.

rhukster, 2020-05-25:
I just want to make sure it's not a physical board issue before I post on the Marlin repo, but that was definitely my next stop if folks like yourself who are familiar with the SKR Pro, couldn't find any obvious issues. Thanks!

GhostlyCrowd, 2020-05-25:
> I just want to make sure it's not a physical board issue before I post on the Marlin repo, but that was definitely my next stop if folks like yourself who are familiar with the SKR Pro, couldn't find any obvious issues. Thanks!

Just to be clear so i understand, youre using a probe for levelling, the probe is NOT the z endstop, youre using a limit switch as z endstop and the BTT SKR board is showing the proper led configuration for open and trigger of the endstop on z?

If the above is all correct i think you have found an actual bug.

rhukster, 2020-05-25:
I intend on using the probe, but it's not connected right now.  Right now i just have a mechanical endstop plugged into the Z_MIN on the board.  So I have configuration in the files for the probe, but have not gotten that tested yet.  

I originally tried the probe connected to the same Z_MIN (as you are supposed to configure it as it uses it's own controller box).  It did trigger on the EZABL Pro control board correctly (lights up), and also triggered the SKR Pro Z endstop light.  So basically it did exactly the same thing as the mechanical switch.  However, both the mechanical switch and the EZABL triggering still did not show any change via the `M119` command. So I don't think this is probe related as there's not any special probe pins required here.

It seems like the board is seeing the endstop being triggered just fine, but Marlin is not able to detect this via `M119`.  I am using the stanard Z_ENDSTOP pin (`PG8`)

rhukster, 2020-05-26:
Was my own fault! I had the endstops connected to Signal + V, instead of Signal + Ground.

closed 2020-05-26 by rhukster