bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #146: Build errors

anm8tor, 2020-05-26:
I wish I understood more about this to find the right information on my own. I have attempted a few fixes found elsewhere in GitHub, but to no avail. I am building Marlin 2.0.5.3 for the BTT SKR Pro V1.1 for my Eryone Thinker S (a CR10-like printer). Here is the information that AutoBuildMarlin outputs, specifically where the issues occur and create a failed build. I am running VSCode in Admin mode.

Linking .pio\build\BIGTREE_SKR_PRO\firmware.elf
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/persistent_store_api.cpp.o:D:\TSMarlin2053/Marlin\src\HAL\shared/persistent_store_api.h:53: multiple definition of `persistentStore'; .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/eeprom_api.cpp.o:D:\TSMarlin2053/Marlin\src\HAL\shared/eeprom_api.h:53: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M20.cpp.o: in function `GcodeSuite::M20()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M20.cpp:33: multiple definition of `GcodeSuite::M20()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M20.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M20.cpp:33: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M21_M22.cpp.o: in function `GcodeSuite::M21()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M21_M22.cpp:33: multiple definition of `GcodeSuite::M21()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M21_M22.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M21_M22.cpp:33: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M21_M22.cpp.o: in function `GcodeSuite::M22()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M21_M22.cpp:38: multiple definition of `GcodeSuite::M22()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M21_M22.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M21_M22.cpp:38: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M23.cpp.o: in function `GcodeSuite::M23()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M23.cpp:38: multiple definition of `GcodeSuite::M23()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M23.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M23.cpp:38: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M24_M25.cpp.o: in function `GcodeSuite::M24()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M24_M25.cpp:50: multiple definition of `GcodeSuite::M24()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M24_M25.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M24_M25.cpp:50: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M24_M25.cpp.o: in function `GcodeSuite::M25()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M24_M25.cpp:91: multiple definition of `GcodeSuite::M25()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M24_M25.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M24_M25.cpp:91: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M26.cpp.o: in function `GcodeSuite::M26()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M26.cpp:33: multiple definition of `GcodeSuite::M26()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M26.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M26.cpp:33: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M27.cpp.o: in function `GcodeSuite::M27()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M27.cpp:35: multiple definition of `GcodeSuite::M27()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M27.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M27.cpp:35: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M28_M29.cpp.o: in function `GcodeSuite::M28()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M28_M29.cpp:61: multiple definition of `GcodeSuite::M28()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M28_M29.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M28_M29.cpp:61: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M28_M29.cpp.o: in function `GcodeSuite::M29()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M28_M29.cpp:71: multiple definition of `GcodeSuite::M29()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M28_M29.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M28_M29.cpp:71: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M30.cpp.o: in function `GcodeSuite::M30()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M30.cpp:33: multiple definition of `GcodeSuite::M30()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M30.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M30.cpp:33: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M32.cpp.o: in function `GCodeParser::has_value()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M32.cpp:43: multiple definition of `GcodeSuite::M32()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M32.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M32.cpp:43: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M524.cpp.o: in function `GcodeSuite::M524()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M524.cpp:35: multiple definition of `GcodeSuite::M524()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M524.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M524.cpp:35: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sdcard/M928.cpp.o: in function `GcodeSuite::M928()':
D:\TSMarlin2053/Marlin\src\gcode\sdcard/M928.cpp:35: multiple definition of `GcodeSuite::M928()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/sd/M928.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\sd/M928.cpp:35: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/temperature/M104_M109.cpp.o: in function `GCodeParser::has_value()':
D:\TSMarlin2053/Marlin\src\gcode\temperature\../parser.h:141: multiple definition of `GcodeSuite::M104()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/temp/M104_M109.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\temp\../parser.h:141: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/temperature/M104_M109.cpp.o: in function `GCodeParser::seen(char)':
D:\TSMarlin2053/Marlin\src\gcode\temperature\../parser.h:149: multiple definition of `GcodeSuite::M109()'; .pio/build/BIGTREE_SKR_PRO/src/src/gcode/temp/M104_M109.cpp.o:D:\TSMarlin2053/Marlin\src\gcode\temp\../parser.h:149: first defined here
c:/users/anm8t/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/BIGTREE_SKR_PRO/src/src/gcode/temperature/M105.cpp.o: in function `GcodeSuite::M105()':
[... cut ...] .cpp:27: multiple definition of `lang'; .pio/build/BIGTREE_SKR_PRO/src/src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/language/language.cpp.o:D:\TSMarlin2053/Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\language/language.cpp:27: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1
=========================================================== [FAILED] Took 119.85 seconds ===========================================================

oldman4U, 2020-06-09:
Hello anm8tor.

Have you been able to solve the reported issue in the meantime? If not, could you please attach your config and config.adv files.

Thank you

anm8tor, 2020-06-10:
Yes, I ended up fixing the issue. If I remember correctly, it had to do with setting the environment accurately in the ini file, then fixing some options for the menu that were on by default.

oldman4U, 2020-06-10:
This sounds great.

Please help us to keep this system usable and close all tickets you do not need anymore by clicking on the Close button right besides the Comment button.

Thank you

oldman4U, 2020-10-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

anm8tor, 2020-10-15:
Closing due to resolving the issue

closed 2020-10-15 by anm8tor