bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #147: Octopi with uart pins

rnelson197, 2020-05-30:
Is there a way to use the dedicated uart pins on the Skr pro to connect to the raspberry pi.   I connected the RX to TX and TX to RX and the gnd to gnd.  Updated Octopi config.txt and port.    When I attempt to connect to printer through octopi, it cannot connect.  

Unexpected error while connecting to serial port: /dev/ttyAMA0 SerialException: '[Errno 13] could not open port /dev/ttyAMA0: [Errno 13] Permission denied: '/dev/ttyAMA0'' @ comm.py:_openSerial:2691 (hook default)

sketch4x4, 2020-06-07:
have you got 5v into the raspberry pi? if you connect a monitor to the pi does it boot up ok? that would be where I would start.

jbuck2005, 2021-02-08:
This is a permissions problem for the user you are trying to access the hardware serial port with.
you need to:

sudo usermod <<username>> -a -G dialout,gpio,tty

where <<username>> is your <<user>>@<<hostname>> for the command line
you may need to add whatever "user" OctoPrint is being launched as ..