bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #148: Downloading .bin file via usb

MikeEitel, 2020-05-31:
I tried to download a new  firmware.bin file on a running marlin 2.5.3 marlin.
I have not been able to do this. 
It is a pain in my diy deltas to exchange the sd for every new version. 
What do i wrong? 
Thx Mike

P. S. 
2x  skr pro with 6x tmc and tft 3.5

oldman4U, 2020-06-02:
Hi. What is the size of the firmware you try to upload?

More details what exactly does not work could also help.

Thank you

MikeEitel, 2020-06-02:
Can not remember but it is no matter of size.
When i write directly on the sd it works perfect. 
I can not access the sd via ubs on the mks pro only. And i know generally how to do. As doing the same process with mks sbase works perfect.

AnHardt, 2020-06-02:
> What do i wrong?

Absolutely nothing.
It does not work now.

Up to now this is not implemented for the STM32F4xx boards.
There are shy tries at https://github.com/stm32duino/Arduino_Core_STM32/pull/586 to prepare the MMC/CDC composite device for the used framework. But this is currently for SDIO only. Until a SPI version will reach Marlin another year will be gone.

MikeEitel, 2020-06-02:
That is a shame.
The board is nice. 
But producing a board based on a chip that has not full sw support for the intented main market. That does not earn my respect.
Mike

oldman4U, 2020-06-02:
Maybe you will ask for other missing features like usb stick support or
print from sd card. Simple led support or filament runout sensor
connectors. - nope. But you get superb power handling, three controllable
fans and ESP01-S FW update through the board.

Hope this helps

MikeEitel, 2020-06-02:
Does that mean that ESP01-S can write the firmware.bin to the onboard sd, so i do not have to dissasamble my printer for every change?

And sorry, i stick to my opinion: upgrade of the firmware via usb is a basic feature since long time. As an producer neglecting such a basic functionality is in my eyes not state of the art customer orientation

AnHardt, 2020-06-02:
The other way around. SKR-PRO-V1.1s boot-loader can update a connected ESP01-Ss firmware from the onboard SD-card.

oldman4U, 2020-06-02:
Hi Mike.

We all would love to see better USB and SD card support. At least the
possibilities and what does not work should be defined much better.

oldman4U, 2020-06-08:
Hi Mike.

Do you still need this ticket? If not, could you please be so kind and close it.

Thank you

MikeEitel, 2020-06-08:
To be honest
I'm verry tempted to let it in, as missing possibility is a main design feature. A customer oriented supplier should take such sw related drawbacks into acount!
You won't buy a fine sportscar when forced to fuel it by can just behind the rear seats.
The board itself is very nicely made hw, and i guess would have high sales potential. But my impression is that BTT does not do enough for firmware support to assist the marlin comunity. Myself will probaly not think good about btt everyytime i do upgrade on my two fully equipped 6axis boards.

MikeEitel, 2020-06-08:
But to be no too nasty i will close this not solved request

closed 2020-06-08 by MikeEitel

oldman4U, 2020-06-08:
Hi Mike.

I agree with you and I am not working for BTT or get paid by them. I am just a user who tries to clean up the tickets so someone new can find a topic before he starts a new ticket.

AnHardt, 2020-06-08:
Recently there was some movement.
https://github.com/stm32duino/Arduino_Core_STM32/pull/1088
https://github.com/MarlinFirmware/Marlin/pull/17222
https://github.com/MarlinFirmware/Marlin/issues/18157

So maybe i was a bit to pessimistic. (But still a long way to go.)

MikeEitel, 2020-06-08:
Thx Anhardt and oldman4u
I'm aware of the progress in Marlin. Looking nearly every day whats going on.
It's this situation that good peoples like you are trying to solve problems, and not a company like BTT what has to be clearly stated.

Again Big thx for you
Mike

oldman4U, 2020-06-08:
Hi Mike. Feel free to be people like Anhardt and me😁

Sineos, 2020-06-08:
See here for a potential work around I'm using: https://github.com/OctoPrint/OctoPrint/issues/3522