bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #149: 24v fan

drmshady, 2020-06-11:
hi

can I use 24v fan and hot end heater
thank you

oldman4U, 2020-06-11:
Hello drmshady.

This depends on your power supply. The SKR supports 3 different power inputs for Motor, Main and Bed and you can connect 12V OR 24V to each of the inputs. In case you connect a 24V power supply to the MAIN input, you will get 24V on the Fan and hot end output - in case you connect 12V, you will get 12V on the outputs.

Please let me know if you ned any additional information

drmshady, 2020-06-12:
Thank you
I have 24v power supply, so I think  it is ok

oldman4U, 2020-06-12:
It is. I need a downconverter to drive my 12v noctua fans.

Please do not forget to close your ticket once you don’t need it anymore.

Thank you

closed 2020-06-12 by drmshady

pedrovanzella, 2020-09-11:
@oldman4U how did you hook up the buck converter to power your 12V Noctua Fans on a 24V Power Supply?

oldman4U, 2020-09-11:
Hi. 

I connected the converter to the power supply output which also feeds the mainboard main input.

In another setup i used the 24v fan out and connected two fans in a row. But this does not work so well.