bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #151: MicroSD card reader no longer recognises any cards

Malaquitte, 2020-06-27:
Hi,

The micro SD port on the motherboard suddenly stopped working. The TFT35 screen displays the following message "Read onboard SDCard error". I tried a lot of different sdcards, with different format type, never mind. I am now completely stuck because I have no way to flash the firmware on my motherboard.

Any ideas are welcome...

oldman4U, 2020-06-27:
So this happened after a firmware update?

Malaquitte, 2020-06-28:
Maybe, to tell the truth, my last flash was a few days ago and since I use the TFT35 usb port to print, I didn't realize it immediately.

oldman4U, 2020-06-28:
And firmware you can still upload?

Malaquitte, 2020-06-28:
none anymore, that's the problem. No matter what file I have on a card, the message is always the same.

oldman4U, 2020-06-28:
Turn the printer off. Disconnect the TFT from the mainboard. Compile a firmware.bin file, copy it onto a SD card and insert the card into the SD card slot of the mainboard. Turn the printer on. Can you see the firmware update LED flash on the mainboard for approximately 10 seconds, before it stops flashing?

Malaquitte, 2020-06-29:
It's perfect, I managed to flash my motherboard and by plugging my screen I can now flash again.
Thanks for the advice, I would never have thought to incriminate the screen...
That said, it's hard to understand it, sometimes the screen stays black under the Marlin interface after a flash, sometimes the .bin file is not deleted but I see the green led blinking!

oldman4U, 2020-06-29:
So there is some improvement, that's great.

Please let me know in case there is anything else I can help you with. In case everything works fine, please help us to keep this system usable by closing this ticket

Thank you

closed 2020-06-29 by Malaquitte