bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #152: SKR Pro v1.1 TMC2209v1.2 does not sensorless homing

Majestix1970, 2020-06-28:
Hi all,

I have a problem with my new SKR Pro v1.1 and my  TMC2209 v1.2 Driver.

The driver have been used fully functionally in my old SKR 1.4Turbo. After switching the board using the old Marlin 2.0.5.4 configuration sensorless homing stopped working because it does not move at all when doing G28. 
First I thought I need to readjust the sensitivity, but reducing it from the old values (x=130 y=190) to x=54 and y=100) it started to move but bumps hard into the end position with about 3 sec of making noise...
If I adjust the new values  eg. by +1 to X=55 it stops moving again... same for y. 
Then I thought it might be the TMC current. I changed it from the default Marlin value 800mA to 1100mA for all stepper drivers. Unfortunately did not help either.

Any ideas what the issue might be?
Printer is an cartasian Geeetech i3 Pro C Prusa clone.
Was working perfectly with the SKR1.4Turbo...
TMC2209 also work perfectly with the SKR1.4 Turbo.

Hope you can help.
Best regards
Majestix1970

Majestix1970, 2020-06-28:
[Configuration_and_Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4842881/Configuration_and_Configuration_adv.zip)

Majestix1970, 2020-06-28:
here are my .h files...

Majestix1970, 2020-06-29:
sorry closed it unintentionally....

oldman4U, 2020-06-29:
Hi.

I use the same board and driver and it works perfectly. I am sure we do not use the same motors and maybe also other factors are different, so there is not a lot to say - but it does work in general. I saw huge difference depending on the stiffness of the belts and i use 200mA for sensorless homing.

Hope this helps

Majestix1970, 2020-06-29:
Hi oldman4u
Thanks for your reply...
My motors are simple NEMA17 driven with 24V
Belt strength I did check already 
and overall the exact same mechanical setting is working fine with the SKR 1.4 Turbo😂
I assume I got a faulty board.. Grrr

oldman4U, 2020-06-29:
I have 2 SKRs in the meantime, one for testing and one in my printer;-)

So what are you going to do next?

Majestix1970, 2020-06-30:
I am requesting an HW replacement of the SKR Pro 1.1 board
I do not see any issue with my configuration! 

The board is also behaving completly unpredictable. One time the same sensitivy setting makes the print bed hit hard the end position. Next time it does not move at all because it is too sensitive.

oldman4U, 2020-07-01:
Sounds good and makes sense.

Last question. Are you using 12V or 24V and which power supply are you using?

Majestix1970, 2020-07-01:
Hi 
I use 24V and a regulate switched Power supply with 15A. 

Why?

oldman4U, 2020-07-01:
I saw some other tickets in the past, where a broken or cheap power supply also caused random issues.Maybe i can find some time to update to your Marlin version, which version are you using at the moment? 2.0.5 or the latest and greatest from GitHub?

oldman4U, 2020-07-11:
Hi. Any news regarding the reported issue?

oldman4U, 2020-08-03:
@Majestix1970 

Please close this ticket in case you do not need it anymore 

Thank you

oldman4U, 2020-08-26:
Have you replaced the board in the meantime?

Majestix1970, 2020-08-26:
Hi Oldman4U,
have returned the board and closed the ticket.
Thanks for your help

closed 2020-08-26 by Majestix1970

oldman4U, 2020-08-26:
Perfect!

Hope you have more luck with another board.