bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #153: C62 and C67 are too large and cause too-slow crystal start to use system bootloader in DFU mode

zbrozek, 2020-07-02:
The older system bootloader (BOOT0 set high, BOOT1 set low) present on the STM3240xxx series only waits 780 microseconds for the crystal to start. The SKR Pro 1.1 comes with 22 pF capacitors that over-load the crystal such that it can't start quickly enough to enable USB support within the system bootloader.

I replaced these capacitors with 12 pF capacitors and now the startup is fast enough that the SKR Pro 1.1 is able to enumerate as a USB DFU device.

mqt0029, 2020-08-09:
Can someone @bigtreetech verify this is an issue with USB enumeration? Is this why my SKR Pro V1.1 USB-A port to OctoPi or Pronterface never worked no matter what I tried?

zbrozek, 2020-08-09:
@mqt0029 This is only relevant if you're trying to use the DFU bootloader, which is unlikely if you're thinking about OctoPi. Also presumably you meant the USB-B connector?

mqt0029, 2020-08-09:
Apologies was typing it up halfway through trying to get OctoPi to work with this board. And yes I meant the USB-B connector. So far trying all the firmware flashes which claims to have bootloader or USB serial does nothing. #115 .bin reads all F in ST-Link Util, #72 brings it back to where it all started, and scrambling 1, -1, and 6 in latest Marlin, latest bugfix, and BTT Demo for SERIAL_PORT only works with ESP-01 and TFT 35 V2.0, USB Virtual COM never came up and error out in Windows. 

Sorry for the wall of text. It is a long shot, but at this point I'm beginning to suspect everything possible.

1300371, 2024-11-08:
I think I'm facing the same issue with a brand new SKR PRO V1.2. 
I'am unable to switch the board in DFU Mode :

````
[   73.378996] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[   73.465795] usb 1-1.3: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[   73.465834] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   73.465853] usb 1-1.3: Product: BLACK_F407VE CDC in FS Mode
[   73.465870] usb 1-1.3: Manufacturer: STMicroelectronics
[   73.465885] usb 1-1.3: SerialNumber: 205839853950
[   73.507782] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[   73.507873] usbcore: registered new interface driver cdc_acm
[   73.507881] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

# Putting jumper to BOOT0 and push reset button

[  110.587343] usb 1-1.3: USB disconnect, device number 4
[  112.818473] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  113.109970] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[  113.954292] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  113.961401] usb 1-1-port3: attempt power cycle
[  115.114205] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  115.962191] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  115.969244] usb 1-1-port3: unable to enumerate USB device
````

I'm trying to upload Onstep firmware for a telescope gotomount project with `dfu-util`

zbrozek, 2024-11-08:
You're definitely suffering the same problem. SKR Pro 1.2 did not change these capacitor values from SKR Pro 1.1. C62 and C67 are 22 pF and ought to be 12 pF. You can fix the issue by replacing those two capacitors.

I do not believe that BTT monitors Github issues or makes product changes based on findings within them.

1300371, 2024-11-08:
I'm not a specialist but if i understood well: 
I change these capacitor and i will be able to see USB device of type `STM32 Bootloader` ?

zbrozek, 2024-11-08:
That's correct!

1300371, 2024-11-08:
Will give a try : it's very very tiny 😨 🤞

1300371, 2024-11-17:
Well, I changed the capacitors : still no DFU mode 😭

zbrozek, 2024-11-17:
Looks like you ordered smaller capacitors than what comes on the board, making the swap slightly more challenging. Did you set the BOOT0 jumper to get it to start in DFU mode?

1300371, 2024-11-17:
> Looks like you ordered smaller capacitors than what comes on the board, making the swap slightly more challenging. Did you set the BOOT0 jumper to get it to start in DFU mode?

Yes

zbrozek, 2024-11-22:
Try using an oscilloscope to see how long it takes the oscillator to start up.

1300371, 2024-11-22:
😞 Don't have one. 
As I mentioned, i'm not experimented that much ... 
Have you an advice ? I know some mini oscillope exists  ( 40 € on Aliexpress) : is it valuable ?