bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #154: Ymax endstop is not working

zacomposites, 2020-07-04:
hi guys. 

i am struggling to make y max endstop works. reason its ymax and not y min is because the y homing direction not right so i change it to 1, ive inverted everything on ymax endstop still when i check m119, nothing is triggered.

oldman4U, 2020-07-04:
Hi. Which version of the firmware are you using?

zacomposites, 2020-07-04:
Marlin bugfix 2.0.

oldman4U, 2020-07-04:
And do you have physical endstops or are you using sensorless homing?

You may want to check this: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/103#issuecomment-599070627

zacomposites, 2020-07-04:
> You may want to check this: #103 (comment)

I have physical endstop. Using 2208 driver uart. The board doesnt have +y
endstop like mks board etc. Does it mean i can only use min and not max?

oldman4U, 2020-07-05:
Hi

I use senseless homing on min and software on max. Maybe you can change an unneeded endstop to be used as max, but i do not know. Have you tried to change the y endstop pins like described in the link i sent to you?

zacomposites, 2020-07-05:
It Works!!! :) swap the Y min max value. Thanks !! 1 issue sorted. now my z axis doesnt want to move. all limit switch working fine. any idea?

closed 2020-07-05 by zacomposites

oldman4U, 2020-07-05:
I learned from the best. Gurmeet.

But the z axis moved before? Also 2208?

oldman4U, 2020-07-05:
Just saw you closed the ticket. Seems all good. ;-)))

Thank you and happy printing

zacomposites, 2020-07-05:
Hm.. before i compiled with cheetah from kay3d using similar build size
printer. Z axis does work.  Problem is that it uses 2 z motor..

My printer is the cr3040. It does have dual z ball screw but really its
just single motor with belt attached to both.

So..yeah.  now it doesn't want to work

oldman4U, 2020-07-05:
So, please reopen your ticket or start a new one. This ticket is closed...

The SKR Pro v1.1 can be setup in Marlin using a single Z and you can simply remove the jumpers from the second z Axis and connect a motor there. It will move both motors automatically.

AnHardt, 2020-07-06:
```cpp
//
// Limit Switches
//
#define X_MIN_PIN                           PB10
#define X_MAX_PIN                           PE15
#define Y_MIN_PIN                           PE12
#define Y_MAX_PIN                           PE10
#define Z_MIN_PIN                           PG8
#define Z_MAX_PIN                           PG5
```
When using 'endstop interrupts' `Y_MAX_PIN` (PE**10**) conflicts with `X_MIN_PIN` (PB**10**). The STM32 hardware allowes only one interrupt per end-**number**. All registers (A to J) share the same interrupt vector.

Either disable 'ENDSTOPINTERRUPTS' or swap Y_MAX_PIN with an other and connect the switch there.