bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #155: Z homing not in center of the bed and Z motor not moving.

zacomposites, 2020-07-06:
Hi GUys, my homing direction and endstop are working fine now for X and Y. now to sort out Z axis. after X and Y homing done, the hotend goes to opposite direction all the way to other end and crush. does not goes to center of the bed and start probing(bltouch v3.1) 

heres my setting for z safe homing

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT Y_CENTER  // Y point for Z homing when homing all axes (G28).
#endif

oldman4U, 2020-07-06:
Hi.

Does Z work properly without Z_SAFE_HOMING?

What is the size of your printing bed and what have you defined in config file?

zacomposites, 2020-07-06:
i havent tried to remove z safe homing

/ Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

// @section machine

// The size of the print bed
#define X_BED_SIZE 310
#define Y_BED_SIZE 310

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 440

/**
 * Software Endstops
 *
 * - Prevent moves outside the set machine bounds.
 * - Individual axes can be disabled, if desired.
 * - X and Y only apply to Cartesian robots.
 * - Use 'M211' to set software endstops on/off or report current state
 */

oldman4U, 2020-07-06:
Besides the Z_MAX_POS, we use identical settings. I even have #define Z_SAFE_HOMING also defined.

But i use a Marlin from April at the moment, and there the settings are a bit different:

oldman4U, 2020-07-06:
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

zacomposites, 2020-07-06:
Hi Oldman4U

Thanks for this . just tried now, hm.... no change, x and y perfect, but z is not in the center. hm.............. any other suggestion?

oldman4U, 2020-07-06:
I am slowly running out of ideas. You could try v2.0.5.3 like i use it.

zacomposites, 2020-07-06:
ok. will do now. :)

oldman4U, 2020-07-11:
Hi. Any news regarding the reported issue?

oldman4U, 2020-08-03:
Hi @zacomposites.

What is the status of the reported issue?

Please do not forget to close this ticket, in case ou do to need it anymore.

Thank you

zacomposites, 2020-08-03:
Ok will do. I changed and use the skr 1.4 turbo board

oldman4U, 2020-08-04:
And the new board already works properly?

zacomposites, 2020-08-04:
I use it on my cr20pro. Not on my cr3040, but uses latest marlin not bugfix
nor cheetah.

I notice one of the driver tmc 2208 uaet is not working. Out of 5 that i
received. So i swap it. And the the motor move..

So today am getting new driver from biqu. I will use the same. Hopefully it
will work. Last time i was trying with marlin bugfix and cheetah.

I will try with new marlin.

oldman4U, 2020-08-04:
Seems you have no luck with hardware!😜

closed 2020-08-05 by zacomposites