bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #157: SKR PROv1.2

Roman-ru, 2020-07-25:
Hello, how to connect UART FYSETC 2209v3.0 to SKR PROv1.2

oldman4U, 2020-07-25:
Hi. Could you please check if this helps. https://github.com/GadgetAngel/SKR-PRO-V1.1-Stepper-Driver-Jumper-Configuration-Manual

Roman-ru, 2020-07-26:
No, it doesn't help, there's no Fysetc 2209v3.0

oldman4U, 2020-07-26:
Ok, sorry. In this case i can not help because i use btt tmc2209 with this
board.

I would recommend to use the bugfix Marlin from their github page and start
with the basic functions and add advanced features later on.

Hope someone uses the same stepper driver and can help.

Roman-ru, 2020-07-26:
Thank you anyway.

oldman4U, 2020-07-26:
Make sure to desolder the diag pin in case you are not using sensorless homing  and set the jumpers on the mainboard for UART. Then you can set all other functions in software. Maybe you follow the BTT 2209 guideline from the document above and make sure the FYSTETC is setup for UART too.

Roman121275, 2020-07-26:
Not working, no connection. Sending M122 connection error.

oldman4U, 2020-07-27:
It is ANOTHER MAINBOARD but describes you driver:

https://3dwork.io/en/complete-guide-skr-v1-4-and-tmc2209

Roman121275, 2020-07-27:
Yes, I saw it, but I can't figure out where to put the jumper on my board

oldman4U, 2020-08-03:
Hi.

Any news regarding the reported issue?

Roman-ru, 2020-08-04:
Hello, there is nothing new, I ordered a Bigtreetech 2209 and will receive
it today.

Roman121275, 2020-08-04:
Hello, there is nothing new, I ordered a Bigtreetech 2209 and will receive it today.
--
Отправлено из Mail.ru для Android вторник,

oldman4U, 2020-08-04:
This is great. Let me know how it works

Roman-ru, 2020-08-10:
Work without problems

closed 2020-08-10 by Roman-ru

oldman4U, 2020-08-10:
👍🏻

markosko, 2020-09-15:
[image]
In case someone have same problem with combination of SKR Pro and Fysetc TMC2209 v3.0, no need for soldering or cable wiring, uart jumper as normal, then another jumper on board between RX and TX lines, and under board jumper in middle row on left side as shown in image, M122 shows every connection ok.

oldman4U, 2020-09-15:
👍🏻

@gadgetangel is this something for you?

Roman-ru, 2020-09-15:
Yes,thank you!