bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #158: Can you hook up neopixels to this board?

NovaViper, 2020-07-29:
Hi, Im wondering if you can attach 60 neopixels to the board, I have them being powered externally as shown in the photo
[image]

oldman4U, 2020-08-03:
Hi.

There is an old three which covers your question here: https://github.com/MarlinFirmware/Marlin/issues/15355

I hope this helps

Please let me know

oldman4U, 2020-08-13:
Hi.

Any news? Have you been able to connect the Neopixel?

Please let us know

NovaViper, 2020-08-14:
> Hi.
> 
> Any news? Have you been able to connect the Neopixel?
> 
> Please let us know

Hey sorry for the delay! I haven't had a chance to hook up the board because I've been busy trying to get ready for college

oldman4U, 2020-08-14:
Ok. So what are the next steps?

oldman4U, 2020-09-30:
Please help the community and close the ticket once you do not need it anymore. Thank you

NovaViper, 2020-10-01:
I finally got around to getting the neopixels hooked up to the board last night, but to my dismay, it seems the pixels require some sort of level shifter to get them to work properly (setting different colors seem to cause this weird flickering where the LEDs flash different lights periodically despite not telling it to do so). But it does seem like getting a neopixel (or any sort of LED) is possible with this board, however since there isn't a 3.3V signal anywhere, a level shifter is needed. Im currently working on getting a level shifter and im gonna close the issue.

closed 2020-10-01 by NovaViper