bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #160: USB Device Descriptor Request Failed

mqt0029, 2020-08-09:
Board: SKR Pro v1.1 (not sure if it's any different from v1.2?)

Firmware: Marlin release 2.0.6, Marlin master branch, Marlin bugfix, BTT provided Marlin on this repo

The problem: USB detected the device but throw "USB Device Descriptor Request Failed" out of the box. TFT35 V2.0 still connects, M503 from TFT (and ESP-01 web UI still works when using combination 1 and 6 for serial ports), but no USB to OctoPi or Pronterface. 

What I tried:
- Installed Virtual COM driver provided from this repo, no change
- Try different bootloader: **#25** and **#72** (**WORKS** - out-of-box state, USB error), **#115** (**DOESN'T WORK** - can't flash, file's all F).
- Scramble SERIAL_PORT configurations between -1, 1, and 6, using single or both #define statement:
``` 
#define SERIAL_PORT -1    // or 1, or 6 while #define SERIAL_PORT_2 is commented out 
```
```
#define SERIAL_PORT -1    // or any unique combination between -1, 1, and 6
#define SERIAL_PORT_2 1
``` 
- Disconnect everything, remove all jumpers except microSD, power (tried both Vin and USB), and USB/HOST jumpers (in #48 someone suggested to rotate them jumpers 90<sup>o</sup> but I'm not ready to fry another one of these from random wiring), same result

Tools I used: FTDI breakout set to 3.3V, and an ST-Link V2

Any suggestion about where to go from here? I read a few more issues of people having close but not really the same problem, so I'm not sure if I should reference to those as well.

oldman4U, 2020-08-09:
Hi. Marlin currently supports  a maximum of two serial ports but you try to connect three serial devices. You can have usb OR WIFI besides the TFT but not all three. 

Hope this helps

mqt0029, 2020-08-09:
Please read the description again, I only enabled TFT or WiFi to proof that the firmware wasn't at fault. To clarify, I always define one of the serial port for USB serial, which I believe to be -1 on SKR Pro, which means at all time, one of these two lines must be -1.
[image]

oldman4U, 2020-08-09:
Yes. -1 has to be activated for usb connection. 

Could you attach your configuration files here so i can test with my board

mqt0029, 2020-08-09:
Here's the config. I'm running TMC (2130s on XYZ and 2209 on E) so I enabled TMC but I doubt TMC section has anything to do with the USB Serial.

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5048550/Configuration.zip)

oldman4U, 2020-08-10:
Hi.

It is already very late here and I wanted to give you a quick answer.

I tested the USB connection on my printer and it did not work . which surprised me, because I did most of the configuration like PID tuning using Printrun and USB connection.

So I took a brand new SKR Mainboard and connected it out of the box with the USB cable which came with the mainboard, started Printrun, clicked connect and voila - it worked.

Tomorrow I will try to install the firmware on my "production machine" which has stepper motors and thermistors and a 24V power supply attached.

Attached is the OLD firmware which came with the mainboard. May you wanna give it a try even it is v2 bug fix.

[FIRMWARE.bin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5053647/FIRMWARE.bin.zip)

mqt0029, 2020-08-11:
Thanks for helping and the information. I'll run it through and see if USB pops up. Could you also provide the configs files, since I'm running multiple TMC drivers?

oldman4U, 2020-08-11:
Like I wrote, this is the firmware from the SD card which came with a board which I bought a few months ago (spare). Unfortunately there are no binaries, but in case this also works for you you can expect that the problem is really in software and that your hardware is fine in general.

oldman4U, 2020-08-11:
Please find attached the config files of Marlin bug fix v2 from today 11.8.2020

I changed in platformio.ini:
default_envs = BIGTREE_SKR_PRO

In Configuration.h
Serial Port -1 
and 1
#define BAUDRATE 115200
[Archive.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5057779/Archive.zip)

as well as
#define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_1

I can connect via USB using Printrun.

Hope this helps

oldman4U, 2020-08-13:
Hi.

Any news?

mqt0029, 2020-08-13:
Apologies for the delay I am currently away from the setup. I will update as soon as I get back.

oldman4U, 2020-08-19:
I would disconnect everything except the usb cable and try the configuration like i wrote above. This works withy my two mainboards.

mqt0029, 2020-08-19:
Already did, the board was empty (no display, no ESP, no stepper drivers) with all it's original jumpers in place. Nothing. Probably time to move on. Thanks for the help.

closed 2020-08-19 by mqt0029