bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #162: Extruder only turning one direction

22norbi, 2020-08-18:
Hi, I have been trying to use the BTT SKR Pro v1.1 for soem time now however was am not able to get the Extruder to work. I have tried the 2208 and now the DRV8825 and the motor turns in one direction for the first second or two then it turns into the opposite direction automatically for the rest of the extrusion. 

Here are my two configuration files that I am using:
[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5091191/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5091193/Configuration_adv.h.txt)

Here is a video to illustrate the problem
[ExtruderError.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5091224/ExtruderError.zip)

Any assistance to get this resolved will be appreciated as I have been searching the net without any success of finding a solution. I have also replaced the cable and motor on the extruder and tested different stepper drivers. They all provide the same results, maybe it is something in my configuration that is incorrect.

oldman4U, 2020-08-24:
Hi.

What happens, when you do a manual UNextrude or extrude? Does it move in different directions initially?

nk-111, 2020-08-24:
Hi, yes both manual and printing the extruder first second it moved correctly then it automatically reverses and "UNextrude" every-time

oldman4U, 2020-08-24:
Have you tried to remove everything except what is absolutely needed and
tried to extrude using pronterface?

Has this working before?

oldman4U, 2020-08-24:
Remove everything and try to extrude using pronterface. 

Has this been working before?

nk-111, 2020-09-01:
Hi, I had to setup everything, thank you for the suggestion, however also with pronterface I had the same challenge. I think I have something wrong in my configuration settings, not sure.

oldman4U, 2020-09-02:
Finally this is most likely an hardware issue. If it would run wrong in a
single direction it would be easy to fix. But in your case it starts right
and changes the direction after that. I am not aware of a setting which in
the firmware which could cause this.

🥴

oldman4U, 2020-09-02:
You could try to turn around the extruder cable on one side 180 degrees.

oldman4U, 2020-09-07:
Hi.

Any news?

oldman4U, 2020-09-09:
you tried to deactivate

#define E0_AUTO_FAN_PIN 255

oldman4U, 2020-09-14:
@22norbi 

Could you please reply with the latest news or close this ticket.

Thank you