bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #163: SKR Pro + TMC5160 + PT100 using MAX31865+ BTT TFT35 v2 = no attached printer

code505, 2020-08-24:
I'm having issue working this configuration
`#define SERIAL_PORT -1
#define SERIAL_PORT_2 1`

I'm using this guide to connect MAX31865 in software SPI configuration:
[https://drive.google.com/file/d/1zOky07F_R_DO_-XxZx6C3xULqhuxYMOi/view?usp=sharing](url)

once I change TEMP_SENSOR_0 to  -5, I got the no attached printer, otherwise TFT is working

any thoughts

oldman4U, 2020-08-25:
Please send pictures showing the connection between tft and mainboard. Check the baudrate setting of mainboard and tft.

code505, 2020-08-25:
that is the picture, the reset wire is the last on the left,

[image]

the baudrate is 115200 on both the board and the TFT

what I want to emphasise on, is that when I put,
 TEMP_SENSOR_0 1 >>>> TFT is working with no issue
if
TEMP_SENSOR_0 -5 >>>> TFT show no printer attached

oldman4U, 2020-08-25:
Maybe then it would be good to start a ticket in the general TFT Firmware
repository. Seems TFT related and not SKR.

oldman4U, 2020-08-25:
Saw you opened the ticket, great. Reading your answer again i am not sure if this is SKR or TFT any more, but there are more users watching, so it is always a good idea to start a ticket there. Hope someone can help.

May i ask you why you are using this temp sensor and not a common one?

code505, 2020-08-25:
thank you so much for your cooperation, I installed V6 hotend and I thought it is a good idea to upgrade to pt100 for more consistant heat management and for carbon fiber printing

oldman4U, 2020-08-26:
You are welcome.

I believe there are many parameters in a 3D printer which vary too much in
a very short time to give you accurate results which makes such a precise
measurement unnecessary.

PETG based carbon filament is my favorite material. Best results with a .6
Nozzle, approximately 240 degrees and VERY VERY slow print speed. Usually
60-80mm. Carbon 10mm. But the results are worth. 😁

oldman4U, 2020-08-26:
Forgot. NO Fan!!! Very important.

AnHardt, 2020-08-26:
Back to the problem.

When there is any configuration for the serials where communication is possible - keep that.
When the communication fails when `TEMP_SENSOR_0 -5` but not with `TEMP_SENSOR_0 0` something with the MAX31865-system is wrong. The protection systems detect a MIN_TEMP_ERROR, a MAX_TEMP_ERROR or maybe a communication error and sends Marlin into KILL_STATE - stopping all communication.

Checklist:
How many wires has the PT100/1000 sensor?
Is the MAX31865-module prepared for that number of wires?
If the sensor has 3 wires use only two different colour ones and prepare the module according to that. Marlin does not support 3_WIRE_MODE!
Is the sensor connected to the module?
To witch of the SPI-busses is the module connected to?
Is Marlin configured to use the same buss?
Are the buss-wires not to long, not broken, connected to the right pins? Don't cross MISO and MOSI like you'd do with serials RX and TX.
Check the last tree points also for SS.

Finally correct the `400` to a '430' in `temperature.cpp` at about line 1378 (there is only one occurrence)
```
            #if ENABLED(MAX6675_IS_MAX31865)
              max31865.temperature(100, 400)  // 100 ohms = PT100 resistance. 400 ohms = calibration resistor
            #else
```

If all that does not help try to check the hardware with different software like https://github.com/adafruit/Adafruit_MAX31865/blob/master/examples/max31865/max31865.ino

AnHardt, 2020-08-27:
As far i can see the hardware SPI variants in the above guide can't work.
The software uses the 'default SPI bus' (SPI = SPI1) where the onboard SD-Card is connected to.
The wires shall be connected to the TMC-SPI bus (SPI3).
That compiles but can't work.

GadgetAngel, 2020-08-28:
#

> 

> As far i can see the hardware SPI variants in the above guide can't work.
> The software uses the 'default SPI bus' (SPI = SPI1) where the onboard SD-Card is connected to. The Adafruit library in the currently by Marlin used version is not able to use an other one. The current version of the library can be connected to any SPI buss but uses Adafruit BussIO what is currently not supported by Marlin.
> The wires shall be connected to the TMC-SPI bus (SPI3).
> That compiles but can't work.
> 
> Up to now i couldn't find a deal beaker for the Software SPI variantes. (Even if i did not try it physically.)
> 
> Edit: Sorry. Even the newest Adafruit_MAX31865 library (1.2.2) is not able to use an other than the default SPI hardware. They missed to expand the constructor.

## Please educate me, so I can correct the guide.  (To restate what you have written and to ensure I understand): You state that Marlin 2.0.6 can not work on SPI3.  Marlin 2.0.6 will only use SPI1 for hardware SPI for ANY MCU. Is this correct?

#### As far as I am aware, the Hardware SPI is a master slave communication protocol. The SKR PRO V1.1 has three SPI buses on the MCU. SPI1 is the SD card reader; SPI2 is EXP2 and SPI3 is the Stepper Motor Driver Sockets. 

## Are you also saying that Marlin 2.0.6, does not allow the user to use Hardware SPI because at this time Marlin does not support Adafruit BUSIO? So the ONLY WAY to even use Adafruit’s  MAX31865 is by using Software SPI?

AnHardt, 2020-08-28:
The BTT SKR PROs have 3 SPI-buses. Marlin, in general, is able to use all of them. 
It's Adafruit_MAX31865 library what does not allow to select an other than the default SPI-buss (here SPI = SPI1).
Hardware SPI for the MAX31865 is currently only possible if connected to the onboard-SD-SPI-buss - what results in a mess because there are no connectors.

Even in the newest version of the MAX31865 lib we have no way to specify the "buss".
```
  Adafruit_MAX31865(int8_t spi_cs, int8_t spi_mosi, int8_t spi_miso, int8_t spi_clk);
  Adafruit_MAX31865(int8_t spi_cs);
```
While the underling BussIO now can handle it:
```
  Adafruit_SPIDevice(int8_t cspin, uint32_t freq = 1000000,
                     BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                     uint8_t dataMode = SPI_MODE0, SPIClass *theSPI = &SPI);

  Adafruit_SPIDevice(int8_t cspin, int8_t sck, int8_t miso, int8_t mosi,
                     uint32_t freq = 1000000,
                     BitOrder dataOrder = SPI_BITORDER_MSBFIRST,
                     uint8_t dataMode = SPI_MODE0);
```
 - last parameter of the first constructor.

In general alone from the SS-pin it's impossible to detect the right buss.

GadgetAngel, 2020-08-28:
> The BTT SKR PROs have 3 SPI-buses. Marlin, in general, is able to use all of them.
> It's Adafruit_MAX31865 library what does not allow to select an other than the default SPI-buss (here SPI = SPI1). [...]

#### Is it not possible to write a software wrapper that exposes the pointer to the SPI BUS from the  underlying BUSIO Function call?  Is that not what a Constructor does in C++? Marlin is written in which programming language?

## Which MAX31865 amplifier board on the market today can a user of Marlin use to access the correct SPI bus? Is there even one available?

GadgetAngel, 2020-08-28:
> I'm having issue working this configuration
> `#define SERIAL_PORT -1 #define SERIAL_PORT_2 1`
>
> I'm using this guide to connect MAX31865 in software SPI configuration: [...]

### Please place your Marlin folder (Zip the folder up) onto google drive and supply us with the shared link to that File. 

### It makes no sense that Software SPI is not working for you. 

### I will update the Guide to reflect the information about the current state of the Adafruit MAX31865 library

code505, 2020-08-28:
Here is the compressed folder, I really appreciate your cooperation guys.

[https://drive.google.com/file/d/1K1XbNh0lFrJSDPzNua3jOYGHsJ_Dz9JK/view?usp=sharing](url)

GadgetAngel, 2020-08-28:
### @code505 I downloaded the latest Marlin version 2.0.6.1 and made the changes that I have on my printer.  Do you have a filament runout sensor attached on your printer?   Are you using the filament runout port on the TFT screen?   

### If you have a filament runout sensor attached please comment out the Marlin sections related to that equipment.  Leave them commented out until we get the MAX31865 in Software SPI mode to work.

### Are you running V2.0.26 firmware on your BTT TFT35 V2.0 screen?

## Here are the differences (that I think are relevant) I found between my setup and yours:

## Differences in Platformio.ini file (your file is on the LEFT, my setup is on the RIGHT)--- I think this is where your problem is:
[image]

# *** In my opinion: In Platformio.ini file: Change your default_evns to be BIGTREE_SKR_PRO and change include_dir to be Marlin

## Difference in Configuration.h file (you have two screens enabled in Marlin):
### You have Your Stock CR10 Displayed enabled yet the above picture shows no connections to EXP1 or EXP2 connectors:
[image]
### You have the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled in Marlin along with your CR10 stock display:
[image]

# *** In my opinion: In Configuration.h file, you need to just disable both the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and the CR10_STOCKDISPLAY **_AND_** ensure the Marlin line #define SDSUPPORT  is uncommented.

### Here is a video to watch to connect the BTT TFT35 V2.0 to the SKR PRO V1.1 board: https://youtu.be/9i3Ee6UtC-Q

### In the above video He does not mention enabling any LCD screen in the Marlin firmware download.  All he enables in Marlin Firmware are the following: Stepper Drivers used and enable are A4988 drivers;  SERIAL_PORT is -1;  SERIAL_PORT_2 is 1;.  MOTHERBOARD is BOARD_BTT_SKR_PRO_V1_1     ; In Platformio.ini file: default_envs = BIGTREE_SKR_PRO.  He sets his baudrate on the TFT35 V2.0 screen to 250000

GadgetAngel, 2020-08-29:
## @AnHardt Could I use this device and hack into the SPI Lines for the SKR PRO V1.1 and GTR V1.0 boards? Maybe that would work?
https://www.amazon.com/gp/product/B088WB5L8R/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

[image]

AnHardt, 2020-08-29:
Yes - one could.
But it also has no pins to connect the SPI-bus to. So it's has no advantage over soldering wires/pins directly to the mainboard at the SD-Card-slot or the processor. (Maybe the pins at the ESP are a bit larger?)

GadgetAngel, 2020-08-29:
@AnHardt, Thanks for setting me straight on the Adafruit library issue!  I am now fixing the guide so that I will not be giving out bad information.  Just to be clear on another point, the reason my guides provide redundant information is to make it perfectly clear to the user that they have options.  

Some people cannot extrapolate one solution into another solution.  

My goal is to open up this hobby of 3D printing to more users.  The more people that get involved with 3D printing the better the firmware will get and the MCU boards will get upgraded.  

I do not intend for my manuals and guides to be idiot proof.  But I do want to make it easier for people to hook up the devices.  If they have a sort of cook book to follow, they are given a starting point and then learn from there.  Sometimes that is all that is needed to get people to try out a new item.  

I never intended my guides and manuals to be insulting.  In fact, if anyone ever finds wrong information in my guide or manuals PLEASE let me know.  I need to learn and then I can fix the guide.  Just like you did with the PT100 Guide.  I really appreciate the input you gave.  Thanks again.

code505, 2020-08-29:
@GadgetAngel 
Hi,
I also downloaded Marlin 2.0.6.1, and edited all the configuration including Platformio.ini file, 
I have filament run out sensor and I commented out it's setting 
installed V2.0.26 firmware on TFT35
I had the cr10s display and I thought I would give it a try without the TFT, but that also didn't work, that why I set the setting for the original cr10s display, but I revert every thing back now
but until now, no luck
one thing to mention, when I set `#define SERIAL_PORT 1` and `#define SERIAL_PORT_2 -1`

the TFT give these messages:
[image]
[image]

code505, 2020-08-30:
I have  PT100 amplifier board (INA826 amp) , and I will connect it using table 21 

to which pin should I use for 3.3v supply, ground and signal.

thank you

GadgetAngel, 2020-08-31:
@code505 The easiest way is to use any of the ADC lines on extension-1 heade, get 3.3VDC and GND from power expansion header.

Extension-1 header:
[image]

Color PIN diagram of SKR PRO V1.1 Board:
[image]

code505, 2020-08-31:
can I use 3.3v and GND from E1 endstop socket?, I have only one extruder

AnHardt, 2020-08-31:
On this board VREF+ is not different from VCC_3.3V and VDDA is not accessible - so yes for the 3.3V.
On this board we have more or less separate systems for GND and GNDA - so piking GNDA from a unused T input could make more sense to avoid some noise (but for functionality that should not make a difference - in the end they are connected via 0 Ohm R31).

[image]

GadgetAngel, 2020-09-06:
> can I use 3.3v and GND from E1 endstop socket?, I have only one extruder

yes

GadgetAngel, 2020-09-06:
### @AnHardt You wrote this: 

> Finally correct the 400 to a 430 in temperature.cpp at about line 1378 (there is only one occurrence)
> 
>             #if ENABLED(MAX6675_IS_MAX31865)
>               max31865.temperature(100, 400)  // 100 ohms = PT100 resistance. 400 ohms = calibration resistor
>             #else
> 

## 400 Ohm is the calibration resistor.   So for the Adafruit MAX31865 for PT100 you must change the calibration resistor to 430 in temperature.cpp. If you bought Adafruit MAX31865 for PT1000 you must use max31865.temperature(1000, 4300) in temperature.cpp.  Is this correct?

AnHardt, 2020-09-06:
# **Yes!**

Do you really have to be that loud - always? (big fonts) My eyes are bleeding. :-)

GadgetAngel, 2020-09-09:
> Yes - one could.
> But it also has no pins to connect the SPI-bus to. So it's has no advantage over soldering wires/pins directly to the mainboard at the SD-Card-slot or the processor. (Maybe the pins at the ESP are a bit larger? Ahhh - and if ruined this is a bit cheaper.)
> 
> Something like this would be even cheaper.
> [image]

## @AnHardt Can you find any document showing the 15P cable and which PINS correspond to MOSI, MISO, SCK ?   I have looked and I can not find any document showing which PINS on the extension cable correspond to the SPI lines

AnHardt, 2020-09-09:
One side of the cable is a sd-card with some wires.
The other a sd-card-socket with some wires. 
To make that work the wiring has to be straight, one by one, no crossings.
The cable can also be used to extend SDIO connected sockets - not only SPI.

https://www.google.com/search?sxsrf=ALeKk02d7DxB8_5DES9ppG2VqZWjRveTVw:1599686685325&source=univ&tbm=isch&q=sd+card+spi+interface&sa=X&ved=2ahUKEwjmwoWlgd3rAhXGlqQKHQs-AGAQsAR6BAgMEAE&biw=1280&bih=577

GadgetAngel, 2020-09-11:
> One side of the cable is a sd-card with some wires.
> The other a sd-card-socket with some wires.
> To make that work the wiring has to be straight, one by one, no crossings.
> The cable can also be used to extend SDIO connected sockets - not only SPI.
> 
> https://www.google.com/search?sxsrf=ALeKk02d7DxB8_5DES9ppG2VqZWjRveTVw:1599686685325&source=univ&tbm=isch&q=sd+card+spi+interface&sa=X&ved=2ahUKEwjmwoWlgd3rAhXGlqQKHQs-AGAQsAR6BAgMEAE&biw=1280&bih=577
> 
> Soldering to the flat-cable is difficult. Better crack the housing of the sd-socket and solder to the contacts.

This issue is the SD card socket has 15 pins on the back when I crack it open.  The sd card on the other end has 8 pins.  I agree they should be straight connect but why have 15 pins to 8?  If every other pin was Ground then I would expect 16 pins to 8.  I looked up the part number on the SD card socket: AoSheng100921TF .  It is a patented product from China.  I can not find how they go from 15 pins to 8 pins.  Do you have any idea what they did?

AnHardt, 2020-09-12:
8 pins have 7 gaps. 8+7 = 15. Yes, likely there are GND in between - even if having 3 GND lines together does not make sense - maybe additional GNDs are at the outer lanes. Measuring helps to know exactly.

code505, 2020-09-12:
@GadgetAngel Hi there,

I've used your guide to connect pt100 with amplifier board:

[image]

I changed TEMP_SENSOR_0 to 21 and TEMP_0_PIN to PF7

now hotend temp is reading 1, not changing
when I change  TEMP_SENSOR_0 to 20
it works but temperature is higher by 3-4 degrees, and it is not stable, some time shooting 10-15 degrees up/down and come back again, this happen within second 

using MAX31865, still no luck, :(

GadgetAngel, 2020-09-12:
> @GadgetAngel Hi there,
> 
> I've used your guide to connect pt100 with amplifier board:
> 
> [image]
> 
> I changed TEMP_SENSOR_0 to 21 and TEMP_0_PIN to PF8
> 
> now hotend temp is reading 1, not changing
> when I change TEMP_SENSOR_0 to 20
> it works but temperature is higher by 3-4 degrees, and it is not stable, some time shooting 10-15 degrees up/down and come back again, this happen within second
> 
> using MAX31865, still no luck, :(

Which Amplifier board are you using? the E3D PT100 Amplifier board or the MAX31865 board?  If you are using the E3D PT100 Amplifier board and connected up as shown in the diagram, then TEMP_SENSOR_0 set to 20 will give you the wrong readings because that conversion table assumes the excitation voltage is set for 5V.  You are using a 3.3VDC supply voltage.   Try creating your own translation table.   Keep your Technique #2&Method#1 wiring but try Technique #8 to generate your own translation table (create a custom thermistor table).  

Double check your wiring to and from the PT100 sensor and the PT100 amplifier board are good. 

 Check your PT100 sensor: to see if it changes resistance with temperature.  Use your multimeter and measure the resistance of the PT100 sensor at room temperature.  Now place the PT100 Sensor next to your soldering iron so it can read temperature above room temperature.  See if the PT100 sensor resistance changes.  

It sounds like you have more than one problem you are trying to overcome.  It sounds like you are getting data back to the MCU, now you just need to figure out why it is stuck at 1.  

You can make Marlin show the ADC values it is getting from your Amplifier board by turn on the Marlin variable "SHOW_TEMP_ADC_VALUES".  Please watch this video to understand how to create your own thermistor table (translation table) from scratch: https://www.youtube.com/watch?v=brl1weuD8bs&feature=youtu.be.  The YouTube video is called "Marlin Thermistor Calibration" but it walks you through understanding how Marlin uses the thermistor table.

code505, 2020-09-12:
after doing a search, I think I'm not alone,
https://github.com/MarlinFirmware/Marlin/pull/19246

by the way, I used your guide also to connect smart filament sensor, is it okay to use 5v supply to provide power to the sensor, I think it can be operated ok from 3.3v from the limit switch E0

[image]

GadgetAngel, 2020-09-13:
> after doing a search, I think I'm not alone,
> [MarlinFirmware/Marlin#19246](https://github.com/MarlinFirmware/Marlin/pull/19246)
> &
> [MarlinFirmware/Marlin#16744](https://github.com/MarlinFirmware/Marlin/issues/16744)
> 
> by the way, I used your guide also to connect smart filament sensor, is it okay to use 5v supply to provide power to the sensor, I think it can be operated ok from 3.3v from the limit switch E0
> 
> [image]

The BIGTREETECH SFS requires 5V supply but try the 3.3VDC to see if it works,  I have not tried it with 3.3VDC.  If it works, it works

AnHardt, 2020-09-13:
Please update to the complete facts.
Use 20 if module supply is the same for module and ADC-reference. For the SKR PRO, if the module allows, 3.3V / 3.3V. For a 5V processor board, 5V / 5V.
Use 21 if the module gets 5V but the processor 3.3V. Don't use this above ~500°C.

It would make sense to wait until https://github.com/MarlinFirmware/Marlin/pull/19246 or a replacement/followup is merged. Then slightly rephrase, to avoid complete redundancy. A manual should provide a second chance to understand what is going on.

GadgetAngel, 2020-09-14:
> Please update to the complete facts.
> Use 20 if module supply is the same for module and ADC-reference. For the SKR PRO, if the module allows, 3.3V / 3.3V. For a 5V processor board, 5V / 5V.
> Use 21 if the module gets 5V but the processor 3.3V. Don't use this above ~500°C.
> 
> It would make sense to wait until [MarlinFirmware/Marlin#19246](https://github.com/MarlinFirmware/Marlin/pull/19246) or a replacement/followup is merged. Then slightly rephrase, to avoid complete redundancy. A manual should provide a second chance to understand what is going on. For example avoid "excitation".

### I agree, I will state the following:
### Use table 20: When the power supply for the PT100 Amplifier board is equal to the ADC reference voltage for the MCU.  

### For SKR PRO V1.1/V1.2, SKR V1.3/V1.4/V1.4 Turbo, GTR V1.0, M5 V1.0 all use 3.3 VDC as the ADC reference voltage for the ### MCU.
### Therefore use table 20 when the PT100 Amplifier board is powered by 3.3VDC

### Use table 21: When the power supply for the PT100 Amplifier board is different to the ADC reference voltage for the MCU.  

### The E3D PT100 Amplifier board wants 5V as its power supply. But the SKR PRO V1.1 uses 3.3V as the ADC reference voltage  for the MCU.  My current manual recommends using 3.3VDC for powering the E3D PT100 Amplifier board.  But now that you all have made me aware of this issue with the translation table I would recommend using 5V for powering the E3D PT100 Amplifier board.   In this case, I would also recommend using a 3.6V Zener Diode (reverse biased hookup) to protect the MCU from 5V input.  If the PT100 sensor gets damaged or becomes disconnected from the PT100 amplifier board while the PT100 amplifier board is powered on, a 5V output signal will occur and fry the MCU board.  This type of problem may or maynot happen but I would prefer to protect the MCU board. Therefore, I will update my wiring diagrams so that they show the wiring for the Zener Diode.  Again, Thank you both for making me aware of this issue

code505, 2020-09-14:
Using MAX31865, Is there any changes have to be done in temperature.cpp?? other than the pin assignemtn in pins_BTT_SKR_PRO_common.h

#ifndef MAX31865_CS_PIN
#define MAX31865_CS_PIN TEMP_0_PIN
#define MAX6675_SS_PIN MAX31865_CS_PIN
#define MAX31865_MOSI_PIN PD2
#define MAX31865_MISO_PIN PD5
#define MAX31865_SCK_PIN PE0
#endif

I found this [https://reprap.org/forum/read.php?415,809669](url)
is this applicable for the current Marlin release?

AnHardt, 2020-09-15:
No.
Meanwhile MAX31865 is integrated into Marlin.
That message is from a wile before that. So likely the appended `temperature.cpp` is to old to not cause problems.

code505, 2020-09-15:
I'm still trying to make it work with MAX31865, when compiling I got this:

> Compiling .pio/build/BIGTREE_SKR_PRO/src/src/sd/SdFile.cpp.o
> Compiling .pio/build/BIGTREE_SKR_PRO/src/src/sd/SdVolume.cpp.o
> _**In file included from Marlin/src/module/temperature.cpp:48:
> .pio/libdeps/BIGTREE_SKR_PRO/Adafruit MAX31865 library/Adafruit_MAX31865.h:84:32: warning: passing NULL to non-pointer argument 1 of 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass*)' [-Wconversion-null]
>    84 |   Adafruit_SPIDevice spi_dev = NULL;
>       |                                ^~~~
> In file included from .pio/libdeps/BIGTREE_SKR_PRO/Adafruit MAX31865 library/Adafruit_MAX31865.h:55,
>                  from Marlin/src/module/temperature.cpp:48:
> .pio/libdeps/BIGTREE_SKR_PRO/Adafruit BusIO/Adafruit_SPIDevice.h:60:29: note:   declared here
>    60 |   Adafruit_SPIDevice(int8_t cspin, uint32_t freq = 1000000,
>       |                      ~~~~~~~^~~~~**_

AnHardt, 2020-09-16:
That's a problem in `Adafruit MAX31865 library`.

Use a recent Marlin and compile with PlatformIO. The newer versions of `Adafruit MAX31865 library` are purposely excluded (MAX6675_IS_MAX31865     = Adafruit MAX31865 library@~1.1.0).
Alternatively downgrade the library via Arduinos library manager to `1.1.0`.

GadgetAngel, 2020-09-17:
> I'm still trying to make it work with MAX31865, when compiling it compile successfully, but I got this:
> 
> > Compiling .pio/build/BIGTREE_SKR_PRO/src/src/sd/SdFile.cpp.o
> > Compiling .pio/build/BIGTREE_SKR_PRO/src/src/sd/SdVolume.cpp.o
> > ___In file included from Marlin/src/module/temperature.cpp:48:
> > .pio/libdeps/BIGTREE_SKR_PRO/Adafruit MAX31865 library/Adafruit_MAX31865.h:84:32: warning: passing NULL to non-pointer argument 1 of 'Adafruit_SPIDevice::Adafruit_SPIDevice(int8_t, uint32_t, BitOrder, uint8_t, SPIClass_)' [-Wconversion-null]
> > 84 |   Adafruit_SPIDevice spi_dev = NULL;
> > |                                ^~~~
> > In file included from .pio/libdeps/BIGTREE_SKR_PRO/Adafruit MAX31865 library/Adafruit_MAX31865.h:55,
> > from Marlin/src/module/temperature.cpp:48:
> > .pio/libdeps/BIGTREE_SKR_PRO/Adafruit BusIO/Adafruit_SPIDevice.h:60:29: note:   declared here
> > 60 |   Adafruit_SPIDevice(int8_t cspin, uint32_t freq = 1000000,
> > |                      ~~~~~~~^~~~~_*_

## @code505 The only changes I am aware of at this time to temperature.cpp in the latest release of Marlin 2.0.6.1 are as follows:
 
[Changes to max31865.temperature.cpp in Marlin for MAX31865 board from PRO V1.1_ Version 6.1.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5239624/Changes.to.max31865.temperature.cpp.in.Marlin.for.MAX31865.board.from.PRO.V1.1_.Version.6.1.pdf) or

[image]

[image]

# Please place your current Marlin folder (Zip the folder up) onto google drive and supply us with the shared link to that File. I would like to look at the info messages you are getting from Marlin myself.  Maybe I can see what is going on in the code.

code505, 2020-09-17:
Thank you so much guys for your input, this is the file:

[https://www.dropbox.com/s/rc9imppk85hk6wd/Marlin.zip?dl=0](url)

this is the board I'm using:
[https://www.aliexpress.com/item/32952386281.html?spm=a2g0s.9042311.0.0.27424c4d7vWIqI](url)

the sensor is working with amplifier board but not accurate and have lots of fluctuation, I've two MAX31865 and both doesn't work so the must Likely cause is setting

GadgetAngel, 2020-09-18:
> Thank you so much guys for your input, this is the file:
>
> [https://www.dropbox.com/s/rc9imppk85hk6wd/Marlin.zip?dl=0](url)
>
> this is the board I'm using: [...]

# I tried to open your Marlin folder under VScode but I could not.  You need to zip your whole Marlin folder with the platformio.ini file included in the zip folder.  I get the following message when I try to open your project under VScode:
[image]

GadgetAngel, 2020-09-18:
> https://www.aliexpress.com/item/32952386281.html?spm=a2g0s.9042311.0.0.27424c4d7vWIqI

# The device you bought is _**NOT**_ an Adafruit MAX31865 board.  

## Notice the Rref is at 431 Ohms.  So change the value from 430 to 431 in temperature.cpp. 
[image]

# I have no idea if this clone board uses the correct SPI timing needed to get it to work with the Adafruit MAX31865 library.  Please buy yourself the Adafruit MAX3165 board. You can find it here:
 https://www.digikey.com/products/en?mpart=3328&v=1528

[image]

# I know it cost $10 more than the clone. But, the Adafruit MAX31865 board is the ONLY board that has been tested with the Adafruit MAX31865 library and Marlin

AnHardt, 2020-09-18:
Nonsens!
https://www.electronics-notes.com/articles/electronic_components/resistors/smt-smd-resistor-codes-markings.php chapter 3.
A SMD resistor marked "431" **has** 430 Ohm

GadgetAngel, 2020-09-18:
> Nonsens!
> https://www.electronics-notes.com/articles/electronic_components/resistors/smt-smd-resistor-codes-markings.php chapter 3.
> A SMD resistor marked "431" **has** 430 Ohm

Thanks for the input

GadgetAngel, 2020-09-18:
# @AnHardt I am confused about something. In Marlin, by default the LCD SD (EXP2) is enabled. Does that mean that the SPI Bus for LCD SD (EXP2) is now the default Hardware SPI BUS in Marlin?

###  To state it another way:  Which ever SPI bus that LCD SD (EXP2) is on,  Marlin will use that SPI Bus as the default Hardware SPI bus?  

### So for SKR PRO board, the LCD SD (EXP2) is on SPI BUS number 2.  So does Marlin automatically defaults to use SPI BUS number 2 as the Hardware SPI bus?  The reason I ask is due to comments I see in the PINS file for the MCU.  For example the comment in the orange square box in pins_BTT_SKR_PRO_common.h is what I am confused about :
[image]

### It states in the comments: Onboard SD card (TF card reader) must use software SPI because Marlin's default hardware SPI is tied to LCD's EXP2 port (LCD SD).

### For SKR PRO board the EXP2 is on SPI bus number 2, while the TF card reader is on SPI bus number 1.  

# So Marlin's default hardware SPI bus is LCD EXP2 on each MCU board? 

## For SKR V1.3 board,  SPI bus 0 is for LCD EXP2.  While SKR PRO use SPI bus 2.  So the default hardware SPI bus for SKR V1.3 is SPI bus 0 and the default hardware SPI bus for SKR PRO is SPI bus 2.  Is am getting this right?

# A while ago I thought I understood that the default hardware SPI bus would be used by the Adafruit MAX31865 board.  Now does that mean the Adafruit MAX31865 library will default to the LCD EXP2 on each MCU or does it mean the Adafruit MAX31865 library will default to use the first SPI bus on the MCU board? 

## For example the first SPI bus on SKR PRO is SPI bus number 1 which is the TF card reader. But Marlin uses the LCD EXP,2 which is on SPI bus 2, as the default hardware SPI bus. 

# So does the Adafruit library use SPI bus 1 or SPI bus 2 on the SKR PRO board?

AnHardt, 2020-09-18:
The SKR PRO has 4 Spi busses. One to the onboard SD-reader that is connected to pins able to be used as SPI1 - but currently used in software SPI because of a pullup at the MiSo-line (or MoSi - don't remember) . One is the SPI buss for the TMC and the SPI/connector - hardware SPI3. Two are on the EXP Connectors - one for the SD-slot on the dispazs board (hardware SPI2) and another software one for the display.

The default SPI is always a hardware SPI. Currently SPI for the SKR PRO indeed seems to be SPI2. :-(

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/variants/BIGTREE_SKR_PRO_1v1/variant.h#L239/L245
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/variants/BIGTREE_SKR_PRO_1v1/variant.h#L69/L72

code505, 2020-09-20:
I'm sorry being late, here is the file:
[https://www.dropbox.com/s/oq3pcovnan059h5/Marlin-bugfix-2.0.x%204.zip?dl=0](url)

have you tried authentic Adafruit board and worked with your settings?
I have two of the clones and it's soldered perfectly, I don't know about the power is enough?? I will try to supply with external power

GadgetAngel, 2020-09-20:
> The SKR PRO has 4 SPI busses. One to the onboard SD-reader that is connected to pins able to be used as SPI1 - but currently used in software SPI because of a pullup at the MiSo-line (or MoSi - don't remember) . One is the SPI buss for the TMC and the SPI-connector - hardware SPI3. Two are on the EXP Connectors - one for the SD-slot on the displays board (hardware SPI2) and another software one for the display.
> 
> The default SPI is always a hardware SPI. Currently SPI for the SKR PRO indeed seems to be SPI2. :-(
> 
> It's changed, a bit hidden :-( , from the default SPI1 to SPI2 here:
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/variants/BIGTREE_SKR_PRO_1v1/variant.h#L239/L245
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/variants/BIGTREE_SKR_PRO_1v1/variant.h#L69/L72

@AnHardt Thank you for the information.  Sorry about the large letters.  I just read your note on that.  Please forgive me. I will stop yelling. 

I believe the pullup is on the CLK line which makes the clock idle at HIGH. Therefore, forcing the hardware SPI to be mode 3. Apparently the issue is that Marlin uses Mode 0 for the default hardware spi bus and that is why they force it to be software SPI. 

Say I have the following situation: I have a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled as an LCD and I try to hook up a MAX31865 board with Hardware SPI.  

Since by default the pins_BTT_SKR_PRO_common.h file sets  #defene SDCARD_CONNECTION LCD, the default hardware SPI bus number is 2 (as you pointed out in your last post, which for SKR PRO V1.1 is the EXP2 connector), meaning the MAX31865 board should be spliced into the EXP2 flat ribbon cable. 

So here is my current question: After reading this post: https://github.com/MarlinFirmware/Marlin/issues/10640#issuecomment-394061808,   it sounds like the the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER does NOT release the SPI bus. So if anyone has the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled in the firmware this issue will cause the MAX31865 board in hardware SPI not to work as expected because the MAX31865 board will never get a chance to be the selected slave on the default hardware spi bus number 2.  So are there any other LCD screens I also need to be aware of that behave in this manner?

@code505 I own three MAX31865 boards,  I have not tried them out on the SKR PRO V1.1 board yet, that is why I am trying to help you figure out why its not working.  While I think about your issue, I am also writing the manual for the GTR V1.0 board which is based on the SKR PRO V1.1 board.  Since you pointed out my error with the Thermistor tables, I am now changing my techniques to use ONLY 5VDC power source for the PT100 amplifier board and the MAX31865 board .  I am updating the wiring diagrams to include a zener diode to protect the input of the SKR PRO V1.1 board or GTR V1.0 board.  Both boards are 3.3VDC logic levels.  

@code505 I suspect that your issue is the clone board, so please buy a real Adafruit MAX31865 board.  It's the next logical step in debugging the MAX31865 board problem.  The software SPI should work.  But, "Shoulda woulda coulda",  Should does not mean it is working.  Please continue on with the next step.

GadgetAngel, 2020-09-21:
> I'm sorry being late, here is the file:
> [https://www.dropbox.com/s/oq3pcovnan059h5/Marlin-bugfix-2.0.x%204.zip?dl=0](url)
>
> have you tried authentic Adafruit board and worked with your settings?
> I [...]

@code505 I think a new download of the platformio.ini file needs to occur.  In the folder I see the following: one the left is your platformio.ini file and on the right is the one I just downloaded an hour ago:

[image]

I would expect only a few lines would be RED but the file has been changed.  

The **UPPERCASE is very important for the Marlin variables stored in platformio.ini file**.  When "MAX6675_IS_MAX31865" is changed to "max6675_is_max3186" the variable is a different name so platformio saw you needed the MAX31865 library and downloaded the latest and greatest version.  Here is what I saw on the max6675_is_max31865 line:

[image]

Please redo the download, change the line to : "**MAX6675_IS_MAX31865 = Adafruit MAX31865 library@1.1.0**".  By default Marlin adds a ~ after the @ sign.  I know it works with the V1.1.0 version so set it to that exact version.

[image]

Here is the way to tell that you are using the right library.  When you compile the code after the success message appears, scroll to the very top of the output listing and look for the library version the compiler used to create the .bin file:

[image]

To uninstall the wrong library versions go to the Platformio HOME page and select "Libraries", then select the Installed tab, then in search field type "Adafruit MAX31865". Select the uninstall button for all library that are not V1.1.0.

[image]

AnHardt, 2020-09-21:
The `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` has two SPI busses.
One for the SD-card and
one for the ST7920 disply driver.
Usually the SD-SPI is driven by a hardware SPI buss. On the SKR PRO SPI2 = SPI.
The SPI bus for the ST7920 is usually a software SPI. 
The ST7920 has a hardware error. The chip itself ignores the SS (CE or what ever you want to name it) line. If any other device is used on the same SPI buss the display will show random pixel garbage. So don't put any other devices on that software SPI buss. [Bobs note](https://github.com/MarlinFirmware/Marlin/issues/10640#issuecomment-394061808) is about the REARM. That has not enough pins, so both of the `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`s SPI busses are shared (end on the same processor pins) what causes the problems.
The SD-SPI is on EXP2. The Display-SPI is on EXP1.
https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf .

----

http://www.hpinfotech.ro/ST7920.pdf page 26 second, paragraph is about the CS line. This is kind of an errata. Previous datasheets told, its possible to use the ST7920 normally.

----

The used by the software SPI pins names are not obvious. In the pins-file they still follow the naming scheme of the 20x4 display `REPRAP_DISCOUNT_SMART_CONTROLLER` for the same positions on the plug.
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h#L30-L32
```cpp
#define ST7920_CLK_PIN  LCD_PINS_D4
#define ST7920_DAT_PIN  LCD_PINS_ENABLE
#define ST7920_CS_PIN   LCD_PINS_RS
```

GadgetAngel, 2020-09-21:
@AnHardt You have been doing 3D printing for awhile.  What do you mean REARM? I google it an nothing comes up that makes sense

> The `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` has two SPI busses.
> One for the SD-card and
> one for the ST7920 disply driver.
> Usually the SD-SPI is driven by a hardware SPI buss. On the SKR PRO SPI2 = SPI.
> The SPI bus for the ST7920 is usually a software SPI.
> The ST7920 has a hardware error. The chip itself ignores the SS (CE or what ever you want to name it) line. If any other device is used on the same SPI buss the display will show random pixel garbage. So don't put any other devices on that software SPI buss. [Bobs note](https://github.com/MarlinFirmware/Marlin/issues/10640#issuecomment-394061808) is about the REARM. That has not enough pins, so both of the `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`s SPI busses are shared (end on the same processor pins) what causes the problems.
> The SD-SPI is on EXP2. The Display-SPI is on EXP1.
> https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf .
> 
> http://www.hpinfotech.ro/ST7920.pdf page 26 second, paragraph is about the CS line. This is kind of an errata. Previous datasheets told, its possible to use the ST7920 normally.
> 
> The used by the software SPI pins names are not obvious. In the pins-file they still follow the naming scheme of the 20x4 display `REPRAP_DISCOUNT_SMART_CONTROLLER` for the same positions on the plug.
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h#L30-L32
> 
> ```c++
> #define ST7920_CLK_PIN  LCD_PINS_D4
> #define ST7920_DAT_PIN  LCD_PINS_ENABLE
> #define ST7920_CS_PIN   LCD_PINS_RS
> ```
> 
> [image]
> 
> Add a third plug to the EXP2 cable.

@AnHardt  I really do appreciate your knowledge and attention to detail.  Thank you so much for sharing this information!

AnHardt, 2020-09-21:
https://www.panucatt.com/Re_ARM_for_RAMPS_p/ra1768.htm
https://github.com/MarlinFirmware/Marlin/issues/10640#issuecomment-388699863

GadgetAngel, 2020-09-21:
@AnHardt I wish they would add comments in each PINS file about which lines on the EXP1 are Software SPI pins.  Currently they are like this:
[image]

May be updating the comments like this may help: 
[image]

GadgetAngel, 2020-09-24:
@code505 I just found something that might be the cause of your MAX31865 board not running properly.  In my manual I told you to do the following in pins_BTT_SKR_PRO_common.h file:
```
#define TEMP_0_PIN  PD0
#ifndef MAX31865_CS_PIN
#define MAX31865_CS_PIN TEMP_0_PIN
#define MAX6675_SS_PIN MAX31865_CS_PIN
#define MAX31865_MOSI_PIN PD2
#define MAX31865_MISO_PIN PD5
#define MAX31865_SCK_PIN PE0
#endif
```
You want the MAX31865 board to use Software SPI to communicate with the board.  After digging some more into the Marlin software I just noticed something:
[image]

In the  (...\Marlin\src\module\) temperature.cpp file as shown above, you can see the statement:
```
Adafruit_MAX31865 max31865 = Adafruit_MAX31865(MAX31865_CS_PIN
    #if MAX31865_CS_PIN != MAX6675_SS_PIN
      , MAX31865_MOSI_PIN           // For software SPI also set MOSI/MISO/SCK
      , MAX31865_MISO_PIN
      , MAX31865_SCK_PIN
    #endif
  ); 
```

This code checks to see if MAX31865_CS_PIN is not equal to MAX6675_SS_PIN.  If they are equal then the Adafruit MAX31865 library call is passed only one argument, the CS pin for the MAX31865 board. 

By only passing one argument the Adafruit MAX31865 library will default to the hardware SPI bus for the board!! 

So you are actually using hardware SPI not Software SPI.  I told you wrong.   I am sorry for misleading you.  What you need to do to get Software SPI is the following which will ensure that MAX6675_SS_PIN and MAX31865_CS_PIN are NOT equal:
```
#define TEMP_0_PIN  PD0
#ifndef MAX31865_CS_PIN
    #define MAX31865_CS_PIN TEMP_0_PIN
    //now MA6675_SS_PIN is NOT equal to MAX31865_CS_PIN, which will force Software SPI to be used
   #define MAX6675_SS_PIN PG13      //ensure you are not using this PIN, this pin is on EXTENSION-1 header
   #define MAX31865_MOSI_PIN PD2
   #define MAX31865_MISO_PIN PD5
   #define MAX31865_SCK_PIN PE0
#endif
```

I think this has been the problem all along.  Again I am sorry I didn't catch that earlier.  I will fix it in the manual in an update.

code505, 2020-09-24:
Halaloiaaaaa

Thank you a million bro, mow it is working like charm, I really appreciate your effort MAN

I have only some issues now with PID autotune future, this message appear,
[image]

and also octoprint can't connect because I think of the same problem:
[image]
[image]

GadgetAngel, 2020-09-24:
@code505 To try to get rid of that problem I suggest you read the section in configuration_adv.h file on temperature. 

I have not personally used this Marlin variable yet.  But try searching for MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED and uncomment the Marlin variable. 

Set MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED to 10 or some number that get rid of the Temperature measurement error. 

This is what I set mine to:

[image]

GadgetAngel, 2020-09-24:
@code505 I found where the **TEMP MEASUREMENT MAX6675** error message is coming from:

[image]

The above code in temperature.cpp is using the Marlin variable **THERMOCOUPLE_MAX_ERRORS**.  

**In configuration_adv.h set THERMOCOUPLE_MAX_ERRORS to 20** and see if the errors go away.  

[image]

I know you are using a PT100 temperature sensor and you may be asking why would a **THERMOCOUPLE_MAX_ERRORS** variable (which is concerned with Thermocouples) be affecting the software.  From what I can figure out from the Marlin code, Marlin reuses as much of the code when any new features are added.  It appears that thermocouples where developed first.  When they added the feature for PT100 temperature sensors, Marlin layered it on top of the thermocouple code.  So **THERMOCOUPLE_MAX_ERRORS**  does affect PT100 sensor also.  I would leave in **MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED** Marlin variable and play with the **BOTH values** to see if you can fix the error.

If the errors **do not go away**, do the following:
Check the wiring to the MAX31865 board.  Check the wires for continuity.  All the wires.  
    1. Check continuity from PT100 to MAX31865. 
    2. Check continuity from MAX31865 to SKR PRO V1.1 board. 

Noise is getting into the temperature reading.  Check the power supply going to the MAX31865.  

If you are using a seperate power supply (not the 3.3VDC from the board) ensure you tie the GND of the separate power supply to the GND on the board or you will induce a ground loop which will cause noise with the temperature reading.

code505, 2020-09-24:
still no luck with all suggestions, I think the problem is here:

[image]

[33 more replies]