bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #165: USB serial printing does not work (Linux, slic3r, RepetierHost, SKR Pro V1.2)

Yenya, 2020-09-10:
Hello,

I am trying to configure SKR Pro V1.2 for printing over the USB-serial port from Linux with either slic3r or RepetierHost. Both of these programs work on the same computer correctly with my Arduino Mega + RAMPS 1.4 + Marlin 1.9 setup. With SKR Pro, I can see a new serial port /dev/ttyACM1 created, when I do "`cat /dev/ttyACM1`" and from the other terminal "`echo M105 > /dev/ttyACM1`", I see the correct output "`ok T:-15.00..."` in the first terminal. However, when I try to connect from slic3r's Controller tab, I get about 5 seconds pause, and then "Connection failed. Check serial port and speed." error message. With RepetierHost, it connects to the serial port, but immediately says "6 commands waiting, and no communication except the initial M105 without response" is shown in the log window.

 I have the following in my `Configuration.h`:

```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE 250000

```
It might be that the USB serial port emulation inside the STM32 does not handle for example the stop bit, echo, rts/cts, or other serial settings from the Linux host.

I tried the master branch of Marlin as well as the bugfix-2.0.x branch - both behave the same.

Does anybody here print on SKR Pro V1.2 over the USB serial port from Linux?

Thanks,

-Yenya

Yenya, 2020-09-11:
Exploring this further, even Pronterface does not work. When it tries to connect, the following errro message is written to the kernel log:

`cdc_acm 1-10:1.0: failed to set dtr/rts`

When I do "`cat /dev/ttyACM1`", and from another terminal I repeatedly run "`echo M105 > /dev/ttyACM1`", the output looks garbled sometimes:
```

$ cat /dev/ttyACM1 
ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0
echo:Unknown command: "ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0"
ok
echo:Unknown command: "echo:Unknown comma0 B:-15.00 /0.00M105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commaM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commokM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown comhoM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown comma commM105""
ok
echo:Unknown command: "ok"
ok
ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0
echo:Unknown command: "echo:Unknown omchok B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0"
ok
ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0

```
There seems to be something horribly wrong with USB serial port emulation on SKR Pro MCU.

oldman4U, 2020-09-14:
Hi.

Would it be helpful for you if I try the same on macOS and Printrun and SKR Pro v1.1?

Yenya, 2020-09-14:
@oldman4U - as far as I can see v1.1 and v1.2 have the same MCU, so yes, it might be helpful to try it under MacOS as well.

I see this repo has even drivers for the serial port for Windows, so it might be that the Windows driver papers over some non-standard behaviour, which other systems with native serial drivers see.

oldman4U, 2020-09-14:
Hi.

I used an a bit older Marlinversion 2.0.5.3 and serial port -1 and 1 as well as 115200 Baudrate. No problem to connect using Printrun and no extra drivers installed. I only tested to connect and more the axis - no actual print tested but I guess this would also work. No drivers installed.

Do you have a reliable 12V or 24V PS connected or USB only?

Anything else I can help with?

Yenya, 2020-09-14:
Does the Printrun attempt to reset the printer with lowering the DTR signal, set/clear the RTS/CTS handshake or something? My guess is that something related to emulating the actual physical serial port fails with SKR Pro.

My test setup has 14V/2A power supply, from which the SKR Pro, TFT35E3, and one stepper on X axis driver are powered. Tried also the USB-only power (no stepper, of course), with and without the TFT35E3. No difference.

Yenya, 2020-09-14:
Another interesting thing - but maybe not related to this issue: when I instruct Marlin to home the X axis, where I have a testing stepper only but no endstop, the homing fails with "kill() called - reset the printer" or something like that. But hitting reset does not help - the TFT display reports "No printer attached". The only thing that helps is to power the whole setup (SKR Pro + TFT) down, wait for a few seconds, and then power it up again.

oldman4U, 2020-09-14:
Sorry, but I do not know what Printrun actually does to connect.

How do you instruct Marlin to home, using the TFT in Marlinmode?

No printer attached when using the touch mode I guess? Hitting Reset where, on the SKR board? Serial cable attached? I have the same display btw.

Yenya, 2020-09-14:
Tried using TFT, and using Pronterface (the rare case I managed to connect). TFT is in Touch mode, connected over Serial 1. Tried both reset buttons, no difference.

Yenya, 2020-09-14:
Another interesting thing is that hitting reset button (any of them) does not result in USB disconnect and reconnect events in the Linux host.

oldman4U, 2020-09-14:
When I reset the SKR board I get this in Printrun:

[ERROR] Can't read from printer (disconnected?) (OS Error 6): Device not configured
[ERROR] Can't write to printer (disconnected?) (SerialException): write failed: [Errno 6] Device not configured

Yenya, 2020-09-14:
Would you mind to re-test with the latest Marlin and my configuration attached to the above mentioned Marlin issue? Or vice versa - could you send me the exact version (git commit hash) of your firmware, and your configuration (and other possible local modifications)?

oldman4U, 2020-09-14:
I can upload. It is a test machine

2.0.6.1 or latest latest bugfix?

Yenya, 2020-09-15:
Either one is welcome, thanks!

oldman4U, 2020-09-15:
Hi.

I just downloaded the Bugfix Marlin from today. Can you send me your configuration files and I will test it.

Yenya, 2020-09-15:
It is attached to the other issue, mentioned above: https://github.com/MarlinFirmware/Marlin/files/5207416/skrpro-config.zip

Thanks!

oldman4U, 2020-09-15:
OK.

So I used the firmware above, changed the env. in PlatformIO.ini and replaced the two config files with yours. Updates the firmware on my SKR Pro v1.1 and even v1.2 was defined (MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2) it took it (led close to the CPU flashed) and I had to change the baud rate to 250000 to get the TFT connected. So everything tells me that the firmware was loaded.

Opened Printrun, clicked on Connect - Printer is now online, and I could control my mainboard.

I use 12/24V Power and of course Host USB selection like shown in the attached picture.
[image]

For me this works...

Yenya, 2020-09-15:
So maybe it is someting like half-baked serial port emulation, which Printrun does not fully use and thus does not see a problem. Can you test also from Linux?

oldman4U, 2020-09-16:
Or it is a hardware issue....

Unfortunately not, sorry. Otherwise i would have offered from the beginning.

Yenya, 2020-09-16:
HW issue might be the case, altough I can't imagine how would this work, as the serial port is emulated inside the USB connection. Anyway, thanks for checking my configuration.

More observations:
Marlin compiled with RepRap Discount LCD support (`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`), and TFT35-E3 connected with ribbon cables in Marlin mode, everything powered from USB only. After power-up, I see Marlin info screen as expected, and the /dev/ttyACM1 serial port appears in the Linux host. When push the RESET button (either on the TFT or on the SKRPro board, they behave the same), the /dev/ttyACM1 device is disconnected from Linux and doesn't come back. Also, the TFT now displays only the "LCD12864 simulator" title only. So it seems that after reset, the Marlin fw does not boot again.

From dmesg output on the Linux host:
```
[1178283.921758] usb 1-10: new full-speed USB device number 67 using xhci_hcd
[1178284.242468] usb 1-10: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[1178284.242470] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1178284.242471] usb 1-10: Product: STM32F407ZG CDC in FS Mode
[1178284.242472] usb 1-10: Manufacturer: STMicroelectronics
[1178284.242472] usb 1-10: SerialNumber: 207731745252
[1178284.259523] cdc_acm 1-10:1.0: ttyACM1: USB ACM device
_(here I hit the reset button)_
[1178288.269298] usb 1-10: USB disconnect, device number 67
```

Yenya, 2020-09-16:
Another observation: tried the binary firmware which I got with the SKRPro board - fortunately I saved it beforehand - I am able to connect to it and communicate with it with Slic3r, RepetierHost, and Pronterface. Also tried the SKRPro v1.1 firmware blob for TMC2209 drivers from this repository:

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/firmware/SKR%20ProV1.1-Marlin_v2.0-12864p-TMC2209Uart_Drivers-Test/firmware.bin

- this does not work at all over the USB serial connection, the same way as my own compiled firmwares. But maybe the reason is that it is for v1.1.

Another interesting thing is that when I hit the RESET button, the /dev/ttyACM1 device disappears and doesn't come back until power cycle, but when I eject the microSD card and plug it back, the device appears again in few seconds, and the LCD12864 simulator displays the Marlin boot screen and then the info screen.

----

It might be something with the toolchain (compiler/binutils version, libraries, or something like that). @oldman4U - could you please send me the firmware binary you have compiled from my configuration? Either attach it here, or mail it to me - my address is kas @ <my github nickname> .net. Thanks!

oldman4U, 2020-09-16:
Here you are. Just tested the reset on the SKR and it makes the SKR and the TFT reset. After that, the TFT connects properly to the SKR using touch screen (can not test Marlin because i removed the knob.... don't ask;-))

[Archive.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5231607/Archive.zip)

Yenya, 2020-09-16:
I am sorry but I see only Configuration.h and Configuration_adv.h in this zip file. What I want is the resulting firmware.bin compiled on your system with your toolchain. Thanks!

oldman4U, 2020-09-16:
As far as i know this is not allowed. But you can send me your private email or a upload link and i can do so.

Yenya, 2020-09-16:
My e-mail is kas@(my github nickname).net, as I wrote above. Thanks!

oldman4U, 2020-09-16:
It is on the way.

Let me know how it works

Yenya, 2020-09-16:
Got it, thanks! It behaves more-or-less the same as my own FW: after power-up, I see /dev/ttyACM1, after reset it disappears and doesn't come back (and the TFT display reports "No printer attached"), after ejecting the microSD card and plugging it back the device appears again, and the TFT display connects. Connection from slic3r or RepetierHost does not work as before, Pronterface seems to be working correctly (as it does most of the time with my own compiled firmware - I guess my previous problems with Pronterface were in fact related to the reset problem, now I am able to use Pronterface reliably).

So, if I understand it correctly, the reason could be:

- my hardware is faulty
- there is something wrong with serial port emulation in the current Marlin, which breaks the way both slic3r and RepetierHost use the serial port on Linux.

I have also discovered that even though I can connect from Pronterface, I get truncated output from the M43 command. Each time it is truncated at slightly different place, usually somewhere between the PD* and PE* ports. Does this work for you?

oldman4U, 2020-09-16:
[image]
[image]
[image]

I have to go, so taking pictures was the fastest way to send you the M43.

Yenya, 2020-09-16:
So, it seems your output is also truncated, albeit at different points than mine.

oldman4U, 2020-09-16:
Where do you see the truncation?

You are welcome

Yenya, 2020-09-16:
Your output ends sometimes with PH1, sometimes with PD0, etc. It should end with the last pin name every time.  Mine is truncated around PD(last) to PD(0 or 1).

oldman4U, 2020-09-17:
I never took care because it works. But you are right. 

Do you have a WIFI module?  I use it for this kind of communication.

lamikr, 2020-09-17:
Could it be that the modemmanager or some other software like that is trying to connect simultaneously with the /dev/ttyACM0?

Yenya, 2020-09-18:
Interesting guess, but no - I use XFCE and have no modemmanager running. As far as I can tell, (fuser(1)) nobody has /dev/ttyACM1 opened.

No WiFi module here.

oldman4U, 2020-09-30:
Please help the community and close the ticket once you do not need it anymore. Thank you

Yenya, 2020-10-02:
It seems the same problem - not being able to print over serial port using slic3r and RepetierHost, while Pronterface apparently works - is also on SKR v1.4 Turbo. Keeping the issue open for now.

oldman4U, 2020-10-02:
Ok. Thank you for the update

evandene, 2020-11-25:
[image]

Did you checked this jumper?

Yenya, 2020-11-25:
Yes, of course.

evandene, 2020-11-25:
Great, I hope you could solve your issue.
I'm struggling with my limit switches on my skr pro 1.2 and my DELTA printer. Tried everything and from BTT 0.0 support.