bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #167: bl touch m999 error skr mini e3 v2.0

jimthedinosuar, 2020-09-24:
Hello, I recently installed a bl touch version 3.1 on my ender 3 pro and while when the test worked fine when I went to auto bed level the m999 error came up. I tried rewiring it but that did not work it still brought up the m999 error I have looked at different git hub issues and tried them but nothing seems to work. Thank you if you have a solution

oldman4U, 2020-09-24:
Hi.

This is the wrong repository for your mainboard.

Please close this ticket and start a new ticket here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues

Thank you

jimthedinosuar, 2020-09-24:
Thank you @oldman4U

oldman4U, 2020-09-24:
👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-09-28 by jimthedinosuar