bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #169: endstops not working after plugging in a driver (TMC2209)

outtacontrol216, 2020-09-25:
without a driver in the slot the endstop is working fine (red LED is ON and goes off when i trigger the endstop).
When i power off the board and plug in a TMC2009, the red LED is not ON and the endstop for this slot is not working anymore.
setup of the endstops are COM to ground and NC to Signal.
in marlin 2.0 endstop inverting is on false.

thank you for your help

oldman4U, 2020-09-25:
Are you going to use sensor-less homing?
If not, have you removed the diagram pin from the 2209?

outtacontrol216, 2020-09-26:
i am not going for sensor-less homing.
there is absolutely no difference when i remove the DIA pin of the TMC2209. 
as is said, the endstops are working and the red LED`s near the driver are ON only when there is no driver in the slot.

outtacontrol216, 2020-09-26:
update: u cutted the wrong pin
https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf
helped me alot
thank you

closed 2020-09-26 by outtacontrol216

oldman4U, 2020-09-26:
👍🏻