bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #172: Board not accepting firmware

quad121, 2020-10-09:
When trying to flash firmware to the board using the sd card and 'firmware.bin' file won't install to board.

oldman4U, 2020-10-09:
Which firmware. Sd card size and format. Pictures. 

Please

quad121, 2020-10-09:
I am trying to flash 'klipper' to the board using a Sandisk 32gb sd card, which has been formatted in the 'fat32' format (new card).

quad121, 2020-10-09:
[image]

oldman4U, 2020-10-09:
I would use a card smaller 8GB for firmware flash, best the one which was included. Klipper I do not know. Sorry

oldman4U, 2020-10-09:
But I would try it with the standard Marlin for a test of the hardware.

quad121, 2020-10-09:
Many thanks for your time, I will try and flash the board with marlin and see how it goes.

oldman4U, 2020-10-09:
Let me know.

quad121, 2020-10-09:
I will.

quad121, 2020-10-09:
O.K, I've downloaded the latest Marlin and configured and built using VS Code. 
The file still never got accepted!

quad121, 2020-10-09:
I can't believe that a large manufacturing firm like Big TreeTech keep on supplying faulty goods?
I have had 2 skr 1.4 turbo mobo' s and both turned out to be faulty, as both had faulty mosfets on the part cooling fan output.

oldman4U, 2020-10-09:
Can you zip you config and advanced and attach it here?

oldman4U, 2020-10-09:
And use the sd card which came with the board.

You can rename the firmware file and try it directly with it.

quad121, 2020-10-09:
[untitled folder.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5355483/untitled.folder.zip)

This file was loosely configured, just to try to see if it would load!
I also did use the SD Card which came with the board

quad121, 2020-10-09:
I also tried the BTT's Firmware also!!!

quad121, 2020-10-09:
There also doesn't seem to be any lights lit either? Ive checked the fuses etc

oldman4U, 2020-10-09:
Yes, please do so...

I mean you have connected almost everything possible, but you should start with the basics and add more stuff later on.

quad121, 2020-10-09:
I've found this, I don't know if it will work but I suppose it's worth a go.

https://www.youtube.com/watch?v=7O_ZOdnM0SU&feature=youtu.be

oldman4U, 2020-10-09:
Maybe you want to check if it takes power from USB first (check the jumper configuration) and also if the fuses are ok.

quad121, 2020-10-10:
While waiting for the ST Link I thought I would connect the printer using Pronterface. Which it does connect! No function of the Axis though
[image]

quad121, 2020-10-10:
This must be firmware which was installed via the Company!

quad121, 2020-10-10:
Ok, I've sorted the problem out and also have got the firmware to flash also

closed 2020-10-10 by quad121

dzervas, 2021-12-23:
but HOW did you solve it? @quad121

afbenevides, 2022-01-18:
Hi, I would be interested in new solution ideas too if one has? I'm stuck with SKR pro 1.2 too not accepting firmware. Thanks, @dzervas @quad121

afbenevides, 2022-01-18:
Thanks @quad121 the YouTube link + GitHub link did the trick for 2 of them... now I have to investigate the other 2 problem...

I put them here in case it helps someone else:

https://www.youtube.com/watch?v=7O_ZOdnM0SU
https://github.com/P-C-R/SKR_PRO_BOOTLOADER_REPAIR

olijouve, 2022-11-09:
Hello @quad121, what was your issue that used to make the fuses blow immediately after replacing, and how did you fixed it ?