bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #173: Increasing z-axis offset while running

HorstHardcore, 2020-10-11:
Hi,

I am not sure, if this is the right place for my problem, but I couldnt get help so far. So I want to ask for help here now.

My printer is running with a SKR 1.1 pro with TMC2208 drivers. Both of the stepper motors are controlled by the same stepper driver, but both motors got their own connector on the mainboard. The problem is, that the longer my printer is running and moving in z-axis, the more tilted my x-axis carriage (where the extruder is mounted) gets.
That means, I calibrate my z-axis, so that the distance between z-axis and print bed is the same on both sides (right and left) and it constantly sets off one side of the z-axis, when the printer is running. And its reproducable. Its about 0,5 mm offset for about 1 or 1,5 hours of print time. When i flip the connectors of stepper motors on the mainboard, it also flips the offset (on which side the offset happens). 

Now I am curious, if my mainboard has an electric defect, because i cant imagine that this problem is up to my stepper drivers, because both motors are controlled by the same driver and just to be sure, I already exchanged the driver for the z-axis and this didnt help at all. Or maybe there is a totally different reason for the problem?

I would be very glad, if somebody can help. If further information are neccessary, please ask.

dineshvyas, 2020-10-12:
Keep same arrangement and only swap cables. See if problem persists.

oldman4U, 2020-10-27:
@HorstHardcore 

Any news?

dineshvyas tried to help, but never got an answer from you.

HorstHardcore, 2020-11-22:
@dineshvyas and @oldman4U I am very sorry for my late answer. I appreciate the help but i had very little time to spend on my printer lately and couldnt test it as much. I swapped the cables and it seems to be better, although i cant say yet, that the problem is completely solved.
Despite having little time to test it properly, I also ran into another problem which prevents me from verifying my assumption. I am sorry.

oldman4U, 2020-11-22:
No reason to be sorry, but if you can not follow the problem it would be best to close the ticket and reopen it once you have time.

Thank you

closed 2020-11-23 by HorstHardcore