bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #175: Caught on Fire

rjdandy, 2020-10-25:
I wired the board to 24VDC for the motors and the board and powered up with no issue. But when i went to connect my computer to the board through USB the processor on the board caught fire, the USB cable melted and my usb ports on my computer were damaged and not longer work.

oldman4U, 2020-10-26:
Hi. How can we help you?

rjdandy, 2020-10-26:
I was wondering if anyone has seen this before or tell me what i did wrong that would have caused this. I am going to order a new board and i don't want this to happen again.

oldman4U, 2020-10-26:
This makes sense.

What kind of power supply do you use? A broken CPU is not very common.

rjdandy, 2020-10-26:
FYI, I am building a Mostly Printed CNC machine.

When i first powered up the board to check firmware i powered the board through USB and everything was ok as well. Once i got the machine built and wired to the external power supply i moved the jumper between IN and VCC

Padarsey DC 24V 15A 360W Power Supply
https://www.amazon.ca/gp/product/B07CPTHG7N/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1

oldman4U, 2020-10-27:
Looks like a decent power supply. Have you checked the output of it to be 24V?

Especially the SKR Pro is very stable when it comes to power input and internal power management and I have two of them working flawlessly.

I used the USB only for Printrun at the beginning, now I do not use it at all and use WIFI instead. So my experience with USB is very limited I must say.;-(

Hope this helps