bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #176: BIGTREETECH-SKR-PRO-V1.1 + max31865 + tft35 v2.0  "no printer attached"

daniagos, 2020-10-26:
Hi, 
i am runnng a skr pro v 1.1
tft 35 v2
tmc2209, uart mode, sensorless homing
marlin 2.0.x, latest release
While trying to change TEMP_SENSOR_0 from 1 to -5 (which indicates i am using a max amplifier) the tft disconnects.

i can't also upload firmware via usb but i need to upload it every time with sd card, quite annoying (i use vscode)

https://drive.google.com/drive/folders/1DBuVxjGbWB0jsPYh_vwEzkzLrf90pklb?usp=sharing

this is the link with the full marlin version i am using, here the tft is not working

GadgetAngel, 2020-10-27:
Please see this following issue:  https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/163#issuecomment-680798334

GadgetAngel, 2020-10-27:
> Hi,
> i am runnng a skr pro v 1.1
> tft 35 v2
> tmc2209, uart mode, sensorless homing
> marlin 2.0.x, latest release
> While trying to change TEMP_SENSOR_0 from 1 to -5 (which indicates i am using a max [...]

Please ZIP up the Marlin folder using WINZIP.  The folder takes to long to download uncompressed.  I need to download the ZIP file containing the whole Marlin folder.

GadgetAngel, 2020-10-27:
> Hi,
> i am runnng a skr pro v 1.1
> tft 35 v2
> tmc2209, uart mode, sensorless homing
> marlin 2.0.x, latest release
> While trying to change TEMP_SENSOR_0 from 1 to -5 (which indicates i am using a max [...]

How do you have your MAX31865 wired up to the SKR PRO V1.1 board? 

On top of that I really need to see the error message coming from Marlin for the MAX31865 board, but since you are using a TFT screen you will not see the message all you will see in "no printer attached".  Please use a 12864 LCD on EXP2 so you can see what error message Marlin is sending you.  Yes, the 12864 LCD uses both the EXP1 and EXP2 connectors but you just need to clamp on a tap to EXP2 to access the hardware SPI lines (https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/1658622-1/AKC10B-ND/825411)

But from looking at your firmware files it looks like the Hardware SPI configuration you are using the MAX31865 should work.

To do any more debugging we need to know the exact error message Marlin is sending out.

GadgetAngel, 2020-10-27:
## @daniagos I noticed that in **configuration.h** file you have not **enabled** the Marlin variable **SDSUPPORT**.  I believe you need to enable this parameter.

If you enable SDSUPPORT and get the 12864 LCD we can test to see if the hardware SPI is working on the SKR PRO V1.1 board by seeing if you can use the SD card on the LCD screen.  Without SDSUPPORT the SD card is not supported because the SD_DETECT_PIN is not defined:

[image]

[image]

## With your tft screen, try just enabling SDSUPPORT in configuration.h and see if that change effects the problem.

daniagos, 2020-10-27:
[image]

this is the wiring setup

is with this connector ((https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/1658622-1/AKC10B-ND/825411)) possible to attach both lcd and use spi hardware cables?

I enabled sd card pin, and i recieve the error " Printer halted. kill() called! " as soon as it turns on, then disconnects

daniagos, 2020-10-27:
you left me on facebook, now i want to use only one of it, to add the second later, but even with only one it can't work

daniagos, 2020-10-27:
i tried commenting all the "THERMAL_PROTECTION" but nothing changes, i have the same error and then it disconnects

daniagos, 2020-10-27:
I tried to set the "TEMP_SENSOR_0 -3", i still have the same error, but the tft stay connected, i can see bed temperatures.
i disabled all the security features, set min temp to -100, but with "TEMP_SENSOR_0  -5" there is no way to have the tft connected

once i switch back to 1 all runs and the thermocouple works fine.

daniagos, 2020-10-27:
i have the same error even with "TEMP_SENSOR_0 147" and 1047, there is something really strange

daniagos, 2020-10-27:
[image]

it seem as if the option is disabled even if it is defined correctly

GadgetAngel, 2020-10-28:
> [image]
> 
> it seem as if the option is disabled even if it is defined correctly

It appears disabled because you have not **enabled** SDSUPPORT in configuration.h file.  Please remove the two forward slashes on the SDSUPPORT line in configuration.h file.  

Yes, this connector ((https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/1658622-1/AKC10B-ND/825411) will allow you to add a third connector to EXP2 which will then allow you to access the hardware SPI lines that the LCD screen uses.  The default hardware SPI bus is used by the EXP2 connector or the LCD screen.  As long as the MAX31865_CS_PIN is different than the SDSS PIN then both can use the same hardware SPI bus. 

PUT you MUST enable the Marlin variable SDSUPPORT in configuration.h file.

daniagos, 2020-10-28:
[image]
[image]

As you can see i enabled sd card, but that part still seem disabled.

i can't understand why i have also error if i use pt100 set as "TEMP_SENSOR_0 147", and the pt100 connected to the thermocouple pins, this should work even without spi, right?

GadgetAngel, 2020-10-28:
@daniagos upload your zipped Marlin folder again to your google drive, so I can look at the changes you currently have in your firmware.  You will have to get a new shared link to the new zipped Marlin folder and place the link here in the thread so I can download your Marlin folder.

> i can't understand why i have also error if i use pt100 set as "TEMP_SENSOR_0 147", and the pt100 connected to the thermocouple pins, this should work even without spi, right?

I do not understand what you are trying to do here.  Did you remove or disable the following in pins_BTT_SKR_PRO_common.h when you set TEMP_SENSOR_0 to 147?:
```
#ifndef MAX38165_CS_PIN
  #define MAX6675_SS_PIN                   TEMP_0_PIN
  //#define MAX6675_SS2_PIN                  TEMP_1_PIN // Force hardware SPI to be used making MAX38165_CS_PIN equal to  MAX6675_SS_PIN
  #define MAX38165_CS_PIN                  MAX6675_SS_PIN
#endif
```
Did you change TEMP_0_PIN also? Why are you using a thermocouple? I thought we were hooking up a PT100 and MAX31865 board.  Let's take on problem at a time.  Which situation do you  want to debug?  

When I looked at your Marlin folder earlier and I enable SDSUPPORT with the TEMP_SENSOR_0 set to -5 and pins_BTT_SKR_PRO_common.h set as follows 

```
#ifndef MAX38165_CS_PIN
  #define MAX6675_SS_PIN                   TEMP_0_PIN
  //#define MAX6675_SS2_PIN                  TEMP_1_PIN // Force hardware SPI to be used making MAX38165_CS_PIN equal to  MAX6675_SS_PIN
  #define MAX38165_CS_PIN                  MAX6675_SS_PIN
#endif
```
the section listed as 
```
#if SD_CONNECTION_IS(LCD)

  #define SD_DETECT_PIN                     PF12
  #define SDSS                              PB12

```
was enabled!

So you have changed too many things and now I have no idea what state your Marlin firmware is in.  So again I ask that you upload you current Marlin folder to your google drive so I can help you debug what is going wrong.  

Which PT100 cartridge are you using?    

How did you hook up the PT100 sensor (or cartridge) to the thermistor port without an amplifier board?  

Most PT100 need amplification "The E3D PT100 amplifier board amplifies the signal from the PT100 to a voltage level that your printer controller can read easily and accurately" but the E3D PT100 amplifier board means you would use thermistor table 20 or 21.  I know you can use this PT1000 sensor and use thermistor table 1047 (https://e3d-online.com/products/pt1000-temperature-sensor?_pos=1&_sid=26fb8c09c&_ss=r). 

I do not know why your PT100 did not work with table 147 because I do not have all the above information.  Lets stick to trying to get the MAX31865 board (amplifier board) working with hardware SPI and the PT100 sensor connected to the MAX31865 board.

I have never used thermistor table 147 with PT100 temperature sensor on a SKR PRO V1.1 board.

daniagos, 2020-10-28:
i want to debug the e3d pt100 with max amplifier.

starting from stock marlin to see if there are other things that gives me tft error, all security removed, SDSUPPORT enabled

1) i started with "TEMP_0_PIN 1" with a termocouple installation as in stock marlin.
all works fine, no error, temp corrected.
1.1) removed the sensor, it show -15 degrees but still no error as expected

2) i tried to see if there is problem with the pt100 installed as the termocouple, so removed termocouple, installed pt100 in the same 2 pins, set "TEMP_0_PIN 147".
ERROR: printer halted, kill() called

this make me think of a library problem more than a hardware problem

3) i tried to connect the max amplifier

wired it
checked that SDSUPPORT is enabled

wrote in pins 

#ifndef MAX38165_CS_PIN
  #define MAX6675_SS_PIN                   TEMP_0_PIN
  //#define MAX6675_SS2_PIN                  TEMP_1_PIN // Force hardware SPI to be used making MAX38165_CS_PIN equal to  MAX6675_SS_PIN
  #define MAX38165_CS_PIN                  MAX6675_SS_PIN
#endif

changed

 #define TEMP_0_PIN  PF4         to      #define TEMP_0_PIN       PE4

And do the other setting on config_adv listed in your picture.
ERROR: printer halted, kill() called.

i think that i don't have the right library installed, on platformio is there only available the "adafruit/Adafruit MAX31865 library@^1.3.0", That is listed in my env and installed, but on the  [features] in platformio i find this:

 max6675_is_max31865 = Adafruit MAX31865 library@~1.1.0

is there something i should correct?

as soon as i can i zip the marlin and upload it on google drive

GadgetAngel, 2020-10-28:
### @daniagos This is my current understanding of the different PT100 thermistor tables in Marlin 2.0.x:

---
### PT100 Thermistor tables and how to use them:

I should use **table "-5"** if I am using a PT100 sensor attached to the MAX31865 board and the SPI bus.

I should use **table "20"** when I use a PT100 sensor attached to a PT100 amplifier board (INA826 amp) AND the power supplied to that PT100/PT1000 Amplifier board (INA826) IS EQUAL TO the ADC reference voltage for the MCU AND there is NO Pull-UP resistor.  --NOTE: For SKR PRO V1.1/V1.2, SKR V1.3/V1.4/V1.4 Turbo, GTR V1.0, M5 V1.0 all use 3.3 VDC as the ADC reference voltage for the MCU. Therefore use table 20 when the PT100/PT1000 Amplifier board is powered by 3.3VDC

I should use **table "21"** when I use a PT100 sensor attached to a PT100 amplifier board (INA826 amp) AND the power supplied to that PT100/PT1000 Amplifier board (INA826) IS DIFFERENT FROM the ADC reference voltage for the MCU AND there is NO Pull-UP resistor.  --NOTE: The E3D PT100 Amplifier board wants 5V as its power supply. But the SKR PRO V1.x (like the GTR V1.0) uses 3.3VDC as the ADC reference voltage for the MCU. If you use 5VDC to power to the PT100 amplifier board, the amplifier can output a maximum signal with 5VDC. To protect the SKR PRO V1.x board (or the GTR V1.0 board, or SKR V1.3 or SKR V1.4 or SKR V1.4 TURBO), use a 3.6V Zener Diode (reverse biased hookup) between the output signal of the PT100 amplifier board and the PT100 amplifier board's ground (to protect the MCU from getting fried from a 5VDC input signal).

I should use **table "147"** when I use a PT100 sensor attached to a PT100 amplifier board (INA826 amp) that uses 5VDC as its input voltage and is wired up to a MCU's analog input PIN which is 5VDC tolerant and has a 4.7k pull-up resistor attached to the analog input PIN.

I should use **table "110"** when I use a PT100 sensor attached to a PT100 amplifier board (INA826 amp) that use 5VDC as it input voltage and is wired up to a MCU analog input PIN which is 5VDC tolerant and has a 1k Ohm pull-up resistor attached to the analog input PIN.

I should use **table "201"** when I have a PT100 sensor attached to a PT100 amplifier board that uses LMV324 amp (used by the 3D printer Overlord V1.1) and the amplifier board uses 5VDC as its input voltage and is wired up to a MCU analog input PIN which is 5VDC tolerant and has NO pull-up resistor attached to the analog input PIN.

I should use **table "1047"** when I have a PT1000 sensor attached to NO AMPLIFIER board and the PT1000 sensor is wired up to a MCU analog input PIN which is 5VDC tolerant and has a 4.7K pull-up resistor attached to the analog input PIN. Here is the PT1000 sensor (https://e3d-online.com/products/pt1000-temperature-sensor)

I should use **table "110"** when I have a PT1000 sensor attached to NO AMPLIFIER board and the PT1000 sensor is wired up to a MCU analog input PIN which is 5VDC tolerant and has a 1K pull-up resistor attached to the analog input PIN.  Here is the PT1000 sensor (https://e3d-online.com/products/pt1000-temperature-sensor)

---

If you look at the processor data sheet for the SKR PRO V1.x or SKR V1.3/SKR V1.4 or SKR V1.4 Turbo or GTR V1.0/M5 V1.0 boards (https://www.st.com/resource/en/datasheet/stm32f407zg.pdf) or (https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Datasheet/LPC17-32-IC.pdf) or (https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Datasheet/LPC17-32-IC.pdf) or (https://www.st.com/resource/en/datasheet/stm32f407ig.pdf) you will see the PINs for the thermistor ports **are 5VDC tolerant as long as they are NOT USED for analog input.**  When they are **used for analog input they are ONLY 3.3VDC tolerant**. See note (4) on processor data sheet for SKR PRO 1.x or GTR V1.0/M5V1.0. See note [2] on processor data sheets for SKR 1.3/V1.4 or SKR V1.4 Turbo.  Therefore, you can use a 3.6V Zener Diode (reverse bias hookup) between the analog output pin for the amplifier board and the amplifier board's ground pin to protect the SKR series boards from getting fried:

[image]

GadgetAngel, 2020-10-28:
> i want to debug the e3d pt100 with max amplifier.
> 
> starting from stock marlin to see if there are other things that gives me tft error, all security removed, SDSUPPORT enabled
> 
> 1. i started with "TEMP_0_PIN 1" with a termocouple installation as in stock marlin.
>    all works fine, no error, temp corrected.
>    1.1) removed the sensor, it show -15 degrees but still no error as expected
> 2. i tried to see if there is problem with the pt100 installed as the termocouple, so removed termocouple, installed pt100 in the same 2 pins, set "TEMP_0_PIN 147".
>    ERROR: printer halted, kill() called
> 
> this make me think of a library problem more than a hardware problem
> 
> 1. i tried to connect the max amplifier
> 
> wired it
> checked that SDSUPPORT is enabled
> 
> wrote in pins
> 
> #ifndef MAX38165_CS_PIN
> #define MAX6675_SS_PIN TEMP_0_PIN
> //#define MAX6675_SS2_PIN TEMP_1_PIN // Force hardware SPI to be used making MAX38165_CS_PIN equal to MAX6675_SS_PIN
> #define MAX38165_CS_PIN MAX6675_SS_PIN
> #endif
> 
> changed
> 
> #define TEMP_0_PIN PF4 to #define TEMP_0_PIN PE4
> 
> And do the other setting on config_adv listed in your picture.
> ERROR: printer halted, kill() called.
> 
> i think that i don't have the right library installed, on platformio is there only available the "adafruit/Adafruit MAX31865 library@^1.3.0", That is listed in my env and installed, but on the [features] in platformio i find this:
> 
> max6675_is_max31865 = Adafruit MAX31865 library@~1.1.0
> 
> is there something i should correct?
> 
> as soon as i can i zip the marlin and upload it on google drive

YES, you only want Adafruit MAX31865 library version 1.1.0.  Un-install Adafruit MAX31865 V1.3.0 library! 

Here is the way to tell that you are using the right library.  When you compile the code after the success message appears, scroll to the very top of the output listing and look for the library version the compiler used to create the .bin file:

[image]

To uninstall the wrong library versions go to the Platformio HOME page and select "Libraries", then select the Installed tab, then in search field type "Adafruit MAX31865". Select the uninstall button for all libraries that are not V1.1.0.

[image]

Once all MAX31865 libraries of version 1.3.0 are un-installed then just recompile the code. Marlin will see that the Adafruit MAX31865 library is not installed and automatically download the correct version 1.1.0.

GadgetAngel, 2020-10-28:
> ok, so the trial i did are completely wrong.
> 
> uninstalled MAX31865 libraries version 1.3.0
> #define EXTRUDERS 1
> changed max6675_is_max31865 = Adafruit MAX31865 library@~1.1.0 to max6675_is_max31865 = Adafruit MAX31865 library@1.1.0
> 
> not founded other Adafruit MAX31865 library
> 
> as soon as i go back home i try again and hope everything works,
> 
> Thanks for precious help

If it works or does not work let me know.

daniagos, 2020-10-28:
it doesn't work neither in this way..

https://drive.google.com/drive/folders/1DBuVxjGbWB0jsPYh_vwEzkzLrf90pklb?usp=sharing

this is the link, there is the zip file of marlin of the actual configuration i tried..

GadgetAngel, 2020-10-28:
@daniagos Try connecting the MAX31865 board up via Software SPI instead of Hardware SPI.  If we get software SPI to work then we at least know the MAX31865 board is working properly.

GadgetAngel, 2020-10-28:
> it doesn't work neither in this way..
> 
> https://drive.google.com/drive/folders/1DBuVxjGbWB0jsPYh_vwEzkzLrf90pklb?usp=sharing
> 
> this is the link, there is the zip file of marlin of the actual configuration i tried..

I looked at your new configuration files from your google drive:

currently in **configuration.h** file you have:
`#define EXTRUDERS 1`
Which is what we want, we want to try to only get ONE hot end to work on MAX31865 board using hardware SPI.

BUT
currently in **pins_BTT_SKR_PRO_common.h** file you have:
```
#ifndef MAX38165_CS_PIN
  #define MAX6675_SS_PIN                   TEMP_0_PIN
  #define MAX6675_SS2_PIN                  TEMP_1_PIN // Force hardware SPI to be used making MAX38165_CS_PIN equal to  MAX6675_SS_PIN
  #define MAX38165_CS_PIN                  MAX6675_SS_PIN
#endif
```

When you ` #define MAX6675_SS2_PIN                  TEMP_1_PIN`  the temperature.cpp file will try to toggle the MAX6675_SS2_PIN to see if it can gather information off the second MAX31865 board.  So please leave everything the same EXCEPT **comment out** the ` #define MAX6675_SS2_PIN    TEMP_1_PIN` line **in pins_BTT_SKR_PRO_common.h file**.  

So I want your **pins_BTT_SKR_PRO_common.h file**:
```
#ifndef MAX38165_CS_PIN
  #define MAX6675_SS_PIN                   TEMP_0_PIN
  // Force hardware SPI to be used making MAX38165_CS_PIN equal to  MAX6675_SS_PIN
  #define MAX38165_CS_PIN                  MAX6675_SS_PIN
#endif
```

I know this might not fix the issue but I would like for you to try this before we have you try to use Software SPI instead for Hardware SPI.  Marlin must be working on the MAX31865 stuff.  I recently got two other users MAX31865 boards to work in software SPI but you are the first user I have worked with trying to get hardware SPI to work.  

Where did you buy your MAX31865 board? Did you get it from Adafruit?  Try the above change then let me know if it worked or not and we will talk about trying to use the board in Software SPI.  

You see, If the MAX31865 will work in Software SPI then we know the MAX31865 board is working.  The next step would be to open a tick under Marlin firmware to let them know we have a MAX3165 working in Software SPI but we can not get it to work in Hardware SPI mode.

GadgetAngel, 2020-10-29:
@daniagos Look at this: https://github.com/MarlinFirmware/Marlin/issues/5723#issuecomment-554800209

There might be a problem with the speed and mode.  From looking at the Marlin code, from HAL_SPI.cpp the SD card is using SPI mode 0 and from SdFatConfig.h the SD card is using Using SPI_SD_INIT_RATE 5 which sets the init clock at 328KHz for sd-card or 0.328MHz. 

For the MAX31865 board, the Adafruit_MAX31865 board in hardware spi mode, the MAX31865 board defaults to SPI MODE 1 (from Adafruit_MAX31865.cpp) and speed of 500000. 

So it appears the HAL_SPI.cpp(Marlin\src\HAL\STM32\HAL_SPI.cpp) for the SKR PRO V1.x board is using SPI MODE 0 while the Adafruit_MAX31865 board is defaulting to SPI MODE 1.  We could try and change the SPI MODE 1 (SPI_MODE1) to SPI MODE 0 or SPI_MODE0 and see if that makes it work. But if you do this and then later switch to Software SPI,  you will need to switch the SPI MODE back to SPI_MODE1.

I think you need to send this to the Marlin firmware guys, (open a ticket under https://github.com/MarlinFirmware/Marlin/issues) because I do not think Hardware SPI works for MAX31865 board on the SKR PRO board.  We are now getting beyond my ability to help.

GadgetAngel, 2020-10-29:
@daniagos Use (Appendix M, Analog Procedure using  2 PT100 amplifier boards) Technique number 1 and Technique number 2 that way you do not need to do the hardware hack on the SKR PRO V1.1 board.

GadgetAngel, 2020-10-29:
@daniagos I am going to test your setup when the TFT35 V2.0 arrives in two days.

daniagos, 2020-10-30:
thanks, in those days i make all the trials you suggest, thank you so much

oldman4U, 2020-10-31:
> @daniagos I am going to test your setup when the TFT35 V2.0 arrives in two days.  
> 
> 

Are you buying hardware to be able to help others, Angel!!??

daniagos, 2020-11-01:
hI, I commented the #define MAX6675_SS2_PIN TEMP_1_PIN, no changes of the problem.

i am still in hardware configuration, i tried looking for the spi mode in HAL_SPI.cpp(Marlin\src\HAL\STM32\HAL_SPI.cpp), but my SPI_MODE is SPI_MODE0 from the beginning of all .

I founded this in pins configuration:

#if SD_CONNECTION_IS(LCD)

  #define SD_DETECT_PIN                     PF12
  #define SDSS                              PB12

#elif SD_CONNECTION_IS(ONBOARD)

  // The SKR Pro's ONBOARD SD interface is on SPI1.
  // Due to a pull resistor on the clock line, it needs to use SPI Data Mode 3 to
  // function with Hardware SPI. This is not currently configurable in the HAL,
  // so force Software SPI to work around this issue.
  #define SOFTWARE_SPI
  #define SDSS                              PA4
  #define SCK_PIN                           PA5
  #define MISO_PIN                          PA6
  #define MOSI_PIN                          PB5
  #define SD_DETECT_PIN                     PB11

from what i understand i do not have the right configuration to let sd use hardware spi, so nothing will work, right?

GadgetAngel, 2020-11-01:
> hI, I commented the #define MAX6675_SS2_PIN TEMP_1_PIN, no changes of the problem.
>
> i am still in hardware configuration, i tried looking for the spi mode in [...]

@daniagos I am still working on it.  

Just got in the TFT screen and I just solder together my Adafruit MAX31865 board.  

I checked yesterday that the Adafruit MAX31865 board will work with the SKR PRO V1.1 and the TFT35 V2.0.  IT does. 

Today is when I work on hardware SPI.  

The reason I did it this way was to ensure that both my PT100 sensor and the Adafruit MAX31865 board are in working condition.  

I also have my SKR PRO V1.1 board  with BIQU TMC2209 in UART mode on X, Y, Z and E0.  

Let me do some more testing and see what I can figure out.

@oldman4U Yes, I did not own a TFT35 V2.0 and I wanted to try it since it does not have the dual mode capability and only has the touch screen.  Neat device.  I even got my logo to come up on the screen.  I like trying out new stuff that is why they call me "Gadget Angel".  I love gadgets!

daniagos, 2020-11-01:
I am trying with software spi mode for one adafruit max31865 right now, i  can't make the adafruit work neither in this case,i have always the same error of printer killed, i really don't know what's wrong with my setting

GadgetAngel, 2020-11-01:
@daniagos I hooked up **my MAX31865 board** as follows:

[image]

---

**I have the same exact board (SKR PRO V1.1) and drivers (TMC2209 in UART mode) as you do and the TFT (TFT35 V2.0) stays connected!**  

I defined my **SERIAL_PORT** and **SERIAL_PORT_2** as follows (remember I just got my TFT35 V2.0 so I compiled the TFT35 screen firmware and updated the firmware for the screen before this test to ensure the screen was not getting into the issue, I highly recommend that you compile your TFT35 firmware and update the firmware on you TFT35 V2.0 screen):

[image]

Here is **a video of the bootup process**, you can see I never touch the TFT screen and the TFT screen will after about 30 seconds start displaying the temperature for the PT100 sensor:

https://drive.google.com/file/d/1OGdkTxWqfj61Y-6039-VQrJ3G60MO_hF/view?usp=sharing

I connected **pronterface** (http://kliment.kapsi.fi/printrun/) to my board via the USB connection and found I was getting a lot of "Temp measurement errors" but the SKR PRO V1.1 board **was still able to read the PT100 sensor temperature values**.  If you look here https://github.com/MarlinFirmware/Marlin/issues/5723#issuecomment-557290199, Marlin knows of this issue but it did not make the TFT disconnect from the SKR PRO V1.1 board.  

Here is are pictures of my hook up:

Green wire is CS of Adafruit MAX31865 board
Purple wire is SDI of Adafruit MAX31865 board
Brown wire is SDO of Adafruit MAX31865 board
White wire is CLK of Adafruit MAX31865 board
Black wire is GND of Adafruit MAX31865 board
Red wire is VIN of Adafruit MAX31865 board

[image]

[image]

[image]

[image]

[image]

To handle the errors you can just set the SERIAL_PORT to 1 and disable SERIAL_PORT_2.  You will still be getting  "Temp measurement errors" but you will not see them on the TFT screen.  **Marlin needs to fix this problem** but as you can see from https://github.com/MarlinFirmware/Marlin/issues/5723#issuecomment-557290199 they are aware that the issue.  We need to really report this as a bug in Marlin.

GadgetAngel, 2020-11-02:
> I am trying with software spi mode for one adafruit max31865 right now, i can't make the adafruit work neither in this case,i have always the same error of printer killed, i really don't know what's wrong with my setting

Check the Adafruit MAX31865 board.  Do you have the MAX31865 board prepared properly? 

Check out this website that shows where you need to solder the jumpers on the MAX31865 board to indicate you will use a PT100 2-Wire configuration: https://voron.dozuki.com/Guide/How+to+Use+a+Pt100+Thermistor+w-+Skr+Boards/73?lang=en

Does your PT100 show approximately 110 Ohms when you use an Ohm meter?

If the PT100 sensor is not showing approximately 110 Ohms at room temperature and the Adafruit MAX31865 board is not prepared properly the Adafruit MAX31865 board will not work in Software SPI or Hardware SPI Mode.  

Double check all your wires for continuity, to ensure that they are not broken.  Double check how you wired up the Adafruit MAX31865 board to the SKR PRO V1.1 board.

Have you looked at Appendix O (https://github.com/GadgetAngel/SKR-PRO-V1.1-Stepper-Driver-Jumper-Configuration-Manual/tree/master/CURRENT-Manual) of my manual and followed the Manual's instructions?  You can download the whole manual in zipped file : https://drive.google.com/file/d/1O82o16ouSXPkhGySXl9ZE93H82mqJ-9m/view

If the Adafruit MAX31865 board does not work in Software SPI then you have a problem with your setup.  Sorry I can not come to your house to debug the hook up problem but this is one you will have to find yourself.  Good luck.  Let me know if you find the error with your hook up.

daniagos, 2020-11-02:
IT WORKSSSSSSSS

O LOVE YOU MAN!!!

I WAS USING STUPID PT100 WITH 25M ohm, i put some other pt100 i have at home and now the tft works, i still have some errors from pt100, but the display now shows it, i have error temperature from max, but it still show me the room temperature.

can i use a second pt100 connected as this one, changing cs pin for temperature1?

do you have suggestions on how to connect the board to the pc?
i installed drivers, but i can't really connect it to proterface or repetier or cura.

i set the baud rate to skr board, com port in pc setting and in the program, what am i doing wrong?
i am using the blue cable that bigtreetech gave me

GadgetAngel, 2020-11-02:
> IT WORKSSSSSSSS
>
> O LOVE YOU MAN!!!
>
> I WAS USING STUPID PT100 WITH 25M ohm, i put some other pt100 i have at home and now the tft works, i still have some errors from pt100, but the display now shows [...]

The SKR PRO V1.1 by design does not do firmware updates via the USB cable.  You MUST use the SD card and manually place the SD card into the SKR PRO V1.1 board's SD card reader and reboot the board for the firmware to update.  I do not know why the SKR PRO and GTR V1.0 boards are made this way.  But you can not update firmware over the USB cable like you can with the SKR V1.3 or SKR V1.4 boards.  Sorry that is just the way it is.

Yes, You should be able to now connect up the second PT100 (110 Ohm) to your second Adafruit MAX31865 board and use Y Jumper cables to connect up the SPI lines, 5V and GND lines but use a seperate wire for the CS line on the second MAX31865 board.

daniagos, 2020-11-02:
All works, i love you man, thanks for all !!

GadgetAngel, 2020-11-08:
@daniagos If you tried hooking up two MAX31865 via Hardware SPI you noticed that the second Adafruit MAX31865 will not show the correct temperature on TEMP_SENSOR_1.  

To fix this issue and to also allow 2 Adafruit MAX31865 boards to run is Software SPI, and to fix the issue with "temp errors" that the Marlin was producing erroneously when an Adafruit MAX31865 board was attached to the SKR PRO V1.1, I have submitted a Pull Request to fix these issues. 

If you can not wait for Marlin to incorporate these fixes here is the Pull Request and a link to my branch of Marlin that currently allows two Adafruit MAX31865 boards to work without issue on the SKR PRO V1.1 board.

PR (Pull Request) I have submitted is at (https://github.com/MarlinFirmware/Marlin/pull/20074#issue-517368163).  This PR uses my branch of Marlin that fixes the issue.

My branch of Marlin that fixes this issue is located at https://github.com/GadgetAngel/Marlin/tree/Marlin-Fix-2AdafruitMAX31865

You can download my Marlin-Fix-2AdafruitMAX31865 branch and try my version if you want to.  My Marlin-Fix-2AdafruitMAX31865 branch was built on Marlin bug-fix-2.0.x  that existed on 11/8/2020 (today).

GadgetAngel, 2020-11-12:
@daniagos If you want to hookup two Adafruit MAX31865 board via Hardware SPI the boards will no longer produce 'error temp measurement errors'. Just download the latest bugfix-2.0.x and the issue has been corrected in the latest branch of bugfix!

GadgetAngel, 2020-11-12:
@daniagos Please close this issue because the latest bugfix-2.0.x resolves all the issues

daniagos, 2020-11-12:
Ok, thanks you so much for your precious help!

GadgetAngel, 2020-11-12:
> Ok, thanks you so much for your precious help!

You are welcome, I was happy to help and together we found a bug in Marlin and got it fixed.

daniagos, 2020-12-15:
Hi, i set the first pt100 and it is working as yours, but i can't make the second pt100 work, it disconnect from tft, and i can't connect the board to the printer.

If i set the second termistor to 1 instead than -5 i can connect to the board (there still are errors in max amplifier) tft work, and i can connect to the board, but i can't see the second termistor temperature in the screen (as if i set only one extruder).

What can i look for?

GadgetAngel, 2020-12-19:
@daniagos I found your last comment that said you took the second one off and placed it on your board as TEMP_SENSOR_0.  That proves to me that your PT100 and second MAX31865 both are working.  

Now I want you to put the first one back on as TEMP_SENSOR_1 or TEMP_1_PIN. 

Now zip up your Marlin folder (with platformio.ini file included) on to your free google drive.  Turn on the link to share the  zip folder and place the link in this thread.  I need to see the firmware to help you debug the issue.

daniagos, 2020-12-19:
https://drive.google.com/file/d/1WEwuyBfSX1LRO_mR0DB9MqsU1oND-LA3/view?usp=sharing

This is the link for the file.
i noticed that with a single max defined while preheating the cartridge i have no problem and shows the temperature correctly,, so i assumed it was working, but while i try homing the tft shows me errors and can't home the z

GadgetAngel, 2020-12-23:
@daniagos  I look at this later today after I wake up.  I have been awake all night but I will look at it.

GadgetAngel, 2020-12-24:
@daniagos 

> @daniagos If you want to hookup two Adafruit MAX31865 board via Hardware SPI the boards will no longer produce 'error temp measurement errors'. Just download the latest bugfix-2.0.x and the issue has been corrected in the latest branch of bugfix!

### @daniagos PLEASE use the latest bugfix branch of MARLIN.  You are using the release branch and the release branch will produce all the error messages.  So try downloading the latest bugfix branch.  Try it out and if it still does not work reupload that Marlin folder for me to look at.

daniagos, 2020-12-26:
ok, now it seem all working good, except i can't set from tft a temperature over 275.

In both board setting and display setting it is set to 550 degrees the max temperature, with pronterface i can heat the cartridge to 550 without problems, but no from tft, why?

GadgetAngel, 2020-12-26:
@daniagos You problem now resides with the TFT firmware NOT Marlin firmware.

I am assuming in Marlin's configuration.h file you have the following:
```
#define HEATER_0_MAXTEMP 550
#define HEATER_1_MAXTEMP 550
```

The TFT firmware is limiting you on the max temperature setting.  

Go to [https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and download the source code for the TFT firmware.  Here is a good YouTube video on how to update the TFT firmware: https://youtu.be/NmreTXdyqhk.  Please watch it.

Look at C:\[your computer location]\BIGTREETECH-TouchScreenFirmware\Copy to SD Card root directory to update\config.ini

In **config.ini file** the following is stated:
```
# IMPORTANT NOTE:
#   Ensure that the following options are enabled in Marlin firmware
#
# General options:
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support (Un)Load menu:
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)
#
```
In **config.ini file**, around line 282, you will see the following:
```
#### Bed / Extruder / Chamber Maximum Temperatures
# Format: [max_temp: T0:<max temp> T1:<max temp> T2:<max temp> T3:<max temp> T4:<max temp> T5:<max temp> BED:<max temp> CHAMBER:<max temp>]
#
# Unit: [°C]
#
max_temp:T0:275 T1:275 T2:275 T3:275 T4:275 T5:275 BED:150 CHAMBER:60
```
**Try changing T0 and T1 to 550 and see if that solves the problem.**  Recompile the TFT firmware in VScode and upload to 8 GB SD card and place the 8GB SD card into the SD card reader on the TFT display. Reboot the printer and the TFT firmware update will occur.

If changing T0 and T1 in the TFT firmware does not work, then you need **to post this issue over at** https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues page because your problem is NOT due to the Marlin firmware but due to the TFT firmware for the screen.

oldman4U, 2020-12-26:
First class help like always. 👍🏻

GadgetAngel, 2020-12-27:
Here is the latest SUMMARY or GUIDE:

# GUIDE:

1. You CAN NOT mix PT100/PT1000 and thermocouples:
 
If you have only one hotend then you can have the following:
- Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0
- PT100 (-5) on E0
- PT1000 (-5) on E0
- Thermocouple (-3) on E0 

If you have two hotends then you can have the following:
- Thermistor (that is an ADC Temp Sensors [uses a pure analog I/O pin]) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
        Examples of this are:
               - PT100 (uses table 20 or 21) on E0 and Thermocouple (-4 - use pure analog pin) on E1
               - Thermocouple (-4 [pure analog I/O pin]) on E0 and PT100 (uses table 20 or 21) on E1
              - PT1000 (uses table 1047) on E0 and PT100 (uses table 20 or 21) on E1
              - too many to enumerate (etc.)
- PT100 (-5) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
- PT1000 (-5) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure  analog I/O pin]) on E1
- Thermocouple (-3) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
- PT100 (-5) on E0 and PT100 (-5) on E1 ====>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT1000 (-5) on E0 and PT1000 (-5) on E1 ====>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT100 (-5) on E0 and PT1000 (-5) on E1 ===>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT100 (-5) on E0 and PT1000 (-5) on E1 IS ALLOWED ===>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- Thermocouple (-3) on E0 and Thermocouple (-3) on E1===>>> can use either Software SPI or Hardware SPI.  They both work for two MAX31855 boards and two thermocouples

If you have two hot-ends then you **CAN NOT have the following**:
- PT100(-5) on E0 and a Thermocouple (-3) on E1 IS NOT ALLOWED
- PT1000 (-5) on E0 and a Thermocouple (-3) on E1 IS NOT ALLOWED
- Thermocouple (-3) on E0 and a PT100 (-5) on E1 IS NOT ALLOWED
- Thermocouple (-3) on E0 and a PT1000 (-5) on E1 IS NOT ALLOWED
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and PT100 (-5) on E1 IS NOT ALLOWED ***
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and PT1000 (-5) on E1 is NOT ALLOWED ***
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and Thermocouple (-3) on E1 is NOT ALLOWED ***

*** If you only have ONE PT100 (or PT1000 or Thermocouple) on your printer and you have two hotends then in Marlin you
must use TEMP_SENSOR_0 as the PT100 (or PT1000 or Thermocouple) even if you have the PT100 (or PT1000 or Thermocouple) hooked up to the second extruder.  Just override your PIN definitions to have TEMP_0_PIN point to where the PT100 (or PT1000 or Thermocouple) is located. 

---

2. Marlin still uses the old naming scheme of the 20x4 display `REPRAP_DISCOUNT_SMART_CONTROLLER` in the pins-file for the `software SPI pins names`:
The EXP1 and EXP2 ports are used for display screens.  EXP1 contains a HARDWARE SPI bus while EXP2 contains a Software SPI bus.  The EXP1 hardware SPI bus signal names are obvious in the pins-file but the EXP1 software SPI bus signal names are **NOT obvious**. Here is the EXP1 software SPI bus signal names used in pins-file with their corresponding SPI software functions:
```
LCD_PINS_D4 is software SPI signal SCK for EXP1
LCD_PINS_ENABLE is the data line or software SPI signal MOSI for EXP1
LCD_PINS_RS is Register select (command/data) or software SPI signal SS for EXP1
```

---

3. Translation table to be used (or Thermistor table) depends on the power you supply to the amplifier board and the ADC reference voltage of the MCU:

- 1. Use table 20: When the power supply for the PT100/PT1000 Amplifier board is equal to the ADC reference voltage for the MCU.
--  For SKR PRO V1.1/V1.2, SKR V1.3/V1.4/V1.4 Turbo, GTR V1.0, M5 V1.0 all use 3.3 VDC as the ADC reference voltage for the MCU.  Therefore use table 20 when the PT100/PT1000 Amplifier board is powered by 3.3VDC
- 2. Use table 21: When the power supply for the PT100/PT1000 Amplifier board is different to the ADC reference voltage for the MCU. 
-- The E3D PT100 Amplifier board wants 5V as its power supply. But the GTR V1.0 (like the SKR PRO) uses 3.3V as the ADC reference voltage for the MCU.  If you use 5VDC power to the PT100 amplifier board, the amplifier can output a maximum signal with 5VDC.  To protect the GTR V1.0  board (or the SKR PRO board), use a 3.6V Zener Diode (reverse biased hookup) between the output signal of the amplifier board and ground (to protect the MCU from getting fried from a 5V input).

[image]

On the GTR V1.0 board, here is a list of PINS that you will need to prevent negative current injection. We prevent negative current from flowing into the MCU by using a Schottky diode between the output signal of the amplifier board and ground.  I do this as preventive measure. It protects the MCU from getting fried.

[image]
[image]

[... cut ...]
     #define MAX6675_SS2_PIN           TEMP_1_PIN    //for second Thermocouple 
     #define MAX6675_SCK_PIN           PI1
     #define MAX6675_DO_PIN            PI2
#endif
```
**In configuration.h**:
`	set TEMP_SENSOR_0 to -3` and `set TEMP_SENSOR_1 to -3`

**NOTE: the first thermocouple is wired to the ONBOARD K_Type Thermocouple (KTEM) connector.  That KTEM connector by default uses PH9 as its Chip select pin.  The first thermocouple is not shown in the wiring diagram.**

Here is a wire diagram for this situation:

[image]

---

END OF GUIDE

daniagos, 2020-12-28:
You were right, i didn't use the bugfix of marlin, honestly idk why, i feel so stupid..

Now the situation is 
pt100 works (both)
sensorless homing works for both x and y, 
bltouch has strange issues:
it randomly trigger itself in the air as if it touches something, then try the second touch and obviously it fails and printer got halted.

sometimes it need 2/3 seconds to trigger itself, sometimes 20 seconds if the bed is low enough.

The bltouch itself works good except for this, it pull down and up the pin with the command from pronterface, and i can see it switching from triggered to open circuit with m119 command.

For the bltouch uncommented:
In configuration.h 
#define BLTOUCH

in configuration.adv.h
 #define BLTOUCH_DELAY 500.
 
 
 in common.pins file i switched the z_min endstop to PA2, which is the pin near the servo one (mainly for a cleaner wiring than real necessity).
 in standard configuration.h
  #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
  is enabled by default
  
  What can i check?

daniagos, 2020-12-28:
https://drive.google.com/file/d/1EPR14O2a6--NThF2DB0I4DMhMOFRZEi3/view?usp=sharing
https://drive.google.com/file/d/1nhShG67IcBj4n0OiuGTBb4F5jTqs9pDf/view?usp=sharing

This is how it acts

daniagos, 2020-12-30:
Ok, for the bltouch there were noises in the printer, i solved with a capacitor between white and black wires, now finally all works fine, thanks @GadgetAngel, you really saved my building!

closed 2020-12-30 by daniagos