bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #177: During print, the printer pauses because it thinks it's out of filament but it's not

bobjones113, 2020-10-30:
Not sure if this is a bug or a feature(?).  I've had a few failed prints now because the printer thinks the filament has run out but it hasn't.  I have a filament sensor(just a simple switch type) and that is always connected.  In the terminal it says the filament had run out, paused waiting for user. I send M108 to resume because I can see there is still filament and also can force filament through the nozzle. It will then heat up the tool again to temp and sits and says paused for user again.  I send another M108 but nothing happens.  I send ten more and nothing happens.  The temperature for the bed and tool stay at temp but no progression.  I have to power cycle the board to continue and fail the print at that point.  I'll post my configs soon.  Any help would be appreciated! Thanks in advance.

oldman4U, 2020-10-31:
Sounds like a problem with the switch or the cable. Remove them and see if it still happens. Depending on your software setup you eventually have to shorten the pins on the mainboard.

bobjones113, 2020-10-31:
I'm thinking it's a board leveling issue.  I think the pressure builds up when the build plate is too high and doesn't allow filament to pass and causing it to think there isn't any filament?

closed 2020-10-31 by bobjones113