bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #178: Error Communicating with SKR PRO V1.1

TechnicalWorld2016, 2020-11-04:
I had uploaded marlin 2.0 firmware on my Bigtree tech SKR PRO V1.1 board successfully and connected with pronterface and checked the motor orientations. Next day I am about to do the PID tuning, but the board was not connecting to my PC, no COM port is detected my PC, I have tried to connect with other laptops in my office, but there is no result. The board is getting power on and taking input supply. When I try to press ENDSTOP manually, the respective led near stepper driver is blinking, resulting in normal functionality. But I am unable communicate with the board neither with Pronterface nor with TFT 2.8. I have tried changing the serial_port, Serial_port 2 in marlin and tried to upload the firmware, but no change in the result, finally I am unable to cummunicate with the Board.

If anyone have a solution, please do reply.

AnHardt, 2020-11-04:
Concentrate to the thermal settings. Having them wrong causes a KILL what causes the loss of communication. Messing with the serials will not have any positive effect.

oldman4U, 2020-11-04:
Hi.

In case something works properly but does not do so without changing anything - a hardware issue is most likely.

For USB and TFT connection of this board, you have to activate serial port -1 and 1.

Maybe attaching the Config files could help, but like I said. Most likely a hardware issue and then I would check the fuses first.

TechnicalWorld2016, 2020-11-05:
Hi,
thank you for your suggestions.

Even i thought some hardware issues, the board doesn't have any traces of burns. It seems fine.

oldman4U, 2020-11-05:
Maybe I am wrong or the provided information is not leading into the right direction. Happened to me a few times now, that I changed something which caused an issue somewhere else and it took some time to understand the relationship.

Btw. For this SKR board, the ESP01 based WIFI module replaced the USB connection completely for me.

Please do not forget to close this ticket in case you do not need it anymore.

Thank you

oldman4U, 2020-12-22:
Please help the community and close the ticket once you do not need it anymore. Thank you

AMarinR, 2021-02-07:
hello, I have exactly the same problem, yesterday everything worked fine, and today the board does not connect, not with the pc via usb, not with the screen, nor with the wifi.

any ideas?

AMarinR, 2021-02-08:
I have solved it by flashing the bootloader

SwiftNick, 2021-02-11:
Try enabling DTR/DSR in your comms settings

TechnicalWorld2016, 2021-02-15:
Really great to know. Is it possible to show how to Boot load it pls?

jim-thompson, 2021-04-03:
> I have solved it by flashing the bootloader

@AMarinR How did you flash the bootloader and where did you find the bootloader binary?

ElekKecskemeti, 2021-05-17:
I had the same problem. I also found a description and bootloader for my motherboard here. That solved it for me. I hope it helps you too. 

[link1](https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/Guide%20ST-LINK%20Utility%20on%20Windows%2010%20for%20ST-Link-V2-Clones%20for%20accessing%20STM32%20Processors.pdf)
[link2](https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders)

AMarinR, 2021-05-17:
I followed the instructions in this video

https://youtu.be/7O_ZOdnM0SU

TechnicalWorld2016, 2021-08-04:
Hi 
every one. Thank you very much for your replies so far.

I have purchased an st-link and tried connecting to the Board, All the connections were correct.

I am unable to read or write i.e., unable to access the memory of the board but my board is getting connected to the st-link utility.

I have checked the readout protection, It was set to 1, and it is protected status.

As 1 can be  replaced 0 as mentioned in the community.. I have tried changing 1 to 0 and tried connecting once again but the same problem repeats again ..  And the Readout protection is not changed to 0, it is still 1 as previous. 

But as per community guides and manual 1 can be overwrite to 0, Only read out protection 2 is not able to overwrite to 0 or 1 .
but 1 can be overwritten to 0.

But in my case I am unable to do it.
And, finally I am unable to upload the boot-loader again.

Someone please help me in this regard.

Thank you