bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #179: [GUIDE] to using PT100 or PT1000 or Thermocouples with SKR PRO V1.x boards or GTR V1.0 boards

GadgetAngel, 2020-11-12:
# GUIDE:

1. You CAN NOT mix PT100/PT1000 and thermocouples:
 
If you have only one hotend then you can have the following:
- Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0
- PT100 (-5) on E0
- PT1000 (-5) on E0
- Thermocouple (-3) on E0 

If you have two hotends then you can have the following:
- Thermistor (that is an ADC Temp Sensors [uses a pure analog I/O pin]) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
        Examples of this are:
               - PT100 (uses table 20 or 21) on E0 and Thermocouple (-4 - use pure analog pin) on E1
               - Thermocouple (-4 [pure analog I/O pin]) on E0 and PT100 (uses table 20 or 21) on E1
              - PT1000 (uses table 1047) on E0 and PT100 (uses table 20 or 21) on E1
              - too many to enumerate (etc.)
- PT100 (-5) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
- PT1000 (-5) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure  analog I/O pin]) on E1
- Thermocouple (-3) on E0 and Thermistor (that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E1
- PT100 (-5) on E0 and PT100 (-5) on E1 ====>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT1000 (-5) on E0 and PT1000 (-5) on E1 ====>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT100 (-5) on E0 and PT1000 (-5) on E1 ===>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- PT100 (-5) on E0 and PT1000 (-5) on E1 IS ALLOWED ===>>> **issue fixed in the latest bugfix-2.0.x branch ONLY**. You can use Software SPI or Hardware SPI for two (2) Adafruit MAX31865 boards.
- Thermocouple (-3) on E0 and Thermocouple (-3) on E1===>>> can use either Software SPI or Hardware SPI.  They both work for two MAX31855 boards and two thermocouples

If you have two hot-ends then you **CAN NOT have the following**:
- PT100(-5) on E0 and a Thermocouple (-3) on E1 IS NOT ALLOWED
- PT1000 (-5) on E0 and a Thermocouple (-3) on E1 IS NOT ALLOWED
- Thermocouple (-3) on E0 and a PT100 (-5) on E1 IS NOT ALLOWED
- Thermocouple (-3) on E0 and a PT1000 (-5) on E1 IS NOT ALLOWED
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and PT100 (-5) on E1 IS NOT ALLOWED ***
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and PT1000 (-5) on E1 is NOT ALLOWED ***
- Thermistor ((that is an ADC Temp Sensor [uses a pure analog I/O pin]) on E0 and Thermocouple (-3) on E1 is NOT ALLOWED ***

*** If you only have ONE PT100 (or PT1000 or Thermocouple) on your printer and you have two hotends then in Marlin you
must use TEMP_SENSOR_0 as the PT100 (or PT1000 or Thermocouple) even if you have the PT100 (or PT1000 or Thermocouple) hooked up to the second extruder.  Just override your PIN definitions to have TEMP_0_PIN point to where the PT100 (or PT1000 or Thermocouple) is located. 

---

2. Marlin still uses the old naming scheme of the 20x4 display `REPRAP_DISCOUNT_SMART_CONTROLLER` in the pins-file for the `software SPI pins names`:
The EXP1 and EXP2 ports are used for display screens.  EXP1 contains a HARDWARE SPI bus while EXP2 contains a Software SPI bus.  The EXP1 hardware SPI bus signal names are obvious in the pins-file but the EXP1 software SPI bus signal names are **NOT obvious**. Here is the EXP1 software SPI bus signal names used in pins-file with their corresponding SPI software functions:
```
LCD_PINS_D4 is software SPI signal SCK for EXP1
LCD_PINS_ENABLE is the data line or software SPI signal MOSI for EXP1
LCD_PINS_RS is Register select (command/data) or software SPI signal SS for EXP1
```

---

3. Translation table to be used (or Thermistor table) depends on the power you supply to the amplifier board and the ADC reference voltage of the MCU:

- 1. Use table 20: When the power supply for the PT100/PT1000 Amplifier board is equal to the ADC reference voltage for the MCU.
--  For SKR PRO V1.1/V1.2, SKR V1.3/V1.4/V1.4 Turbo, GTR V1.0, M5 V1.0 all use 3.3 VDC as the ADC reference voltage for the MCU.  Therefore use table 20 when the PT100/PT1000 Amplifier board is powered by 3.3VDC
- 2. Use table 21: When the power supply for the PT100/PT1000 Amplifier board is different to the ADC reference voltage for the MCU. 
-- The E3D PT100 Amplifier board wants 5V as its power supply. But the GTR V1.0 (like the SKR PRO) uses 3.3V as the ADC reference voltage for the MCU.  If you use 5VDC power to the PT100 amplifier board, the amplifier can output a maximum signal with 5VDC.  To protect the GTR V1.0  board (or the SKR PRO board), use a 3.6V Zener Diode (reverse biased hookup) between the output signal of the amplifier board and ground (to protect the MCU from getting fried from a 5V input).

[image]

On the GTR V1.0 board, here is a list of PINS that you will need to prevent negative current injection. We prevent negative current from flowing into the MCU by using a Schottky diode between the output signal of the amplifier board and ground.  I do this as preventive measure. It protects the MCU from getting fried.

[image]
[image]

The below wiring diagram for PT100 using Analog ADC input using 5V DC for the PT100 amplifier board but the MCU board (GTR V1.0) uses 3.3 VDC ADC reference voltage, therefore the Thermistor table to use for this is Table 21:

[image]

The below wiring diagram for PT100 using Analog ADC input using 3.3 VDC for the PT100 amplifier board but the MCU board (GTR V1.0 board) uses 3.3 VDC ADC reference voltage, therefore the Thermistor table to use for this is Table 20 (see the process data sheet and you will find that PF10 need protection against negative current injection,https://www.st.com/resource/en/datasheet/stm32f407ig.pdf#page=114):

[... 497 lines cut ...]
- If you have a **MCU like the GTR V1.0 board**, which already has a MAX31855 built into the board (I will call this ONBOARD_MAX31855), then to use the **GTR V1.0 board with an external MAX31855** you MUST use the Clock Line (SCK) and Data Output Line (DO)  of the ONBOARD_MAX31855 PINs defined in **pins_BTT_GTR_V1_0.h** file. This would mean you have to tie the second (external) MAX31855 amplifier board to the M5 connector of the GTR V1.0 board so that you gain access to the DO and SCK lines of the ONBOARD_MAX31855 chip.

For example, here is how you would **setup using two thermocouples with the GTR V1.0 board**:
```
#define TEMP_0_PIN   PH9    //CS for ONBOARD MAX31855 chip, See NOTE below
#define TEMP_1_PIN   PH13  //CS for external MAX31855 amplifier board
#ifndef MAX6675_SS_PIN
     #define MAX6675_SS_PIN            TEMP_0_PIN     //for first Thermocouple
     #define MAX6675_SS2_PIN           TEMP_1_PIN    //for second Thermocouple 
     #define MAX6675_SCK_PIN           PI1
     #define MAX6675_DO_PIN            PI2
#endif
```
**In configuration.h**:
`	set TEMP_SENSOR_0 to -3` and `set TEMP_SENSOR_1 to -3`

**NOTE: the first thermocouple is wired to the ONBOARD K_Type Thermocouple (KTEM) connector.  That KTEM connector by default uses PH9 as its Chip select pin.  The first thermocouple is not shown in the wiring diagram.**

Here is a wire diagram for this situation:

[image]

---

END OF SUMMARY

arcmal22, 2020-12-13:
HOW TO SETUP MAX31865 TO HEATED BED....

arcmal22, 2020-12-13:
I already followed the guide and it was success on my heater_0....but now i want to set for my heatbed using the ultimaker pt100 bed....any guide to do using max31865??.....

GadgetAngel, 2020-12-13:
@arcmal22 What MCU board are you using?  Please zip up your Marlin folder (ensure you include the platformio.ini file) and place the zipped folder on you free google drive.  Turn on the share link for the zipped folder on your google drive and place that link here so I can download your Marlin firmware.  This will give me the ability to look at how you are trying to set the MAX31865 up.

Then I can investigate how to use a PT100 on the HEATED BED.

GadgetAngel, 2020-12-13:
@arcmal22 from looking at the Marlin firmware code, using a MAX31865 on the Heated bed (TEMP_SENSOR_BED set to -5) is **not** an option with Marlin firmware at this time.  If you want this option please put in a request for a new feature.

fox631, 2021-01-18:
GadgetAngel, I am using SKR MINI E2 V1.2 and BIGTREE_TFT24_V1_1. Can I use your SKR MINI E3 V2.0 diagram to connect the MAX31865? What will I need to change in your settings?

GadgetAngel, 2021-01-18:
@fox631 let me take a look at the SKR MINI E2 V1.2 and the TFT24 V1.1 and see if I can find a way to do it.  I will get back to you.

fox631, 2021-01-19:
Okay, thank you, I'll be waiting.

GadgetAngel, 2021-01-19:
@fox631 Here is the diagram:

[image]

Here is a link to my google drive which has the diagram at the highest resolution: https://drive.google.com/file/d/1N-hZbQFYBljPhR0FeSUBp3NPB6rK9hTL/view?usp=sharing

fox631, 2021-01-20:
**GadgetAngel**, thanks for the diagram. When I compile the firmware for the software SPI, I get the error:
Marlin\src\module\../libs/softspi.h:85:70: error: cannot convert 'bool' to 'WiringPinMode' for argument '2' to 'void pinMode(uint8, WiringPinMode)'

He does not like the file Marlin \ src \ libs \ softspi.h, this line:
inline void fastPinMode (uint8_t pin, bool mode) {pinMode (pin, mode); }

I used Marlin_bugfix_2.0.x firmware

GadgetAngel, 2021-01-20:
> **GadgetAngel**, thanks for the diagram. When I compile the firmware for the software SPI, I get the error:
> Marlin\src\module../libs/softspi.h:85:70: error: cannot convert 'bool' to 'WiringPinMode' [...]

@fox631 Please take your Marlin folder and place it in a zip file by using winzip or your OS' zip utility.  Ensure that the platformio.ini file is included in the Marlin folder. Upload the zipped Marlin folder onto your free google drive. Ensure that you turn on "Shared Link" for the zipped Marlin folder on your google drive.  Place the link here in this thread.  This way I can download your Marlin folder and try to compile from your source code and see the problem you are talking about

GadgetAngel, 2021-01-20:
@fox631 You found a bug in Marlin, please report it here: https://github.com/MarlinFirmware/Marlin/issues

I have already reported enough problems with the MAX31865 code, so please you need to report the issue.  Just tell them like you told me in the write up. You can also point to this thread.  But I confirm that your problem is a bug in the code.

fox631, 2021-01-21:
**GadgetAngel** Thanks for the answer. I will report a bug. I will try all the options. If this is Marlene's mistake, then there is no need to send the firmware?

GadgetAngel, 2021-01-21:
> **GadgetAngel** Thanks for the answer. I will report a bug. I will try all the options. If this is Marlene's mistake, then there is no need to send the firmware?

No no need to place your zip file on your google drive.     I has another idea I am investigating.  Let me check it out. I may have a work-around.  Have your reported the bug yet?

GadgetAngel, 2021-01-21:
@fox631 Please report the bug here: https://github.com/MarlinFirmware/Marlin/issues

Here is **a work-around that solves your problem!**  You will need to update the configuration for your specific printer.  I did the configuration for my Ender 3 Pro.

The work-a-round is to use my Pull Request (https://github.com/MarlinFirmware/Marlin/pull/20447) branch of Marlin.  Here is how you do that:

1. go to my github repository: https://github.com/GadgetAngel/Marlin-Fix-Temperature-Issues-bugfix-branch
2. Instead of **main** branch, hit the down arrow and select: **SKR-MINI-E3-V1.2-fix** branch
3. now download the SKR-MINI-E3-V1.2 branch. **You should be downloading the file: Marlin-Fix-Temperature-Issues-bugfix-branch-SKR-MINI-E3-V1.2-fix.zip**
4. unzip the folder to an area on your disk drive
5. open this folder up in VScode
6. adjust the parameters for your printer (steps_per_mm,feedrates,etc)
7. The configuration is already using the SKR MINI E3 V1.2 board and has an additional flag in platformio.ini to tell Marlin to use my library for the MAX31865 temperature sensor. So leave the platformio.ini file alone.
8. This branch I created is just like my PULL REQUEST branch that I submitted to Marlin on Dec 12, 2020.  So I submitted this PR a month ago.  I do not understand why it is not being incorporated, but Marlin developers **must have their reasons**.
9. **My branch changes the code in temperature.cpp file** to run a user defined library for the MAX31865  and **stops Marlin from manipulating a PIN (MAX6675_SS_PIN) that is not need when using the MAX31865 library**.  Since the manipulation of that pin (MAX6675_SS_PIN) does not occur in my PR, my PR fixes the ERROR you were getting when compiling the code for the SKR MINI E3 V1.2 board. You can read about the PR (pull request) here: https://github.com/MarlinFirmware/Marlin/pull/20447
10. The branch you downloaded is equal to the Marlin bugfix-2.0.x branch as of today.  
11. I compiled it on my computer and I did not get any errors.  

[image]

I hope this helps.
GadgetAngel

fox631, 2021-01-21:
> > **GadgetAngel** I reported a bug. I am very glad to have a new message from you, thank you very much. But I can only check it tomorrow, it's already night with us. I will report the results.

fox631, 2021-01-23:
@GadgetAngel, I did it. The sensor is working. I made a software connection, according to your diagram, and with your firmware "Marlin-SKR-MINI-E3-V1.2-fix". For some reason, the firmware turned out to be 164kb in size. When I compiled with Marlin 2.0.7.2, the size was 251kb. Also, I had to disable BABYSTEP_ZPROBE_GFX_OVERLAY and LCD_BED_LEVELING in the configuration files. Doesn't compile with them when #define CR10_STOCKDISPLAY is commented out.
I want to try hardware SPI. Do I need to comment out the #define CR10_STOCKDISPLAY line? After all, EXP1 is not used with hardware sleep.

GadgetAngel, 2021-01-24:
> @GadgetAngel, I did it. The sensor is working. I made a software connection, according to your diagram, and with your firmware "Marlin-SKR-MINI-E3-V1.2-fix". For some reason, the firmware turned out to be 164kb in size. When I compiled with Marlin 2.0.7.2, the size was 251kb. Also, I had to disable BABYSTEP_ZPROBE_GFX_OVERLAY and LCD_BED_LEVELING in the configuration files. Doesn't compile with them when #define CR10_STOCKDISPLAY is commented out.
> I want to try hardware SPI. Do I need to comment out the #define CR10_STOCKDISPLAY line? After all, EXP1 is not used with hardware SPI.

👍🏻 

@fox631 If you want to use hardware SPI, then you can go ahead and use the EXP1 port and TFT port. you will have to find another free digital I/O line that does not have a pull up resistor. I think PWR-DET is available.  

Please keep in mind that hardware spi uses the same lines as your micro-SD card reader.  So always remove the power and ground from the MAX31865 board (or put in a switch to disconnect the lines) when you use the micro-SD card for reading or writing. Like when you do a firmware upgrade.

Glad I was able to help.

fox631, 2021-01-24:
@GadgetAngel, thank you very much for your help, I could not have done it without you. I will try hardware SPI later, and post the results. About the micro-SD card with hardware SPI, I remember. I don't need it for printing, there is a micro-SD slot on the display. The main thing is not to forget about it while flashing the board. Best wishes to you.

Pirat-1989, 2021-05-01:
Good afternoon. Which option to use, and is there such an option for this case? I hardly understand programming yet
[https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/197](url)

Pirat-1989, 2021-05-01:
Good afternoon. Which option to use, and is there such an option for this case? I hardly understand programming yet
[https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/197#issue-863087436](url)

GadgetAngel, 2021-05-01:
> Good afternoon. Which option is better to use for the MAX6675 with SKR PRO 1.2, and is there even such an option for the MAX6675 with K-type? I hardly understand programming yet
> [https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/197#issue-863087436](url)

There is an option to use the MAX6675 I can do a GUIDE but I will need at least a week before I can start to work on the GUIDE due to other responsibilities.

I will do one similar GUIDE like I did on the MAX31865 board.  I will use the SKR PRO V1.2 board and the MAX6675 with K-Type Thermocouple.  I will let you know when it is done.  I will publish it on my github repository found : https://github.com/GadgetAngel

jinhong-, 2021-07-13:
Hey @GadgetAngel, great comprehensive guide. I am looking at integrating PT100+MAX31865 using software SPI on the SKR E3 TURBO. I have a few questions I hope you can help me with:
1. Is my understanding correct that for software SPI, any "free and available" IO pins can be used interchangeably with the correct pin mapping configuration?
2. How would I figure out which pins are "free and available"? For eg, you did not include P1.24, P0.27 and P0.28 as possible pins. On reading the schematics on E3 Turbo, I did notice there is a pull-down resistor for P1.24. Is that why it cannot be used? How about for the case of P0.27 and P0.28? What prevents them from being used?

GadgetAngel, 2021-07-13:
> Hey @GadgetAngel, great comprehensive guide. I am looking at integrating PT100+MAX31865 using software SPI on the SKR E3 TURBO. I have a few questions I hope you can help me with:
>
> 1. Is my [...]

Yes, the reason most PINS are considered NOT "free and available" is because they have a pull-up resistor tied to it.  A pull-up resistor will interfere with the PIN's signal when using it for software SPI. P1.24 has a pull-up resistor.

The second reason most PINS are considered NOT "free and available" is because they the PIN'S "Output is open-drain".  P0.27 and P0.28 are both "Output is open-drain". See the Datasheet on LPC1769: https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf. See page 11 of the datasheet.

jinhong-, 2021-07-14:
> Yes, the reason most PINS are considered NOT "free and available" is because they have a pull-up resistor tied to it. A pull-up resistor will interfere with the PIN's signal when using it for [...]

Thank you @GadgetAngel! This is super helpful

jimberg98, 2021-09-08:
The SKR Pro v1.2 has a 6 pin header for SPI. It's on the right side just below the serial connector for the display (south east of the last thermistor connection). Is there a reason we shouldn't use that?

GadgetAngel, 2021-09-08:
> The SKR Pro v1.2 has a 6 pin header for SPI. It's on the right side just below the serial connector for the display (south east of the last thermistor connection). Is there a reason we shouldn't use that?
> 
> [image]

As shown on my Colored PIN Diagram (https://github.com/GadgetAngel/SKR-PRO-V1.2-Stepper-Driver-Jumper-Configuration-Manual/blob/master/CURRENT-Manual/HD-Diagrams/SKR_PRO_V1.2_Colored_PIN_Diagram_300ppi.pdf) the PINS on the SPI header you mentioned are the SPI pins used by the stepper motor drivers.  If you do not have any stepper motor drivers in SPI mode (in other words, all stepper motor drivers are set for UART mode) you could try to use them but if you EVER decide to use just one stepper motor drivers in SPI mode then you will cause a conflict on the stepper motor driver SPI pins.

The Stepper motor drivers' SPI PINS CAN NOT BE SHARED with other peripherals when the stepper motor drivers are using the SPI PINS (or stepper drivers are set to run in SPI mode).

jimberg98, 2021-09-09:
@GadgetAngel I don't ever plan to use SPI mode for the steppers, but I guess I will go the Extension 2 route. Thanks for your help.

shuffleznl, 2021-09-11:
Thanks for this great write-up, very much appreciated!
Quick one on using K-type; this works with Marlin sorta out-of-box on GTR
For RepRap (Gloomy FTW!) I have to define the SPI-pins of the onboard MAX31855K.
One of your [great diagram's](GTR_V1.0_Colored_PIN_Diagram_FINAL_300ppi.pdf) shows the hot junction, but no pins?

Hope you can help, thanks again.

GadgetAngel, 2021-09-11:
> Thanks for this great write-up, very much appreciated!
> Quick one on using K-type; this works with Marlin sorta out-of-box on GTR
> For RepRap (Gloomy FTW!) I have to define the SPI-pins of the onboard [...]

I think this is what you are looking for.  I found these pins defined in the PINS file "pins_BTT_GTR_V1_0.h" of the current bugfix2.0.x branch of Marlin firmware (https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x):
[image]

This is how Marlin defines the built in MAX31855 chip of the GTR 1.0 board.

CLK PIN of the built in MAX31855 = PI1
MISO PIN of the built in MAX31855 = PI2
CS PIN of the built in MAX31855 = PH9

Here are two other places that also indicate that these are the correct PINS for the MAX31855 chip on the GTR board:
1. https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/blob/master/BTT%20GTR%20V1.0/manual/GTR%20V1.0(SCH).PDF. Look at the K-Thermocouple section of the schematic diagram: 
[image]

2.  The current Color_PINS_Diagram (https://github.com/GadgetAngel/GTR-V1.0-Stepper-Driver-Jumper-Configuration-Manual/blob/master/CURRENT-Manual/HD-Diagrams/GTR_V1.0_Colored_PIN_Diagram_300ppi.pdf): 
[image]