bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #180: SKR Pro 1.2   If Limit switches are triggered, no movement response

evandene, 2020-11-23:
Symptom
When the board is fully wired, including BLTOUCH, the axis movements do not respond on triggered limit switches.
When limit switches are triggered, LED's along the stepper connectors do blink on/off accordantly.
When M119 executed all limit switches do stay triggered no matted if activated or not.
configuration files and PIN-file attached
Could somebody please help?
Thanks

[SKR_PRO_1_2.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5582104/SKR_PRO_1_2.zip)

Board SKR Pro 1.2
Delta Printer
PCB limit switches used; VCC GND SIG

evandene, 2020-11-27:
Now for three days I was trying to make my DELTA printer to work with the TMC5160 in PSI mode but without luck.
The X-Y-Z movements are okay but the enstops do not work for a DELTA printer
A DELTA printer needs to home to MAX, and that simple doesn't work
No support from BIGTREETECH as always
What a waste of time and money

evandene, 2020-12-12:
I have decided to scrap both my boards, the SKR PRO v1.1 and the SKR PRO v1.2
Both board have hardware flaws, and for both boards none or crappy support from Bigtreetech. Sad but lesson learned.

oldman4U, 2020-12-15:
Have you checked #103?

evandene, 2020-12-15:
Yep I did

oldman4U, 2020-12-15:
Last try - 502/500 to clear ERPROM?

evandene, 2020-12-16:
SKR PRO v1.1 and v1.2 do not have an EEPROM but created a soft eeprom with enabling SDSUPPORT
After every new firmware installation one should always rum M502 and M500, and so I did.
Thanks for the good work, but the boards are scrapped now and I will move on with LERDGE K boards now.

oldman4U, 2020-12-16:
Perfect. I own two of them.

Maybe you want to close this ticket once you don’t need it anymore.

Thank you

closed 2020-12-16 by evandene

oldman4U, 2020-12-16:
Thank you

Happy printing