bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #181: NTC 3950 100K Thermistor Wrong Temps

HorstHardcore, 2020-11-23:
With the recent firmware i couldnt get the thermistor working properly. Its a NTC 3950 100K Thermistor and i am supposed to use this entry in FW:

11 : 100k beta 3950 1% thermistor (Used in Keenovo AC silicone mats and most Wanhao i3 machines) (4.7k pullup)

so my entry in FW looks like this:

#define TEMP_SENSOR_0 11

For my older FW this worked as intended. Now, at room temp, the thermistor tells me like 100°C and i cant figure out, what is going wrong.

Old FW was a Marlin 2.0.x bugfix version which is at least 7 months old, new version is the recent one here (2 months old)

Any ideas?

oldman4U, 2020-11-23:
Only GadgetAngel can help you👼

HorstHardcore, 2020-11-24:
I have seen, that there were made changes in all the thermistor tables in the newer FW version. But i cant figure out, if and how this affects me and how i could solve this problem in general. So please @GadgetAngel help me :)

HorstHardcore, 2020-11-25:
Well, i finally found out, what the problem was. And i didnt expect that. I downloaded the latest FW version here and there were some major changes to the arrangement of the pin files for this board. In my older FW, all the pin settings for this board were in just one file. In the newer one, they were (probably because the newer SKR Pro 1.2) seperated into "pins_BTT_SKR_PRO_common.h" and "pins_BTT_SKR_PRO_V1_1.h" and inside of ""pins_BTT_SKR_PRO_common.h" is the problem. Someone made this:

#define TEMP_0_PIN                          PF7   //PF4   // T1 <-> E0
#define TEMP_1_PIN                          PF5   // T2 <-> E1
#define TEMP_2_PIN                          PF6   // T3 <-> E2
#define TEMP_BED_PIN                        PF3   // T0 <-> Bed

But when u check the pins of the 1.1 Pro, u can see, that PF7 is part of the "extension pins" and PF4 is supposed to be the original pin for the thermistor and within the older FW PF4 was set to TEMP_0_PIN. So i changed it back to

#define TEMP_0_PIN                          PF4   // T1 <-> E0
#define TEMP_1_PIN                          PF5   // T2 <-> E1
#define TEMP_2_PIN                          PF6   // T3 <-> E2
#define TEMP_BED_PIN                        PF3   // T0 <-> Bed

Now at least this Problem is solved and my NTC 3950 is working as intended again. Maybe some1 with more knowledge about github could check that, so that other ppl dont run into this problem. I am not sure, if PF7 set to TEMP_0_PIN is the way it should be in here.

closed 2020-12-02 by HorstHardcore

GadgetAngel, 2020-12-02:
I just saw you request.  Glad you were able to figure it out without my help.  Yes, the boards have changed.  There must of been a reason for the PIN swap that you experienced.  Sorry you had the frustration.  Merry Christmas!