bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #183: BIGTREE_SKR_PRO, Oops! Make sure you have the LPC1768 environment selected in your IDE.

jlmxyz, 2020-12-08:
**version used :** 
----------------------------------------------------------------------------------------------
commit 3327568142f0294fa2b6e69b8b6567fa345ad205 (HEAD -> 2.0.x, origin/HEAD, origin/2.0.x)
Author: Luu Lac <45380455+shitcreek@users.noreply.github.com>
Date:   Tue Dec 1 23:52:09 2020 -0600

    Help hosts when password-locked (#20348)

**product tunning :** 
-------------------------------------------------------------------------------------------------------------
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index db266b524b..6b34ac5b47 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -103,13 +103,13 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT 1
 
 /**
  * Select a secondary serial port on the board to use for communication with the host.
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-//#define SERIAL_PORT_2 -1
+#define SERIAL_PORT_2 -1
 
 /**
  * This setting determines the communication speed of the printer.
@@ -127,7 +127,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_SKR_V1_1
 #endif
 
 // Name displayed in the LCD "Ready" message and Info menu
diff --git a/platformio.ini b/platformio.ini
index 0abb4ac388..e7f10acfe7 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -18,7 +18,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = BIGTREE_SKR_PRO
 include_dir  = Marlin
 
---------------------------------------------------------------------------------------
**when I try to build the target, I get the following error :** 
---------------------------------------------------------------------------------------
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp.o
In file included from Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_V1_1.h:236,
                 from Marlin/src/HAL/STM32/../../inc/../pins/pins.h:394,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:34,
                 from Marlin/src/HAL/STM32/HAL.cpp:28:
Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_common.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
   29 |   #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_V1_1.h:236,
                 from Marlin/src/HAL/STM32/../../inc/../pins/pins.h:394,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:34,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:25:
Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_common.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
   29 |   #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_V1_1.h:236,
                 from Marlin/src/HAL/STM32/../../inc/../pins/pins.h:394,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:34,
                 from Marlin/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp:23:
Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_common.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
   29 |   #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_V1_1.h:236,
                 from Marlin/src/HAL/STM32/../../inc/../pins/pins.h:394,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:34,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:22:
Marlin/src/HAL/STM32/../../inc/../pins/lpc1768/pins_BTT_SKR_common.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
   29 |   #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
      |    ^~~~~
*** [.pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/HAL.cpp.o] Error 1
*** [.pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp.o] Error 1
*** [.pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/MarlinSerial.cpp.o] Error 1
*** [.pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/HAL_SPI.cpp.o] Error 1
============================== [FAILED] Took 16.24 seconds ==============================
------------------------------------------------------------------------------------------

GadgetAngel, 2020-12-10:
In Platformioini file ensure default_envs= BIGTREE_SKR_PRO

In configuration.h ensure MOTHERBOARD BOARD_BTT_SKR_PRO_V1_1

Looks like you have MOTHERBOARD BOARD_BTT_SKR_V1_1.

BOARD_BTT_SKR_V1_1 is not BOARD_BTT_SKR_PRO_V1_1

jlmxyz, 2020-12-10:
OK my mistake, I was misslead by the hit that is given when MOTHERBOARD is set to BIGTREE_SKR_PRO as said in https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/BIGTREETECH%20SKR%20PRO%20V1.1%20guide%202019%20-%201.pdf while compiling it said that it's deprecated and that I should use BOARD_BTT_SKR_V1_1 instead....
compilation was successfull! thanks, now I can begin tunning for my project!

closed 2020-12-10 by jlmxyz