bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #186: BIGTREETECH-SKR-PRO-V1.2 SD CARD

JanHH2019, 2021-01-06:
I have configured and installed a current version of marlin 2. that's how far everything goes. when i go to print it shows me the onboard sd card and the sd card of the tft35. He shows me the files on the TFT SD card. When I choose the onboard sd card it just shows "Loading ..." and nothing happens anymore.

oldman4U, 2021-01-07:
The TFT shows the on board sd as soon as it is configured in software. Personally I was never able to get the on board sd card to work and use only the TFT.

mitchcecg, 2021-01-14:
I am having the same sort of issue any updates on this would be helpful

JanHH2019, 2021-01-23:
The problem persists. I have now used the Marlin 2.0 bugfix version and activated SDSupport there.
Both SD cards are displayed on the TFT35 V2.0 but I cannot access the onboard SD
[image]

JanHH2019, 2021-01-23:
now I've got an esp module and installed it. I put it on the TFT35 V2 and it reports on the network. i can access it and everything works except for accessing the sd card
now the question how do I get it?