bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #187: TMC init error with TMC2209

eeyoredave, 2021-01-10:
Hi there,

I am trying to configure the 1.2 board for use on a Voron 2.4 running Klipper. I seem to have everything going well until I boot up with the motors connected. This is the error I get within Octopi: "Unable to read tmc uart 'stepper_x' register IFCNT Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software. Printer is halted" and when I check the log, I get the "TMC init error". 

Have checked the voltages and the jumpers and appears to be correct. But has my motor cables connected wrong way? But the error still occurs after I unplug all the motor cables. 

[image]
)

GadgetAngel, 2021-01-12:
Please check this out maybe it can help: https://github.com/KevinOConnor/klipper/issues/2741

oldman4U, 2021-01-12:
Hi GadgetAngel, long time no "see". Hope you are doing fine!

Maybe not related, maybe it is...: https://github.com/MarlinFirmware/Marlin/pull/20731

Wish you the best

NovaViper, 2021-01-13:
I'm having the same issue on mine aswell! The oddest thing is for me however is that UART+Sensorless homing works perfectly on Marlin

NovaViper, 2021-01-13:
OH! I just fixed the issue! Make sure you *do not* have the `tx_pin` parameter declared! The UART+sensorless homing configuration for TMC2209s should look like this
```
[tmc2209 stepper_x]
uart_address: 0
uart_pin: PC13
run_current: 0.760
hold_current: 0.500
stealthchop_threshold: 250
diag_pin: PB10
driver_SGTHRS: 100
```

eeyoredave, 2021-01-14:
Thanks for the suggestions... I had not used the tx_pin either, so that did not help. I've checked the two suggested threads and it did not help. I have given up for the moment, and gone to a SKR-1.4 turbo board, and it works well with the same TMC2209 chips installed. I have got the printer working with the motors, so I am not sure if it is an issue with the Pro hardware or the firmware.

Was there a GPIO pin setting that I needed to enable on the Pro to get it working? I know for the SKR-1.4 I had to compile the GPIO state as !P2.9 and SKR mini-E3 !PC13, which is the "USB_CONN". But I could not find the corresponding pin on the Pro, perhaps someone can point me to that pin?

GadgetAngel, 2021-01-14:
@eeyoredave please post your config file for klipper