bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #189: USB not working with Marlin

StefT7, 2021-01-15:
Hello,
I'm unable to speak with my card using USB cable.
Using this project or last Marlin project, nothing to do, each firmware.bin generated and uploaded has the same problem. I can create and open a connection using powershell commands, I can send M115 command but I can't read any resulting output.
It's not an hardware problem because original firmware stored in SD card well know USB, same test with that firmware was good. 
If you look inside this original firmware, you see that it was compiled using BLACK_F407VE card
[image]

So, is anybody could help me to compil Marlin and make USB working well with BigTree_SKR_Pro?

StefT7, 2021-01-17:
Without any debuging solution I decided to make green led to blink at some break points.
I found that the problem is that lineState propertie is always false. If I don't test lineState value in CDC_connected everything works well.
Someone could tell me how this lineState works?

HTTYD9, 2021-04-28:
Marlin bug fix 2.0.8
TFT 2.0.5.4

Skr Pro v1.1
TFT35 V3.0
TMC2209

I followed instructions but still unable to print from usb pen drive via skr pro v1.1 and when i insert pendrive in TFT35 V3.0 still shows error .

pendrive is formatted to FAT32 , i even try exFAT  but no luck
[image]

ivanroberto, 2021-05-27:
Hi, I figure it out how it need to be setup:

Put jumpers on HOST, and all options like picture above.

With Marlin 2.0.8.1, in configuration_adv.h, i made more two changes:
#define USB_FLASH_DRIVE_SUPPORT
  #define USE_OTG_USB_HOST
 
And compile with enviroment: BiGTREE_SKR_PRO_usb_flash_drive
I did this using Platformio

I hope it helped you

svd05, 2022-01-17:
Hi. What you mean when said ?
>  i made more two changes
What you are did more ?
You have the same environments in stm34f4.ini ?
[image]
If I compiling without flag "DUSBCON" I have error:
[image]
[image]