bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #190: SKR-PRO-V1.2 with TFT 35 V2 and Marlin 2

JanHH2019, 2021-01-18:
I've searched the net for hours and haven't found anything breakable. I can't get my SKR Pro 1.2 and the TFT 35 V2 to work. No matter what I enter in configuration.h for Marlin 2.0, I always get the message "No printer attched". What do I have to enter for serial_port and serial_port2?

Even my attempts to reach the SKR board via Ponterface from my computer don't really work. I once managed to establish a connection but that only worked once.
if I set serial_port to -1 and turn off serial_port_2 it works with the connection. as soon as I switch on serial_port_2 and no matter what, nothing works anymore.

I'm angry now that I even took a SKR board.

JanHH2019, 2021-01-18:
Now the board seems to be finally over. I can do what I want, I can no longer get in touch with ponterface. the computer recognizes the com port STM Virtual com port, but ponterface no longer connects

JanHH2019, 2021-01-18:
now i have activated the original firmware from the sd card that came with the board. and so the tft35 V2 works and also the connection to the computer and ponterface.
Now the big question is how does the configuration in the marlin have to look like so that it works

oldman4U, 2021-01-18:
Check the ReadMe on the first page of the TFT firmware repository for more information.

JanHH2019, 2021-01-18:
Thanks for this helpful information ....
and yes i read that, but that doesn't help me a bit.

Connect the TFT to the Mainboard
Do the following, to be able to use the touchmode of your screen.

Connect the 5pin serial cable according to the manual of your mainboard.
Define a serial port in Marlin, to activate the port used by the TFT.
Make sure the same BAUDRATE is defined in Marlin and the firmware of your TFT (config.ini)

Unfortunately, it doesn't say which com port the tft connection is in the instructions for the SKR 1.2 Pro

 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 1

oldman4U, 2021-01-18:
Try 1

JanHH2019, 2021-01-18:
Ok ...
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
that is my current attitude and that does not work ....

oldman4U, 2021-01-18:
And baudrate is the same?

Cable?

JanHH2019, 2021-01-19:
the baud rate is the same, the cable seems optically ok.
have now tried a new marlin version. now it works every now and then. have now made several new versions to adjust the direction of the axes and the limit switches. but as I said, sometimes it works and sometimes it doesn't. extremely unreliable and strange system

oldman4U, 2021-01-19:
Most likely a cable problem

oldman4U, 2021-01-19:
Or something else connected to the display or mainboard causes this. In software only the serial port and baud rate is relevant for your setup.

JanHH2019, 2021-01-19:
thank you for your help. i found the mistake by accident. the SD card is defective and apparently he could not read the entire bin file. I noticed that because I wanted to rename the .CUR version and he told me that it was a 0 byte file. my attempt to format the SD card failed. with a new sd card it works right away.

closed 2021-01-19 by JanHH2019

oldman4U, 2021-01-19:
Great!

Happy printing