bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #191: SKR_PRO_V1.1 Marlin 2.0 SKR-PRO LAYER SHIFTING

luckyscustomleather, 2021-01-18:
I have installed the SKR-PRO-V1.1 and the TFT35 V3.0 to my Tornado Tevo Printer, managed to get everything working well with the exception of random but constant layer shifting on Y axis.

I have tried modifying the Config files as recommended on various other sites that seem to be having this problem with Marlin 2.0.
I have tried to lower the movement speed by 25%
And still no changes, 

Is there ant issues in my configs, or is this a Marlin 2.0 bug, in which case does anyone have Marlin 1.1.9 for a SKR_PRO_V1.1?

I have attached my Config files, 
[advanced_configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5831860/advanced_configuration.h.txt)
[configuation.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5831861/configuation.h.txt)

radek8, 2021-01-20:
You're using a pretty old Marlin.
I would not downgrade to version 1.1.9
I would upgrade to the latest versions.

luckyscustomleather, 2021-01-28:
> You're using a pretty old Marlin.
> I would not downgrade to version 1.1.9
> I would upgrade to the latest versions.

Hi, I am will try this and see what can be done.

radek8, 2021-01-28:
If the Y-axis step is lost, check the current at the Y motor

radek8, 2021-01-28:
800mA, that could be OK

luckyscustomleather, 2021-02-02:
Hello Radek8, thank you I have tried this and the same results, what I have done now is buy some TMC2209 steppers and am now trying to configure these in UART. with Sensor less homing for x and y, I am not sure the SKR-Pro-V1.1. supports these yet properly.

radek8, 2021-02-02:
TMC22069 drivers are great, I have them too :+1:

luckyscustomleather, 2021-02-02:
Are you using them on the skr pro v1.1? If so can you offer some Config settings to get them to work in UART?

radek8, 2021-02-02:
No . I have SKRV1.4
You have to have it right.
- fitted jumpers and on the SKR board
- jumpers on 2209 controllers (if not from production)
- Configured Marlin
Then it must work

radek8, 2021-02-02:
[image]

radek8, 2021-02-02:
[image]

radek8, 2021-02-02:
[image]

radek8, 2021-02-02:
Set motor currents X Y Z E
  #if AXIS_IS_TMC (X)
     #define X_CURRENT 700

uncoment
 #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E
 #define CHOPPER_TIMING CHOPPER_DEFAULT_24V   // or12V?
 #define MONITOR_DRIVER_STATUS
 #define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     100  // [mm/s]
  #define Y_HYBRID_THRESHOLD     100
  #define Z_HYBRID_THRESHOLD       3
  #define E0_HYBRID_THRESHOLD     30
  #define E1_HYBRID_THRESHOLD     30
  #define TMC_DEBUG

Roman-ru, 2021-03-12:
I have a similar problem, the printer seems to change the steps on the axis y and the detail gives more, although from the beginning the printing starts normally.