bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #192: Z height incorrect

TimListon, 2021-01-21:
I'm not sure where the right place to get assistance is so I'm starting here. If someone has a recommendation of somewhere else to go I'd appreciate that info as well.

I recently swapped out my board with a Bigtreetech SKR Pro 1.2 and a TFT E3 v3 display with my BLTouch. I've compiled Marlin 2.0 Bugfix and provided a link to the config files below.

[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5852579/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/5852580/Configuration.h.txt)

The picture is the result of running the following commands:
G28
G28 Z
G1 F60 Z0

The problem is that whenever I home to Z0 the nozzle ends roughly 20-25mm off the build plate.
[image]

I was able to create a mesh by disabling the steppers and moving the nozzle closer to the bed and then initiating a G29, but if I try to run G29 after G28 the process errors out because it is limited to traveling 20mm to find the build plate. I could increase this but it seems like it should be starting closer to the build plate to begin with.

Perhaps I'm not searching on the right terms to find the answer, but I've been trying to troubleshoot this for about 3 days (on and off).

schnello, 2021-02-02:
Hi.

Take a look at:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, **400**, 96 } // Customized TL

TimListon, 2021-09-24:
I took a break for a while on this, but finally got back around to troubleshooting it.

It appears the problem ended up being that I needed to uncomment the following.
#define USE_PROBE_FOR_Z_HOMING //Customized TL

The printer is now homing correctly and generating its mesh without having to manually move the Z axis closer to the bed.

closed 2021-09-24 by TimListon