bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #193: TFT70 does not connect after reset SKR Pro 1.2

fiveorders, 2021-02-04:
Hello, I am using TFT70 with SKR Pro 1.2 When I reset SKR Pro TFT does not connect after booting up. It started today. any ideas? Thank you

oldman4U, 2021-02-14:
You can try to connect RS232 only to see if this solves the problem.

oldman4U, 2021-02-25:
Please help the community and close the ticket once you do not need it anymore. Thank you

fiveorders, 2021-02-25:
Same thing happening with the FLY board with 9 stepper drivers. Somehow reset does not function as it should do. I am testing further, will report back later. Closing this now.

closed 2021-02-25 by fiveorders

EsaJaatinen, 2021-07-07:
Hi

I'm having issues with my SKR pro 1.2 with the reset button also. Everything works ok untill I hit the reset button. After that it seems that Marlin doesn't restart at all.

After reset I can't:
- Get connection to octopi (jumper wires directly to SKR, works normally after off period)
- Win 10 doesn't see stm32 serial port (works normally after off period)
- Hotend cooling fan doesn't start (starts after off period if temperature over threshold)

I'm thinking that either the stm32 or Marlin doesn't start normally.

'ej

EsaJaatinen, 2023-05-27:
Hi

I'm back to the project and still "no printer attached" after hitting the reset button.

I updated Marlin to 2.1.2.1, Octoprint operation works normally, BTT TFT35 find the printer. No connection to either after reset.

Any ideas?

'ej