bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #194: How do you setup Dual Z?

NovaViper, 2021-02-24:
Hey guys, im running Klipper on my SKR Pro, and Im trying to figure out how to setup Dual Z motors on the board.. my driver is a TMC2208 in UART mode

oldman4U, 2021-02-25:
Using Marlin the firmware does not know about the second one because it is a feature of the hardware itself.

Cheo7811, 2021-02-25:
oldman4U, Marling actually have a section for multiple stepper for Z, from 1 up to 4.

Line 576 in Marlin 2.0.7.2

#define NUM_Z_STEPPER_DRIVERS 1   // (1-4) Z options change based on how many

The question was, how to setup 2 on Klipper.

oldman4U, 2021-02-25:
Maybe you should start a ticket in a Klipper forum then.

Most likely Klipper also has to be configured like a single Z because for
the firmware it works like a single Z.

Cheo7811, 2021-02-25:
NovaViper:

There is a section for multiple steppers in the Klipper config, that should do the trick

https://www.klipper3d.org/Config_Reference.html#additional-stepper-motors-and-extruders

Perhaps that is enough for you to close this ticket

NovaViper, 2021-02-25:
@Cheo7811 Thanks! I got it working and I found someone's config that does Dual Z on a SKR Pro

closed 2021-02-25 by NovaViper