bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #195: SKR PRO V1.2 CPU Got Burned

msmorkan, 2021-03-12:
Hi,

I had an SKR PRO V1.2 recently. I tried to configure it using Marlin in this repo. I could not get it working. Then I switched to the Marlin repository (https://github.com/MarlinFirmware/Marlin). I changed drivers to TMC2208 (UART), board to SKR PRO 1.2, increased maximum values of X/Y/Z  and made several changes written in the manual. Then I builded and suprisingly it worked but I have doubt that it is fully correct. Because the orginal firmware (with TMC2209) was 172 KB size, but my firmware is around 103 KB size and this is huge difference. 

Then I plugged my step motors and drivers and Bigtreetech TFT43. I have driven X/Y/Z without a problem. heated the bed (60) and nozzle (200) When I tried to extrude some filament. It made and annoying sound wih just vibration no movement. Then it said something like "Printer halted kill called" I resetted both boards and reconnected to board with no problem. After I tried several times I was not able to connect to the mainboard anymore. CPU is getting too hot when I plug the power in proably burned. Also he 3V3 regulator is heating up too much, probably because of burned MCU.

Since TMC2208 drivers are controlled by UART what can cause this problem? We will manufacture 3D printers using SKR PRO so we have to make sure that it wont happen again. 

I bought another set of drivers and SKR PRO and I am trying to be very careful to the details.

Here are the 3D printer details:
- 24V 350W power supply (Meanwell)
- Standart hot end.
- double extruder fan
- double cooling fan
- special made extruder block with direct drive
- NEMA17 steppers
- Special made hot bed driven by external mosfet
- 300x300x350 printing area (X/Y/Z)
- CORE XY

In summary here are my questions:
- What can cause burned MCU?
-  In configuration_adv.h file, there is an option with CHOPPER_TIMING and it is set to CHOPPER_DEFAULT_12V. Should I change it to 24V since my power supply is 24V?
- Why the firmware file size is shrinked to 103 KB?
- I sometimes get "unknown command error", how can I prevent screen sending unknown commands to the mainboard?
- For default settings with the given details; Which features should I enable and Which registers should I change to print something without a problem?

Thanks a lot
Have a nice day
Suat

oldman4U, 2021-03-12:
Hi

CHOPPER_DEFAULT_24V is correct in your case.

My experience with this board is very good compared to SKR V1.4, especially when it comes to robustness.

msmorkan, 2021-03-13:
> Hi
> 
> CHOPPER_DEFAULT_24V is correct in your case.
> 
> My experience with this board is very good compared to SKR V1.4, especially when it comes to robustness.

Hello,

thanks for the tip. Do you have any thoughts about the other questions?

Thanks

oldman4U, 2021-03-13:
Printer halted I saw from broken thermistor.

Replace mainboard and drivers if you don't know what the problem was.

I use SKR PRO v1.1 with BTT 2208 UART and it works great.