bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #204: skr pro v1.2 + heat bed ender 3 220w 24vdc

dec3003, 2021-05-30:
hello i read in the manual that is maximum 180W power bed with at minima 0.8ohms for resistance from bed,
the bed from ender 3 is a 2.8 ohm 220w 24 vdc 
if i calculate Amp:
24/2.8= ~8.6A  
if we compare that with a bed 180W 12vdc:
12/0.8= 15A
so we are under max Amp with ender bed configuration

so i can use this heat bed (ender 3 220w) 
can you confirm it please

radek8, 2021-05-30:
Used Mosfet WSK220N04 is designed for 220A with ideal cooling.
A 20A fuse is used in the connection.
So your 8.6A is fine.

dec3003, 2021-05-30:
thank you 
i just burned one mother board skr pro today an search about the bad thing i have done,
so that's not with the bed ^^

closed 2021-05-30 by dec3003

radek8, 2021-05-30:
I also have Ender3
I replaced the board with SKR V1.4 turbo.
It uses the same Mosfet and works well for several months.
What burned down?

radek8, 2021-05-30:
Isn't the problem just in fuse?

dec3003, 2021-05-31:
the stm32f maked some smoke, i heard some sparks, 
but at first look i dont found what's happened , i finish to found a burned driver (tmc2130) and a pin burned on the stm32f
i define it in configuration.h it's the z motor who burned, but when i look on the stm32 (mcu) its de pin PA3 who was fondue E1 motor
i wait for my new motherboard for friday and i m a little affraid about make burn it again

i maked an update about motors i changed the z with a tmc and add a tmc on e1, before that was the a4988, 

maybe remove drivers, make the update, and put in the driver again is better.

radek8, 2021-05-31:
If you had incorrectly configured stepper motor controllers in Merlin, they could be overloaded with a large current and the controller and board chip could burn out.

dec3003, 2021-05-31:
i think that was happened, i taken a look at my configuration.h i have forgotten the jumper on the E1 driver they was  like a4988