bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #207: SKR pro reset button issue, Marlin/stm32 doesn't seem to start after pressing it

EsaJaatinen, 2021-07-08:
Hi

I'm having issues with my SKR pro 1.2 with the reset button. Everything works ok untill I hit the reset button. After that it seems that Marlin doesn't restart at all.

After reset I can't:

-    Get connection to octopi (jumper wires directly to SKR, works normally after off period)
-    Win 10 doesn't see stm32 serial port (works normally after off period)
-    Hotend cooling fan doesn't start (starts after off period if temperature over threshold)

I'm thinking that either the stm32 or Marlin doesn't start normally.

Any hints are most welcome.

'ej

EsaJaatinen, 2023-05-28:
I'm still having this issue. Using Marlin 2.1.2.1 without any issues with serial to Octoprint and BTT TFT35 from power on start. No glitches, and long prints work ok. Only after hitting the reset button neither raspi or tft cannot find the SKR pro.

Any ideas?

yavinmaster, 2025-01-05:
I've having the same issue after updating from Marlin 2.0.5 to 2.1.2.