bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #208: SKR_PRO_1.2 connection issue

Alchoni, 2021-07-25:
Hello, it appears that I broke something in Marlin configs or in my hardware SKR_PRO_1.2 (third option is a win10 comport issues). I had X axis working and then changed few things in configs (add more steppers and axis) then windows stop detecting my comport, after a while I realized that it was in a "show hidden" tab but still, I cannot connect to the board by pronterface. I then tried to use my original, working configs but the problem stayed as it was after changing it to a config with more steppers. I tried every solution to a similar problems from all forums from the first page of google, so things like same comport, comport numbers higher than 2, com-rate compatibility, restarting, rebuilding Marlin... those are all check and did not worked (If I've done everything correct, witch I think I did)

**FEW OF MY RANDOM DISCORD CONVERSATIONS:**
--------------------------------------------
_From Device Manager: Currently, this hardware device is not connected to the computer. (Code 45)

To fix this problem, reconnect this hardware device to the computer.
Ofcourse it is connected, comport in pronterface and device manager is the same, bitrate in config and manager is also the same_
--------------------------------------------
_And once again, every time I restart manager I need to "unhide" to see this comport and it's gray'ish, it seems that's it can't read it
I will try few more solutions from forums, I really hope the usb port isn't dead
Information from events tab in Device manager: Device USB\VID_0483&PID_5740\2090318E5252 requires further installation.
Reinstalling pronter and running as an admin does not do the trick :c_
--------------------------------------------
_I changed comport to 5 and rebuild the Marlin, no success_
--------------------------------------------
_tried to set first comport to -1 and second one to 5 in build but nothing changed, now I can't even see comport in after unhiding_
--------------------------------------------

Alchoni, 2021-07-25:
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/6873539/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/6873540/Configuration_adv.txt)

jimberg98, 2021-09-08:
Have you figured this out yet or are you still waiting for an answer? If you are, can you provide what your setup is? Do you have a display? Etc.?