bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #211: SKR PRO v1.1 and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER not connecting

cabbagecreek, 2021-09-12:
I have a board that have been in the drawer a wile. This weekend i decided to update my MKS Gen 1.4 to SKR PRO v1.1

I did everything from scratch with a fresh Marlin "Marlin 2.0 with bug fixes" download.

I have an old "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" from the MKS Gen 1.4 setup that is working.
I followed all tutorials that I can find to setup this old "Marlin style" LCD board.

I can't get any response from the LCD. It is black all the time. 

When I install the 2 flatcables and press the encoder wheel i hear a beep from the PC and the USB is disconnected. I need to press reset on the SKR Pro board to get USB connected again.

Can this be a broken SKR PRO v1.1 board?

Before sending my congiguration.h and configuration_adv.h I want to get some parts to serach for.

Have done activating the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER by uncommenting and get a build done.
The FW is uploaded with SD card. The 3D printer is working but no REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

radek8, 2021-09-16:
Note that the SKR and MKS boards have a rotated cutout in the connector on the EXP 1, EXP2 or EXP3 connectors by 180 °.
Check which pins are powered.

cabbagecreek, 2021-09-19:
Marvelous!!
"Standard is a good thing. That´s why all manufacturers create their own"
This will be an easy task to just pull off the IDC10 sockets on the PCB and then turn 180 degrees, 
I will be back with the result of this!

cabbagecreek, 2021-09-21:
I turned the sockets on the LCD 180 degrees and put them back on.
The LCD showed the Marlin logo and the LCD was working again.