bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #212: Bricked board?

luckyfp, 2021-09-21:
So i decided to flash klipper on skr. Everything was just fine. I flashed firmware on board, changed serial on printer.cfg and it worked. I just wanted to check if everything is good so I heated the bed and hot end to like 40 degrees, and then lost connection. It reconnected and then disconnected again. It shows up this "mcu 'mcu': Unable to connect".
I decided to check serial with ls /dev/serial/by-id/* but it says that there is no such file or directory. So i tried to flash the firmware again but nothing happend, it didn't flash; the blue light is off all te time. 
I ordered st-link but im curious if anything else could happend?

sealj553, 2022-01-01:
I have a similar situation. I've been running my SKR PRO V1.2 board with Klipper for around 6 months with no issues. At some point last week klippy could not connect to the board and I discovered that I also lost my board under `/dev/serial/by-id/`. I had changed nothing with the wiring at all for months.
`dmesg` shows: 
```
[  291.076704] usb 1-1.4: new low-speed USB device number 3 using xhci_hcd
[  291.177009] usb 1-1.4: device descriptor read/64, error -32
[  291.396975] usb 1-1.4: device descriptor read/64, error -32
[  291.616716] usb 1-1.4: new low-speed USB device number 4 using xhci_hcd
[  291.716988] usb 1-1.4: device descriptor read/64, error -32
[  291.936971] usb 1-1.4: device descriptor read/64, error -32
[  292.057428] usb 1-1-port4: attempt power cycle
```
I tried a different pi, no change. Arduino boards still enumerate.

I reflashed the bootloader over serial as described [here](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/115#issuecomment-653814967) and put a freshly formatted microsd card with klipper. Same results.

I need to debug if it's a problem with the klipper firmware or the bootloader or if the STM32 is fried...

sealj553, 2022-01-01:
`FIRMWARE.CUR` is still created on the microsd card, so it doesn't look like my STM32 is fried

sealj553, 2022-01-01:
I compiled a new version of klipper using USART1 (PA10/PA9) (on the TFT header) instead of USB. After hooking it up to my Pi with a FTDI adapter, it is working again.

fabienlacaze, 2022-12-31:
hi seal 553, same issue for me. can you kindly explain more how to "compiled a new version of klipper using USART1 (PA10/PA9) (on the TFT header)"
thanks