bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #217: Can you use different voltages for the motor power and main board power

jayjar291, 2022-03-07:
Basically, wondering if I can run 24 volts for my motors and 12 volts for the main board to include the heater, fans, etc.

radek8, 2022-04-11:
Yes it is possible.
But you have to power the board 24V. this will provide 24V for the motors.
Fans and heating nozzles and the bed can then supply 12 volts as needed.
Switched to negative branch. Therefore, you can supply the other peripherals with a different voltage. But don't forget to put a fuse in the circuit. otherwise you will detonate the mosfets in the event of an accidental short circuit.

radek8, 2022-04-11:
The SKR PRO board even has a separate power supply for the motors. Then there is nothing to solve :-)