bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #218: Ender 3 v2 display on SKR Pro 1.2?

nuker1110, 2022-03-10:
I found [this issue](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/424) detailing a pinout for the SKR Mini E3 to use the stock display from my Ender 3 v2 on it, but I can't picture how to translate those instructions to the display pinouts for the Pro boards.
[image]
and 
[image]

I came here for help because I don't want to fry anything through trial-and-error.

charlesrg, 2022-05-06:
@bigtreetech could you please port the BTT SKR Mini E3 , Ender 3 V2 display support to the SKR Pro ?
This will add compatibility and more users could order it for their Ender 3 V2s and Clones like the Voxelab Aquila.

charlesrg, 2022-05-06:
I'm going to attempt this: 
  #define BEEPER_PIN                      EXP1_10_PIN
  #define BTN_EN1                         EXP1_08_PIN
  #define BTN_EN2                         EXP1_06_PIN
  #define BTN_ENC                         EXP1_09_PIN

Will see if it works.

nuker1110, 2022-05-10:
> I'm testing using
> #define DWIN_CREALITY_LCD_JYERSUI // Jyers UI by Jacob Myers

Have you gotten this to run yet?

charlesrg, 2022-05-10:
I did, however it did not work. I'm about to order a BTT TFT35 and give up. Wasting too much time on this.

charlesrg, 2022-05-10:
I did, however it didn't work. I'm about to order a TFT35 and quit trying. Wasting too much time on this.

charlesrg, 2022-05-12:
> I did, however it did not work. I'm about to order a BTT TFT35 and give up. Wasting too much time on this.

I retract this. I've decided to do one last shot. And I got it working. Pinout is much simpler than I thought. 
The Ender 3 display cable goes straight to EXT1 with the exception of TX/RX, those you need to connect to the Board TFT  RX/TX TFT.
`
  #define BEEPER_PIN                      EXP1_05_PIN
  #define BTN_EN1                         EXP1_03_PIN
  #define BTN_EN2                         EXP1_04_PIN
  #define BTN_ENC                         EXP1_06_PIN
`

closed 2022-06-02 by nuker1110

charlesrg, 2022-06-02:
It works fine, I gave up on the TFT35 and I'm using the original display.