bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #219: Two new Pro v1.2 board wont flash

Mountain3d, 2022-03-16:
Hello,

I am no stranger to the space, I run a CAD/Printing company in Montana. I've followed the very basic steps to flash probably 1000 boards for myself a folks over the years. 

I picked up these two skr pro 1.2 boards for my IDEX builds for the store and neither of them will flash fw. I BTT products usually are a massive gamble as many of the ones I've either had or sold has some stupid problem. However these were the best option fir the budget of these builds. 

The task? Simple. Flash my proven builds on to these boards and get them installed. I learned a long time ago especially with BTT products to do a "bench flash" before committing to the install and burial of a mabo. 

They aren't shipping with SD cards anymore as I know those as generally the best for fw's

I power down, pop in tf card and power up. Nada. I tried the second board, same same. I'm at a total loss how in the h•e•l•l can I get two mabo's that are trash in the exact same way from two different parts of the globe.

If anyone has any insights I would greatly appreciate the help as these two machines are severely overdue.

TIA!

Mountain3d, 2022-03-28:
> I'm having the same exact issue with 2x new skr pro 1.2 . Like just
> received one today. I'm no stranger to the skr pro 1.2 either . I have
> around 8 or 9 of them

I figured it out.

Insert SD card in to computer

Type in "diskpart" one word.

Your SD will the last drive available so let's just say 3 for example
.

Type " select disk 3"

The. Upon the next prompt type "clean"

This will.strip the sD. You can close the window now

Now, type "Disk Part" two words this time.

Down at the bottom scroll to the bottom of the window and you will see an
unallocated drive being your SD you just stripped. Right click, format,
fat32, now type in 7024 in the next window for available space. Finish
Finish format. Load your .bin and flash the board.. its got some space cap
for cards idk how or why but this is the trick. Read what I wrote three
times. Follow the steps.

Food luck!

Mountain3d, 2022-03-28:
EDIT

SKIPPED A STEP

>
>
> Insert SD card in to computer
>
>
> Type in "diskpart" one word.

 THEN TYPE "LIST DISK"******

SodiumEx, 2022-03-28:
I did what you said and still no luck

Mountain3d, 2022-03-28:
Can you zoom?

SodiumEx, 2022-03-28:
Share your marlin if possible

Mountain3d, 2022-03-29:
If your board isn't converting the .bin to .cur then its not the build. Its
simply not reading the card.

SodiumEx, 2022-03-29:
> If your board isn't converting the .bin to .cur then its not the build. Its simply not reading the card.
> […](#)

It does build to .cur . Just won't do anything after